Koenkk/zigbee2mqtt #6788: UnhandledPromiseRejectionWarning, but working when using npm start

ikke-zelf, 2021-03-22:
## What happened
Running "sudo system status zigbee2mqtt" gives warning and zigbee2mqtt does not work (although it says 'active (running)'.
Starting with the command 'npm start' shows me the correct outcome and even the webinterface is opened correctly.

error output:
`zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-03-22 14:30:46 CET; 2h 36min ago
 Main PID: 553 (node)
    Tasks: 23 (limit: 4915)
   CGroup: /system.slice/zigbee2mqtt.service
           ├─553 npm
           ├─707 sh -c node index.js
           └─708 node index.js

Mar 22 14:31:01 MasterPi4 npm[553]:     at /opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:336:39
Mar 22 14:31:01 MasterPi4 npm[553]:     at Array.forEach (<anonymous>)
Mar 22 14:31:01 MasterPi4 npm[553]:     at rmkidsSync (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:336:26)
Mar 22 14:31:01 MasterPi4 npm[553]:     at rmdirSync (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:329:7)
Mar 22 14:31:01 MasterPi4 npm[553]:     at Function.rimrafSync [as sync] (/opt/zigbee2mqtt/node_modules/rimraf/rimraf.js:301:9)
Mar 22 14:31:01 MasterPi4 npm[553]:     at /opt/zigbee2mqtt/lib/util/logger.js:117:20
Mar 22 14:31:01 MasterPi4 npm[553]:     at Array.forEach (<anonymous>)
Mar 22 14:31:01 MasterPi4 npm[553]:     at DerivedLogger.cleanup (/opt/zigbee2mqtt/lib/util/logger.js:115:21)
Mar 22 14:31:01 MasterPi4 npm[553]: (node:708) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandl
Mar 22 14:31:01 MasterPi4 npm[553]: (node:708) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
~`

## What did you expect to happen
As before. Just to work as a service.

## How to reproduce it (minimal and precise)
running 'sudo systemctl start sigbee2mqtt' and 'sudo systemctl status zigbee2mqtt'

## Debug info
Zigbee2MQTT version:
- node version: v12.21.0
- npm version: v6.14.11

Adapter hardware: CC2531
Adapter firmware version:
- tried both the 'default' & 'source_routing':
- https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20201127.zip

Koenkk, 2021-03-22:
Looks like a permission issue, fix the right on `/opt/zigbee2mqtt` or run zigbee2mqtt as root.

ikke-zelf, 2021-03-22:
When I isseu 'systemctl stop zigbee2mqtt.service' and then ask fro the status I get this as one of the lines:
`(node:708) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 4)`

ikke-zelf, 2021-03-22:
strange... i rebooted a dozen of times already, reinstalled npm etc.
now after issueing: 'systemctl stop zigbee2mqtt.service' and issueing a start again. He Works!

but i did not change anything in my opinion.

ikke-zelf, 2021-03-22:
nevertheless. if it stays running I'm happy. for now, thanks for the help.

closed 2021-03-22 by ikke-zelf