Koenkk/zigbee2mqtt #6796: Get does not work with Saswell SEA801-Zigbee/SEA802-Zigbee

regissg, 2021-03-23:
I have installed that device and I could set parameters like current_heating_setpoint or child_lock but I could not get any parameters. 

For example 
   zigbee2mqtt/TRV/get  {"local_temperature": ""} 
gives
   error 2021-03-22 19:58:50: No converter available for 'local_temperature' () 

Here is an extract of the setup log data
..."interviewing":false,"model_id":"TS0601","network_address":24983,"power_source":"Battery","supported":true,"type":"EndDevice"},{"date_code":"","definition":{"description":"Thermostatic radiator valve","exposes":[{"access":1,"description":"Indicates if the battery of this device is almost empty","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"features":[{"access":3,"description":"Enables/disables window detection on the device","name":"state","property":"window_detection","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"features":[{"access":3,"description":"Enables/disables physical input on the device","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"type":"lock"},{"features":[{"access":3,"description":"Temperature setpoint","name":"current_heating_setpoint","property":"current_heating_setpoint","type":"numeric","unit":"°C","value_max":30,"value_min":5,"value_step":0.5},{"access":1,"description":"Current temperature measured on the device","name":"local_temperature","property":"local_temperature","type":"numeric","unit":"°C"},{"access":3,"description":"Mode of this device","name":"system_mode","property":"system_mode","type":"enum","values":["off","heat","auto"]},{"access":3,"description":"Offset to be used in the local_temperature","name":"local_temperature_calibration","property":"local_temperature_calibration","type":"numeric","unit":"°C"},{"access":1,"description":"The current running state","name":"running_state","property":"running_state","type":"enum","values":["idle","heat"]},{"access":3,"description":"Away mode","name":"away_mode","property":"away_mode","type":"binary","value_off":"OFF","value_on":"ON"}],"type":"climate"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"SEA801-Zigbee/SEA802-Zigbee","supports_ota":false,"vendor":"Saswell"},"endpoints":{"1":{"bindings":[{"cluster":"genBasic","target":{"endpoint":1,"ieee_address":"0x00124b000be89d05","type":"endpoint"}}],"clusters":{"input":["genBasic","genGroups","genScenes","manuSpecificTuya"],"output":["genOta","genTime"]},"configured_reportings":[]}},"friendly_name":"TRV","ieee_address":"0x847127fffe1b6875","interview_completed":true,"interviewing":false,"model_id":"TS0601","network_address":3682,"power_source":"Battery","supported":true,"type":"EndDevice"}]'

strandbein, 2021-04-12:
First thank you very much for the great project.
I have the same problem. All /get commands end with:

Error No converter available for 'get' 'window_detection' ()
Error No converter available for 'get' 'child_lock' ()
Error No converter available for 'get' 'current_heating_setpoint' ()
Error No converter available for 'local_temperature' ()
Error No converter available for 'get' 'system_mode' ()
Error No converter available for 'running_state' ()
Error No converter available for 'get' 'away_mode' ()

It is possible to send a redundant /set, then all values are also reported. But it would be more elegant with a functioning /get.

strandbein, 2021-05-13:
reopen, please

Koenkk, 2021-05-13:
`/get` is not supported for TuYa based thermostats.

strandbein, 2021-05-14:
ok, then the description should be adjusted [here](https://www.zigbee2mqtt.io/devices/SEA801-Zigbee_SEA802-Zigbee.html#exposes)

Koenkk, 2021-05-15:
Done, assuming this can be closed now.

closed 2021-05-15 by Koenkk