Koenkk/zigbee2mqtt #6804: TS0215A

fl1p1, 2021-03-23:
TS0215A from manufacturer _TZ3000_fsiepnrh (in fact it is the Nedis ZBRC10WT)

`debug 2021-03-23 18:12:06: Received Zigbee message from 'testbutton', type 'commandEmergency', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
warn 2021-03-23 18:12:06: Received message from unsupported device with Zigbee model 'TS0215A' and manufacturer name '_TZ3000_fsiepnrh'
warn 2021-03-23 18:12:06: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.`

Strange, because it is in the herdsman devices.js list. But it doesn't seem to work...

Koenkk, 2021-03-23:
Switching to the dev branch should solve this.  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

fl1p1, 2021-03-24:
There is only the branch 1.16.2 
`error: pathspec 'dev' did not match any file(s) known to git`

closed 2021-05-02 by github-actions[bot]