Koenkk/zigbee2mqtt #6815: New coordinator design based on RF-STAR modules (chips cc2652p / cc1352p)

drbios, 2021-03-24:
Hi
I would like to share this excellent work from Egony who created this new design with RF-STAR cc2562p modules.
I was able to replicate his work without problems by following his instructions and I am really happy with this excellent coordinator. after several cups of coffee and bad language I was able to assemble this coordinator. works with 20db with great penetration power in concrete walls to devices on the second floor

If possible add this layout to the list of supported coordinators

The project page can be found here
https://github.com/egony/cc2652p_cc1352p_RF-STAR/wiki

[image]
[image]

if possible add this layout to the list of supported coordinators

greetings from Chile
Jed anderson

drbios, 2021-03-24:
> If English instructions are available I'm happy to accept a PR to add this to the supported adapters list.

Which information do you need?
(I use Google translate to read all the instructions)

Koenkk, 2021-03-24:
A guide on how to create the device yourself (BOM and assembly instructions) + how to flash (and with which firmware)

egony, 2021-03-24:
> > If English instructions are available I'm happy to accept a PR to add this to the supported adapters list.
Welcome.
https://github.com/egony/cc2652p_cc1352p_RF-STAR/wiki/Home-EN

egony, 2021-03-24:
Same for E72 boards:
https://github.com/egony/cc2652p_E72-2G4M20S1E/wiki/Home-EN

drbios, 2021-03-25:
> > > If English instructions are available I'm happy to accept a PR to add this to the supported adapters list.
> 
> Welcome.
> https://github.com/egony/cc2652p_cc1352p_RF-STAR/wiki/Home-EN

Thanks a lot

drbios, 2021-03-25:
@egony can you explain the auto BSL flashing procedure ? (I have mine running on a raspberry pi with raspberry os lite)
Best regards

Koenkk, 2021-03-25:
Cool, please make a PR for both https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/README.md and https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/information/supported_adapters.md

closed 2021-05-03 by github-actions[bot]

egony, 2021-05-06:
> @egony could you explain the auto BSL flashing procedure ? (I have mine running on a raspberry pi with raspberry os lite)
> Best regards

Hi! Sorry for delay. Auto BSL means that you need no pressing buttons for bootloader activation. Flashing software puts dongle into bootloader mode automatically. But it works with not all flashing software.

egony, 2021-05-06:
> Cool, please make a PR for both https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/README.md and https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/information/supported_adapters.md

Hi!

https://github.com/Koenkk/zigbee2mqtt.io/pull/735
https://github.com/Koenkk/Z-Stack-firmware/pull/288

Done.