Koenkk/zigbee2mqtt #6823: Integration with Hubitat

BarryOCathain, 2021-03-24:
## Is your feature request related to a problem? Please describe
Hey guys, I have a hubitat and hive, with multiple tuya TRVs.

I'm writing an app that is able to coordinate the multiple different temperature sensors in a heating zone to enable me to set up routines.

Ideally an mqtt solution would allow me to drive everything off events.

## Describe the solution you'd like
I see lots of docs around adding new device drivers, but I see nothing about integrating with a new zigbee adapter.

Given what I've read, I should be able to use the hubitat as a zigbee adapter in essence, and pass the messages on to mqtt2zigbee, for it to parse and publish them.

I'm happy to do the leg work here (I'm a dev by trade) but I'm looking for info on how the current implementation interfaces with supported zigbee adapters, so that I can understand where the hubitat might fit into the equation.

If it's simply a case of understanding how zigbee2mqtt parses the Hive zigbee data so that I can write a native hubitat driver, I'm happy to take a couple of pointers on how the current implemetation does that.

## Additional context
I have 2 x Hive SLT3 thermostats, and 1 x SLR1b and 1 x SLR2 receivers, so gathering zigbee data shouldn't be an issue if that's required.

Once again I'm happy to do dev work here, but I'd appreciate any pointers you can give.

Cheers,

Barry.

Koenkk, 2021-03-25:
>  but I see nothing about integrating with a new zigbee adapter.

Do you mean using the hubitat hardware as the adapter, I don't know about hubitat but is it possible to install custom software on it?

BarryOCathain, 2021-03-25:
@Koenkk yes I plan to use the hubitat as an adapter, however I don't believe you can install custom software on it.

I had hoped to find a point in the code where the messages are read from the adapter, and hook the hubitat in a that point.

The hubitat can relay messages through an mqtt broker, so it seems like there could be a reasonable fit with some alterations.

Koenkk, 2021-03-25:
@BarryOCathain on what level are these messages? They need to be pretty low-level to integrate into zigbee-herdsman (the underlying zigbee library). Basically the following  interface has to be implemented: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/adapter.ts

Examples of 2 adapters that have been added recently:
- https://github.com/Koenkk/zigbee-herdsman/pull/241
- https://github.com/Koenkk/zigbee-herdsman/pull/317

closed 2021-05-03 by github-actions[bot]