Koenkk/zigbee2mqtt #6824: UNDEFINED model and UNDEFINED manufacturerName

Faustino-ai, 2021-03-25:
I've tried to add support following the guide. I've only reached the point where I put in my devices.js information. However, both manufacturerName as well as modelID showed up as undefined. Based on the database.db, I've included Lockvel as the manufacturer with the manufID of 4098 but modelID still showed up as 'undefined'. I have no idea how to add the information for this part.

## Information about the device + link
.. It is a smart door lock and the model is F518. 
https://lockvel.en.made-in-china.com/product/OFWmLCuMfAhD/China-Smart-Door-Lock-Fingerprint-Door-Lock-Smart-Lock-for-Home-Hotel-Apartment-Office.html

## data/database.db entry of the device
.. {"id":2,"type":"EndDevice","ieeeAddr":"0x60a423fffe828430","nwkAddr":21186,"manufId":4098,"epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2064,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1616634050036}

Koenkk, 2021-03-25:
The following config should do it:

```js
{
    fingerprint: [{type: 'EndDevice', manufacturerID: 4098, endpoints: [
        {ID: 1, profileID: 260, deviceID: 2064, inputClusters: [4096], outputClusters: [4096]},
    ]}],
    model: 'F518',
    description: 'Smart door lock',
    vendor: '??',
    fromZigbee: [],
    toZigbee: [],
    exposes: [],
},
```

Faustino-ai, 2021-03-26:
Thank you! That solves the issue.

closed 2021-03-26 by Faustino-ai

Vebryn, 2021-05-22:
where to put this config ? into a new js file ?

Vebryn, 2021-05-22:
I just found that my device is supported in zigbee-herdsman-converters master branch.

```
Zigbee2MQTT:info  2021-05-22 14:24:16: Starting Zigbee2MQTT version 1.18.3 (commit #f2e39af)
Zigbee2MQTT:info  2021-05-22 14:24:16: Starting zigbee-herdsman (0.13.92)
```

How to get herdsman from master branch ?