Koenkk/zigbee2mqtt #6826: SEA802 Lock state report

bublikOff, 2021-03-25:
There is something wrong with SEA802 lock state report.
I can lock or unlock from HA and I can see that state change in HA and in z2m "State" tab.
In z2m "Expose" tab is always showing "Unlocked" in any actual state locked or unlocked.
There is also problem with lock state reporting when I do lock or unlock thermostat using buton on it (not from HA GUI)

PS
z2m version is 1.18.1-2

salopette, 2021-03-25:
so with me the satus is transmitted

bublikOff, 2021-03-25:
im not sure wghat is actually wrog ... as for eg on my test thermostat child_lock shows "locked" but if I press button on thermostat it shows themperature instead of LC

bublikOff, 2021-03-25:
https://www.youtube.com/watch?v=LOR0fug2SWw

closed 2021-05-03 by github-actions[bot]