Koenkk/zigbee2mqtt #6827: Degraded network performance on 1.18.1

balonchiks, 2021-03-25:
## What happened
After upgrading from 1.17.0 to 1.18.1 I am experiencing a degrading network performance. Some light do not turn on at all, some turn on with a very noticeable delay (up to 10 seconds), xiaomi motion sensors might skip movement detections or report them with a noticeable delay (same, up to 10 seconds). 

Have to mention that some devices on the network, and even the devices that had issues 10 minutes ago - might just (what looks like) accidentally start working for some time.

Downgrading back to 1.17.0 seem to solve the issue, at least there's no noticeable lag and the network works as expected.

## What did you expect to happen
Expect everything to work of course

## How to reproduce it (minimal and precise)
update to 1.18.1 and try switching lights on/off or trigger motion 

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC2538_CC2592_MODKAMRU_V3
Adapter firmware version: [20201010](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.0.x/bin/CC2538_CC2592_MODKAMRU_V3.md)

There is nothing suspicious in the general logs, so here are herdsman logs:

- switching on (0x588e81fffecf514b) [HG06104A](https://www.zigbee2mqtt.io/devices/HG06104A.html) led strip : https://pastebin.com/EZbmKZSn

from what i can see here, these are suspicious lines, is that something that causes the behavior:

```
  zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x588e81fffecf514b:47855,205,1) +615ms
  zigbee-herdsman:adapter:zStack:adapter Discovering route to 47855 +1ms
```

- switching on another (0x588e81fffed88e4c) [HG06104A](https://www.zigbee2mqtt.io/devices/HG06104A.html) led strip: https://pastebin.com/UBNhmr6M

In here it looks like that the request timeouts?:

```
  zigbee-herdsman:adapter:zStack:adapter Response timeout (0x588e81fffed88e4c:3119,0) +10s
  zigbee-herdsman:adapter:zStack:adapter Discovering route to 3119 +1ms
```

Is there anything that has been changed in the 1.18.x version that might cause this? or that's an issue somewhere in my network? 
Thanks.

salopette, 2021-03-25:
https://github.com/Koenkk/zigbee2mqtt/issues/6684

balonchiks, 2021-03-25:
thanks, i have tried to completely remove zigbee2mqtt (as you suggested in #6684) and do a fresh install - but the issue is still here..

balonchiks, 2021-04-04:
after numerous re-flashes and reinstalls everything seems to be fine now. thanks.

closed 2021-04-04 by balonchiks