Koenkk/zigbee2mqtt #6831: Danalock V3: support for opening lock including spring

kalle2000, 2021-03-25:
I am running Danalock with firmware 0.17.5 (zigbee2mqqtt 1.18.1) and in the App there is a setting to open the lock including the spring (for doors without a handle on the outside). In the Danalock App this is triggered by pushing the "Lock" Icon while the lock is locked.

Unfortunately I have not found out which mqtt command triggers this.
I have the following debug output:

1. locking the lock (supported)

```
Zigbee2MQTT:debug 2021-03-25 21:28:49: Received Zigbee message from 'danalock_door', type 'commandOperationEventNotification', cluster 'closuresDoorLock', data '{"data":0,"opereventcode":1,"opereventsrc":1,"pin":0,"userid":65535,"zigbeelocaltime":4294967295}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-25 21:28:49: Received Zigbee message from 'danalock_door', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-03-25 21:28:49: MQTT publish: topic 'zigbee2mqtt/danalock_door', payload '{"action":"lock","action_source":1,"action_source_name":"rf","action_user":65535,"battery":91,"linkquality":63,"lock_state":"unlocked","state":"UNLOCK"}'
Zigbee2MQTT:info  2021-03-25 21:28:49: MQTT publish: topic 'zigbee2mqtt/danalock_door', payload '{"battery":91,"linkquality":63,"lock_state":"locked","state":"LOCK"}'
```

2. unlocking the lock (supported)

```
Zigbee2MQTT:debug 2021-03-25 21:27:52: Received Zigbee message from 'danalock_door', type 'commandOperationEventNotification', cluster 'closuresDoorLock', data '{"data":0,"opereventcode":2,"opereventsrc":1,"pin":0,"userid":65535,"zigbeelocaltime":4294967295}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-25 21:27:52: Received Zigbee message from 'danalock_door', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-03-25 21:27:52: MQTT publish: topic 'zigbee2mqtt/danalock_door', payload '{"action":"unlock","action_source":1,"action_source_name":"rf","action_user":65535,"battery":91,"linkquality":63,"lock_state":"locked","state":"LOCK"}'
Zigbee2MQTT:info  2021-03-25 21:27:52: MQTT publish: topic 'zigbee2mqtt/danalock_door', payload '{"battery":91,"linkquality":63,"lock_state":"unlocked","state":"UNLOCK"}'
```

3. unlocking the lock incl. spring (unknown)
```

Zigbee2MQTT:debug 2021-03-25 21:29:25: Received Zigbee message from 'danalock_door', type 'commandOperationEventNotification', cluster 'closuresDoorLock', data '{"data":0,"opereventcode":2,"opereventsrc":1,"pin":0,"userid":65535,"zigbeelocaltime":4294967295}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-25 21:29:25: Received Zigbee message from 'danalock_door', type 'attributeReport', cluster 'closuresDoorLock', data '{"lockState":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-03-25 21:29:25: MQTT publish: topic 'zigbee2mqtt/danalock_door', payload '{"action":"unlock","action_source":1,"action_source_name":"rf","action_user":65535,"battery":91,"linkquality":63,"lock_state":"locked","state":"LOCK"}'
Zigbee2MQTT:info  2021-03-25 21:29:25: MQTT publish: topic 'zigbee2mqtt/danalock_door', payload '{"battery":91,"linkquality":63,"lock_state":"unlocked","state":"UNLOCK"}'```

I cant see any difference in the debug.log between "unlock" and "unlock with spring"
Any hints how to track this down?

closed 2021-05-03 by github-actions[bot]

martimarkov, 2022-01-12:
@kalle2000 Have you found a solution to this?

kalle2000, 2022-01-14:
unfortunately not as I dont know how to capture the zigbee commands from the Danalock app