Koenkk/zigbee2mqtt #6834: after you upgraded to a new version everything stopped here at home

vpereiradn, 2021-03-26:
after you upgraded to a new version everything stopped here at home i have 22 lamps here none lights up anymore, it changed from the share folder to the config folder ..
what is the solution

m0wlheld, 2021-03-28:
> what is the solution

What is the actual problem? What are „config folder“ and „share folder“?

vpereiradn, 2021-03-29:
Hi! Good Morning.
Thanks more I already decided, I had to uninstall everything and pair all devices, and it worked.
it is more complicated because every time you update you have this kind of problem, thank you very much!

m0wlheld, 2021-03-29:
> it is more complicated because every time you update you have this kind of problem

That is not correct. Yet, you haven't provided any details on what you did and what errors it caused.

vpereiradn, 2021-04-03:
[state.json.zip](https://github.com/Koenkk/zigbee2mqtt/files/6253235/state.json.zip)

i lost all my devices after upgrading to 1.18.2

vpereiradn, 2021-04-03:
I left the status to see what the error

vpereiradn, 2021-05-04:
deu certo aqui obrigado. desinstalei e instalei novamente e refiz todo pareamento dos dispositivos.

closed 2021-06-12 by github-actions[bot]