Koenkk/zigbee2mqtt #6835: Need help to finish implementing support for the product - smart lock

Faustino-ai, 2021-03-26:
I've done almost everything according to the guide how to support new devices. Since my door lock is most probably using tuya zigbee protocol, I followed the "How to support new Tuya devices" as well. Currently I've successfully sniff the traffic and one of my decrypted payload look something like this: 
55 AA 03 00 05 04 00 0C 67 00 00 08 36 36 36 36 38 38 38 38 3E
where
55 AA - Header 
03 - version
00 05 - sequence
04 - command
00 0c - length of the following (excluding checksum)
67 - dpID  
00 - dp type (raw) 
00 08 - dp length
36 36 36 36 38 38 38 38 - password
3E - checksum

This is the payload that I obtained when I used the remote unlock with password function. Currently my password is being set as 66668888. I will implement the rest of the functions once I know how to do the rest. I followed the guide and reached step number 4. Currently I've published to MQTT topic--> zigbee2mqtt/F518/set/tuya_data_point_test (F518 is my device friendly name) with the payload 
"raw, 67, 06, 36, 36, 36, 38, 38, 38, 38"      and many different types of combination for the password section.
According to the guide, I am supposed to get some data in data/tuya.dump.txt but I couldn't find any txt file in my data folder. I double checked that I did include the two converters, fz.tuya_data_point_dump as well as tz.tuya_data_point_test. I tried to unlock my door lock using fingerprint, which should've sent data to fz.tuya_data_point_dump but I cant see it. Please help me on how to proceed from here. 

**I did try to add in the dpID 67 in cluster.js however it doesn't seem like it works. I tried to follow some of the codes but since this is tuya its different, and so far there are no tuya smart lock for me to based on. Please help me on this. Thank you

## Information about the device + link
..It is a smart door lock and the model is F518.
https://lockvel.en.made-in-china.com/product/OFWmLCuMfAhD/China-Smart-Door-Lock-Fingerprint-Door-Lock-Smart-Lock-for-Home-Hotel-Apartment-Office.html

## data/database.db entry of the device
..
{"id":4,"type":"EndDevice","ieeeAddr":"0x60a423fffe828430","nwkAddr":6996,"manufId":4098,"epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2064,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d992da","endpointID":1}],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1616730636781}

Koenkk, 2021-03-26:
I'm not a TuYa expert, but something that can possibly help is to sniff the same command send with both with your current Z2M implementation and the original gateway. In this way you can compare the difference. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Faustino-ai, 2021-03-27:
> I'm not a TuYa expert, but something that can possibly help is to sniff the same command send with both with your current Z2M implementation and the original gateway. In this way you can compare the difference. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Thanks for your reply. Currently I only have 1 CC2531 so it's not possible for me to sniff my Z2M. However I already tried to order another stick and hopefully it arrives soon. One of the things that I am still not sure is that in the cluster.js file, what is the ID? https://pastebin.com/ZAJRdJGw
I set mine as 55aa without knowing if it's the correct way.

Faustino-ai, 2021-03-29:
One thing I've realized that my payload is not exactly the same structure as the one shown in the guide. Apparently in the guide, the structure looks something {status, transid, dp, datatype, fn, data} where as mine looks something like {header with version, transid, datatype, dp, fn, data}. Note that the dp and datatype sequence in reversed but I am not sure if it has something to do with Big Endian or what not. Also I am not sure if the header with version is the same as 'status' but from the debug log, my status always show 0, which shouldn't be the case.

Koenkk, 2021-03-29:
Regarding the cluster, this is the cluster ID which you can sniff when controlling it with the original gateway.

Faustino-ai, 2021-03-30:
After cross referencing with manuSpecificTuya in cluster.js, I didn't manage to find anything related to 55 AA in the code, instead I found 0xEF00 as the ID for manuSpecificTuya. Does this mean that I have to create a new cluster for my device?

Koenkk, 2021-03-30:
It depends on what cluster is used to control this device (which can be obtained by sniffing it when controlling with the original gateway)

Faustino-ai, 2021-03-31:
Apparently it's a manufacturer specific cluster not listed in z2m for the moment. Am really lost with this much complications.

Faustino-ai, 2021-03-31:
I've also checked with another user since I do not have 2 usb sticks to sniff my own z2m instances, he provided me with the sniffed data from his z2m instance using the same way as I did (using mqtt to publish {"tuya_data_point_test": "raw, 21, 31, 32, 33, 37, 38, 39"}). The data obtained was something like 00 03 15 00 00 06 1f 20 21 25 26 27. After arranging this according to the format as shown in  /lib/tuya.js  , I found out that the z2m output is missing the first 3 bytes - the 55 AA (header) and 03 (version). It is also missing the 04 (command) which is supposed to appear before the sequence bytes. Apart from this, the original sniffed data also includes the length of the data excluding the first 8 bytes as well as checksum, which should come after the "command" (04) byte. Finally, the output from z2m doesn't seem to produce a checksum at the end of the data.

Here is a more direct comparison between the z2m and original zigbee traffic:
Original traffic (on top) vs Zigbee2mqtt traffic (at the bottom)
55 AA 03 00 05 04 00 0C 67 00 00 08 36 36 36 36 38 38 38 38 3E
                00 03                15 00 00 06 1f 20 21 25 26 27
Is there anything I can do to fix this?

Faustino-ai, 2021-04-05:
I've realized the one who implemented Tuya in zigbee2mqtt used a weird implementation for length. For Tuya zigbee structure, it uses 2 bytes for 'length' but in 'tuya_data_point_test', instead of using two bytes, 1 byte with 2 separate variables are used. For instance, a variable with for example the name 'dpLength' is enough and using ...default.uint16, but what happens in zigbee2mqtt is that two variables with the name 'lengh_hi' and 'length_lo' are used to indicated the lower bytes and higher bytes. May I know why is that so? Also since my door lock uses a total of 6 bytes extra in total, I have no idea how to include them as of now. All the changes I made to the codes weren't working.

drzony, 2021-04-05:
@Faustino-ai 
1. Length is implemented as hi/lo due to endianness. JavaScript/Typescript are weird when manipulating data buffers, so that's why it's implemented that way
2. You are confusing two protocols ZigBee ZCL and Tuya protocol. When receiving "raw" cluster you are getting full ZCL message (do not confuse this with "raw" data type in Tuya protocol), so it will contain ZCL command ID (which is 00 when sending through "tuya_data_point_test"), and other ZCL data
3. The above means that when sniffing you will get 55 AA 03 00 05 04 00 0C 67 00 00 08 36 36 36 36 38 38 38 38 3E, which seems like a ZCL message, so 04 is "sendData" command (see point 5) instead of old 00 "setData" command. Tuya protocol (as seen in fromZigbee) seems to be: 67 00 00 08 36 36 36 36 38 38 38 38. I don't have any of the "new devices", so I'm not able to verify if "sendData" is OK, you seem to be missing `status` and `transid` before 67, see `src\zcl\definition\cluster.ts` in zigbee-herdsman
4. Try latest dev, since I added new receiving command ID for newer devices
5. You can try changing tuya_data_point_test converter by adding "sendData" as a last parameter to send functions. There seems to be a new command ID in newer Tuya devices, so change ` tuya.sendDataPointRaw(entity, dp, data);` to ` tuya.sendDataPointRaw(entity, dp, data, 'sendData');`

Faustino-ai, 2021-04-06:
> @Faustino-ai
>
> 1. Length is implemented as hi/lo due to endianness. JavaScript/Typescript are weird when manipulating data buffers, so that's why it's implemented that way
> 2. You are confusing two [...]

Thank you very much! I will get on to it with the updated dev version.

Faustino-ai, 2021-04-06:
> 3. The above means that when sniffing you will get 55 AA 03 00 05 04 00 0C 67 00 00 08 36 36 36 36 38 38 38 38 3E, which seems like a ZCL message, so 04 is "sendData" command (see point 5) instead [...]

[image]
Actually they gave 2 data and I chose the one with the longest, thinking that's the complete version. Either way, It seems like the other set starts from 04 00 0C 67 00 00 08 36 36 36 36 38 38 38 38 3E. I have no idea what ### status is but transid should be the sequence, in which it is missing because it is just 1 byte before the 04.

closed 2021-05-14 by github-actions[bot]

flatsiedatsie, 2022-01-03:
@Faustino-ai Did you get this lock to work?

FaustinoSW, 2022-01-04:
> @Faustino-ai Did you get this lock to work?

Nope. This one is a dead end for me. It just doesn't seem like a proper zigbee lock as advertised

Faustino-ai, 2022-01-04:
> 

Nope. This one is a dead end for me. It just doesn't seem like a proper zigbee lock as advertised

Faustino-ai, 2022-01-04:
> @Faustino-ai Did you get this lock to work?

Nope. This one is a dead end for me. It just doesn't seem like a proper zigbee lock as advertised

Faustino-ai, 2022-06-17:
> Hi Gents,
> 
> Can someone give me a TLDR _ I tried to understand the situation, I have tested the TuYa Smart Lock via their Dev Portal. So I have the Gatewaz and via the Dev Portal I can open and close the device. Does this mean that the gateway uses BT for opening the Device and Zigbee only to get the status updates ?
> 
> Am I understanding correctly ? Hence the reason it cannot be properly integrated ?
> 
> Sorry for the questions.
> 
> Cheers,
> 
> Gyula

I think you've stumbled on the wrong thread. Either way the TLDR is that Tuya Zigbee lock ( Zigbee only, not bluetooth, if you have both bluetooth and zigbee then we are probably not referring to the same device ) is currently not possible to be integrated into zigbee2mqtt due to its encryption keys require access to tuya cloud. Sure, you can try to use their openapi to integrate it but it will still require access to their cloud. I wouldn't bother with their openapi though because I don't see the point of using our own zigbee stick but still requiring tuya cloud to obtain some sort of keys to make the lock work. If I want to use their cloud I would just stick with their default gateway instead