Koenkk/zigbee2mqtt #6836: GLEDOPTO GL-C-007P RGBW LED controller - White color not working

garywr72, 2021-03-26:
Hi,

I bought a  GLEDOPTO GL-C-007P RGBW LED controller. 

Device supported but without white color. RGB control works fine.

Perhaps problem can be fixed by adding colortemp property in devices.js (similar to the issue [https://github.com/Koenkk/zigbee2mqtt/issues/4424](https://github.com/Koenkk/zigbee2mqtt/issues/4424))

zigbeeModel GL-C-007P
Original > "extend: preset.gledopto.light_onoff_brightness_color(),"
Replaced by > "extend: preset.gledopto.light_onoff_brightness_colortemp_color(),"

Right now controller works in RGB mode only.

Koenkk, 2021-03-26:
Did you test if it works?

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to edit `devices.js`.

garywr72, 2021-03-28:
Finally I added as a new device in hassio-zigbee2mqtt - I'm a beginner yet :) 
It works, but I get error: No converter available for 'color_temp' and No converter available for 'color_temp_startup'
Probably not fully configured yet.
However, I can change colors and turn on white.

Koenkk, 2021-03-28:
How do you change the color temp to white now?

garywr72, 2021-03-28:
I can change the color temp by clicking on color_temp in menu Exposes - similar like in another controller GL-C-008P fully supported in zigbee2mqtt.

Koenkk, 2021-03-30:
Cool! Added the fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-03-30 by Koenkk