Koenkk/zigbee2mqtt #6837: Sometimes Lqi problem after pairing a new device ( CC2652R zzh stick)

kitopopo, 2021-03-26:
## What happened

Sometimes after pairing a new device appear the following error lines:

```
mar 26 10:55:28 raspberrypi npm[6475]: Zigbee2MQTT:info  2021-03-26 10:55:28: Starting network scan (includeRoutes 'true')
mar 26 10:55:40 raspberrypi npm[6475]: Zigbee2MQTT:error 2021-03-26 10:55:40: Failed to execute LQI for 'Denon plug'
mar 26 10:55:51 raspberrypi npm[6475]: Zigbee2MQTT:error 2021-03-26 10:55:51: Failed to execute routing table for 'Denon plug'
mar 26 10:55:52 raspberrypi npm[6475]: Zigbee2MQTT:info  2021-03-26 10:55:52: Network scan finished

```

the problem is solved unplug and plug again the lqi device. In my case 'Denon Plug' Im my network de Denon Plug is a plug but also a repeater plug. I habe two plug repeaters, some times the other repeater plug also get the same lqi error.

## What did you expect to happen

Pairing all devices withouth lqi errors in the network. it is cumbersome to unplug the repeaters every time we pair a new device

## How to reproduce it (minimal and precise)

Sometimes pairing a new device. Sometimes works fine

## Debug info
Zigbee2MQTT version: "version":"1.18.1-dev"
Adapter hardware: CC2652R zzh
Adapter firmware version: revision":20210120

kwetnico, 2021-03-27:
Hello

I have the same problem with Heiman HS2WD-E Zigbee.

Krazy998, 2021-03-31:
Also seeing the same thing on same adapter and versions.

[image]

Koenkk, 2021-03-31:
The LQI feature is something that has to be supported by the device, not all devices may do this (I believe e.g. gledopto devices don't)

kitopopo, 2021-03-31:
Hi, 

Hi don't have gledopto devices. All my switch are xiaomi devices( xiaomi switchs, water sensor, smart plugs, door sensor, vibration sensor, smoke detector, motion detector...) and philips hue bulbs. 
But sometimes when i pair a new xiaomi device i get lqi error, i need unpug and plug my two zigbee repeaters xiaomi smart plug for solve the problem .
the lqi error is an error that has been creeping since my beginnings with the cc2531, it also happens with the new cc2652r. hope it can be fixed as it is very annoying to unplug my repeaters every time I add a device. Thanks in advanced

Koenkk, 2021-04-01:
Could you provide the debug log of this?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

kitopopo, 2021-04-01:
> Could you provide the debug log of this?
> 
> See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

Ok  i will test. But in order to see the log file , how i get the file? In the information appears:

Important: this is not logged to the log files and is only available on the STDOUT.

What o where is STDOUT?

Thanks

Koenkk, 2021-04-01:
Only the herdsman debug logging is logged to stdout. I don't need the herdsman debug logging, just the normal one which is logged to the log files.

kitopopo, 2021-04-04:
Dear friend,

I have enabled the log_level: debug line in configuration.yaml below advanced. I have added 2 devices today but now both lqi are OK. As the failure does not always occur, I will try every day until the lqi errors occur, then I will send you the log. Thank you.
thanks again for your excellent support. a cordial greeting

kitopopo, 2021-04-09:
Dear @Koenkk ,

Due to enrolling 2 new devices "0x00158d0006a28b66" and"0x00158d0006b635c9", they appear without a line drawn in the network map I removed the devices and again added them back to the network but the error Failed to execute LQI for 'Router plug' is shown several times. after retrying this several times I get satisfactory lqi. 
 
`debug 2021-04-08 19:30:39: LQI succeeded for 'Router plug'`

I think that when the lqi error appears the device appears without lines drawn on the network map.

I leave below the record that I request. thanks in advance

```
 node[shape=record];
  "0x00124b002281a72c" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b002281a72c (0)|0 seconds ago}"];
  "0x00124b002281a72c" -> "0x00158d0002675428" [penwidth=0.5, weight=0, color="#994444", label="98"]
  "0x00124b002281a72c" -> "0x00158d00039bba90" [penwidth=0.5, weight=0, color="#994444", label="94"]
  "0x00124b002281a72c" -> "0x00124b00014eef9b" [penwidth=0.5, weight=0, color="#994444", label="70"]
  "0x00158d000273b8c6" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Sensor salon izquierdo|0x00158d000273b8c6 (12185)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|7 minutes, 7 seconds ago}"];
  "0x00158d000273b8c6" -> "0x00124b002281a72c" [penwidth=1, weight=0, color="#994444", label="87"]
  "0x00158d000273b8c6" -> "0x00124b002281a72c" [penwidth=1, weight=0, color="#994444", label="87"]
  "0x00158d000273597c" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Sensor salon derecho|0x00158d000273597c (9426)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|30 minutes, 23 seconds ago}"];
  "0x00158d000273597c" -> "0x00124b002281a72c" [penwidth=1, weight=0, color="#994444", label="100"]
  "0x00158d000273597c" -> "0x00124b002281a72c" [penwidth=1, weight=0, color="#994444", label="100"]
  "0x00158d0002bfd116" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Sensor habitacion derecho|0x00158d0002bfd116 (17585)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|17 minutes, 9 seconds ago}"];
  "0x00158d0002bfd116" -> "0x00124b002281a72c" [penwidth=1, weight=0, color="#994444", label="78"]
  "0x00158d0002753346" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Sensor habitacion izquierdo|0x00158d0002753346 (53948)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|17 minutes, 34 seconds ago}"];
  "0x00158d0002753346" -> "0x00124b002281a72c" [penwidth=1, weight=0, color="#994444", label="77"]
  "0x00158d0002675428" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Denon plug|0x00158d0002675428 (51818)|Xiaomi Mi power plug ZigBee (ZNCZ02LM)|33 seconds ago}"];
  "0x00158d0002675428" -> "0x00124b002281a72c" [penwidth=0.5, weight=0, color="#994444", label="101"]
  "0x00158d0002675428" -> "0x00158d00039bba90" [penwidth=0.5, weight=0, color="#994444", label="46"]
  "0x00158d0002675428" -> "0x00124b00014eef9b" [penwidth=2, weight=1, color="#009900", label="49 (routes: 51818)"]
  "0x00158d00039bba90" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Router plug|0x00158d00039bba90 (58890)|Xiaomi Mi power plug ZigBee (ZNCZ02LM)|10 seconds ago}"];
  "0x00158d00039bba90" -> "0x00158d0002675428" [penwidth=0.5, weight=0, color="#994444", label="44"]
  "0x00158d00039bba90" -> "0x00124b00014eef9b" [penwidth=0.5, weight=0, color="#994444", label="27"]
  "0x00158d000273da5e" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Sensor puerta entrada|0x00158d000273da5e (61716)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|34 minutes, 26 seconds ago}"];
  "0x00158d000273da5e" -> "0x00158d00039bba90" [penwidth=1, weight=0, color="#994444", label="75"]
  "0x00158d00027532c0" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Sensor lavadero|0x00158d00027532c0 (4072)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|10 minutes, 20 seconds ago}"];
  "0x00158d00027532c0" -> "0x00124b002281a72c" [penwidth=1, weight=0, color="#994444", label="79"]
  "0x00158d0004141d1a" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Sensor movimiento|0x00158d0004141d1a (2715)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|2 minutes, 22 seconds ago}"];
  "0x00158d0004141d1a" -> "0x00158d00039bba90" [penwidth=1, weight=0, color="#994444", label="96"]
  "0x00158d0002f92828" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Sai temperature|0x00158d0002f92828 (30474)|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|28 minutes, 19 seconds ago}"];
  "0x00158d0002f92828" -> "0x00158d0002675428" [penwidth=1, weight=0, color="#994444", label="124"]
  "0x00158d00030a6c7e" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Woofer plug|0x00158d00030a6c7e (61801)|Xiaomi Mi power plug ZigBee (ZNCZ02LM)|12 seconds ago}"];
  "0x00158d00030a6c7e" -> "0x00124b002281a72c" [penwidth=1, weight=1, color="#009900", label="104 (routes: 61801,47197)"]
  "0x00158d00030a6c7e" -> "0x00158d0002675428" [penwidth=1, weight=1, color="#009900", label="157 (routes: 61801)"]
  "0x00158d00030a6c7e" -> "0x00158d00039bba90" [penwidth=1, weight=0, color="#994444", label="76"]
[... cut ...]
  "0x00158d0006b635c9" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x00158d0006b635c9|0x00158d0006b635c9 (49993)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|37 seconds ago}"];
}'
debug 2021-04-08 19:30:10: Received Zigbee message from 'Parking plug', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2021-04-08 19:30:18: Received Zigbee message from 'CC2531 Router', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
info  2021-04-08 19:30:18: MQTT publish: topic 'zigbee2mqtt/CC2531 Router', payload '{"led":false,"linkqualit

Geoff571, 2021-04-24:
> The LQI feature is something that has to be supported by the device, not all devices may do this (I believe e.g. gledopto devices don't)

Confirmed with a vengeance :) The only time I see this issue is with GLEDOPTO devices and I have many.

kitopopo, 2021-05-25:
Any improvement? Thanks in advances

closed 2021-07-04 by github-actions[bot]