Koenkk/zigbee2mqtt #6840: Zigbee2mqtt won't start anymore - error: devices should be object

jpslootmans, 2021-03-26:
Followed the guide and installed zigbee2mqtt on Raspberry Pi 4B with flashed cc2531 USB stick, running Domoticz.
Everything worked fine for 24 hours, but after a reboot of the pi, no more data was received by Domoticz.
Tried to start manually, but after "npm start", I get this message, which I do not understand:
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
            READ THIS CAREFULLY

Refusing to start because configuration is not valid, found the following errors:
- devices should be object

If you don't know how to solve this, read https://www.zigbee2mqtt.io/information/configuration.html

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
This is my configuration.yaml:
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
  # user: my_user
  # password: my_password
serial:
  port: /dev/ttyUSB.CC2531-01
advanced:
  network_key: GENERATE
devices:

Please help,
JP

Koenkk, 2021-03-26:
Can you share your complete configuration.yaml?

jpslootmans, 2021-03-27:
Hello Koen,
this the complete configuration.yaml.
After the first succesful installation, there was a device 0x5c0272fffecbe7a2 with the same friendly name, but because the error said something about the configuration being the problem, I replaced the file with the same one that I used before (https://thesmarthomejourney.com/2020/05/11/guide-zigbee2mqtt/).

jpslootmans, 2021-03-27:
Here is the complete log file. Maybe this helps.

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.11
3 info using node@v14.16.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.18.1~prestart: zigbee2mqtt@1.18.1
6 info lifecycle zigbee2mqtt@1.18.1~start: zigbee2mqtt@1.18.1
7 verbose lifecycle zigbee2mqtt@1.18.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.18.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle zigbee2mqtt@1.18.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.18.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.18.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.18.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.18.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:315:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:315:20)
13 verbose stack     at maybeClose (internal/child_process.js:1048:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
14 verbose pkgid zigbee2mqtt@1.18.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 5.10.17-v7l+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v14.16.0
19 verbose npm  v6.14.11
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.18.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.18.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

m0wlheld, 2021-03-27:
Change last line (devices:) to:

`devices: devices.yaml `

and append a blank line to the end. 

Then restart.

jpslootmans, 2021-03-27:
hello Christoph,

that did the trick!
First I had an error that the node.js version did not match and the suggestion was to rebuild npm, which I did.
Then I received an error that the CC2531 could not be found. Indeed the green LED was not on, so I plugged it in another USB port and changed the port to /dev/ttyACM1 in configuration.yaml.
Now everything is working as before.
Thanks to the both of you.
JP

closed 2021-03-27 by Koenkk