Koenkk/zigbee2mqtt #6845: Not connected to MQTT server

Isilorn, 2021-03-26:
Hello,

I updated my container with the latest version last night (in fact it was done automatically via watchtower) and since then it doesn't seem to want to connect to the mosquitto MQTT broker.

Log : https://pastebin.com/iADpXKxk

Configuration : https://pastebin.com/mjbAfGAP

I can connect to the Mosquitto broker without any problem with MQTT Explorer and the mosquitto log is not showing any error

Isilorn, 2021-03-26:
And the log in debug mode : https://pastebin.com/YcT4fagj

Koenkk, 2021-03-26:
Probably the same as https://github.com/Koenkk/zigbee2mqtt/issues/6314#issuecomment-803546543 (issue in mosquitto v2)

Isilorn, 2021-03-26:
Working fine from MQTT Explorer (which is connecting from my laptop) and from other containers trying to connect to the broker (Jeedom, ZWaveJS2MQTT).

And I've got the listener directive into my Mosquitto config file :

```
root@phoenix:/home/docker/Mosquitto/config# grep listener mosquitto.conf | grep -v \#
listener 1883
```

I checked, it happened after Mosquitto update that happened a few hours ago.

Isilorn, 2021-03-26:
And even with Mosquitto in debug, here are the only lines I get when Zigbee2Mqtt connects (the other clients are much more verbose though) :

```
1616783898: New connection from 192.168.42.7:54184 on port 1883.
1616783898: Client <unknown> closed its connection.
``

Koenkk, 2021-03-26:
@Isilorn Zigbee2MQTT uses the following MQTT library: https://github.com/mqttjs/MQTT.js , probably the bug is down there. 

Can you try the minimal example from: https://github.com/mqttjs/MQTT.js#example and try to run it, does it connect correctly?

Isilorn, 2021-03-26:
I'm really not familiar with node.js and especially in a containerized environment 😢 

Any hint on how to do it (beside docker run -it koenkk/zigbee2mqtt:latest /bin/sh, this one I got it...) ?

Isilorn, 2021-03-27:
Now it works and the only difference is that I unplugged the 2nd Zwave key I have on the machine (with which I am trying to progressively migrate to Zwavejs2Mqtt). Very strange

closed 2021-05-05 by github-actions[bot]