Koenkk/zigbee2mqtt #6847: Moes light switch gets mis-recognized

Zorinik, 2021-03-26:
Good morning,
I have zigbee2mqtt addon for Hassio, version 1.18.1.
I use a CC2538 chip connected via USB.

I just bought a Moes 2 gangs light switch, white-label of Tuya, this one:
https://www.amazon.it/gp/product/B08LBBYH1V/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

When I did run the auto-join, it failed a few times and eventually it got recognized as Tuya TS0012, which I'm not sure if it is the right model honestly, but I can't control it (it can't get or set the state), and it's seen as running on battery (it has neutral cable so it should not, right?).
You can find screenshots attached.
So my guess is that it's detecting the wrong model... is it perhaps an unsupported model? Or maybe is it defective?

Thank you

[image]
[image]

Zorinik, 2021-03-26:
Quick update: after some hours it started reading the values, both the linkquality and the two states, with good accuracy. But it is still unable to send any state change (which is the most important thing to me of course).
This is the error message in the logs page:

> Error Publish 'set' 'state' to 'Luce camera' failed: 'Error: Command 0xbc33acfffed85122/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'Timeout' (9999))'

And it still detects it as running on a battery.

[image]

Zorinik, 2021-03-27:
Ok new update (sorry, but it keeps doing things on its own 😂), now I can even set the state, after like 12 hours, I did nothing and it fixed itself on its own, don't know how or why 😅

Now the only problem left is that it's seen as running on battery, which is actually not (has neutral wire attached), and I'd like for it to be used as a router if possible.

Koenkk, 2021-03-27:
Can you share the data/database.db entry of this device?

Note that the power source is only a cosmetic thing, it does not influence the functionality of the device.

Zorinik, 2021-03-28:
So the fact that it is not acting as a router has nothing to do with the software?

Here is the database.db entry as requested anyway:

`{
  "id": 9,
  "type": "EndDevice",
  "ieeeAddr": "0xbc33acfffed85122",
  "nwkAddr": 6808,
  "manufId": 4098,
  "manufName": "_TYZB01_vzrytttn",
  "powerSource": "Battery",
  "modelId": "TS0012",
  "epList": [
    1,
    2
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 256,
      "inClusterList": [
        0,
        4,
        5,
        6
      ],
      "outClusterList": [
        25
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "65506": 19,
            "modelId": "TS0012",
            "manufacturerName": "_TYZB01_vzrytttn",
            "powerSource": 3,
            "appVersion": 65,
            "zclVersion": 3,
            "stackVersion": 0,
            "hwVersion": 1,
            "dateCode": ""
          }
        },
        "genOnOff": {
          "attributes": {
            "onOff": 0,
            "onTime": 0,
            "offWaitTime": 0
          }
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "2": {
      "profId": 260,
      "epId": 2,
      "devId": 256,
      "inClusterList": [
        4,
        5,
        6
      ],
      "outClusterList": [],
      "clusters": {
        "genOnOff": {
          "attributes": {
            "onOff": 0,
            "onTime": 0,
            "offWaitTime": 0
          }
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 65,
  "stackVersion": 0,
  "hwVersion": 1,
  "dateCode": "",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {},
  "lastSeen": 1616922351797
}`

Koenkk, 2021-03-28:
I see that the device advertises itself as and `EndDevice`, so it indeed has no routing capabilities.

Zorinik, 2021-03-29:
Ok so it's the hardware itself that can't handle it (I thought it was automatic since it's neutral-wired). Thanks for the replies :)
Hence the only thing left is the recognition of the power source, but it's a minor issue at this point :)

Koenkk, 2021-03-29:
I see the power source was already fixed (https://github.com/Koenkk/zigbee2mqtt/issues/6520). 
Changes are available in the dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html) and the next release which will be on 1 April.

closed 2021-03-29 by Koenkk