Koenkk/zigbee2mqtt #6851: Tuya wireless Zigbee switch not pairing with zigbee2mqtt

luk984, 2021-03-26:
## What happened
Switch won't pair with zigbee2mqtt. After putting the switch into pairing mode nothing happens.
## What did you expect to happen
To get switch paired with zigbee2mqtt
## How to reproduce it (minimal and precise)
I have 2 wireless switches Tuya (2-gang) bought on Ali. When I put them into pairing mode (hold for 5-10 sec) nothing happens (indicators on switches are blinking) . Most of the time there are also no logs in zigbee2mqtt. From time to time like every 6-10 attempts some logs appear (pasted below). Is this the hardware malfunction of the switches or some software problem with zigbee2mqtt ? Others devices connects without problem (socket wired switches, aquara motion sensor)
## Debug info
Zigbee2MQTT version: edge (but earlier i also tried the stable version without success)
Adapter hardware: CC2531

Zigbee2MQTT:debug 2021-03-26 20:35:09: Saving state to file /config/zigbee2mqtt/state.json
2021-03-26T20:35:12.010Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}
2021-03-26T20:35:12.017Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
2021-03-26T20:35:12.057Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82]
2021-03-26T20:35:12.057Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82]
2021-03-26T20:35:12.058Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82
2021-03-26T20:35:12.058Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0}
2021-03-26T20:35:12.059Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-03-26T20:35:12.062Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":4,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,5,2,11,254,0]}}
2021-03-26T20:35:12.063Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,4,0,30,6,0,25,5,2,11,254,0,234]
2021-03-26T20:35:12.066Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240]
2021-03-26T20:35:12.066Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
2021-03-26T20:35:12.067Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240
2021-03-26T20:35:12.067Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
2021-03-26T20:35:12.067Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-03-26T20:35:12.078Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
2021-03-26T20:35:12.079Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
2021-03-26T20:35:12.079Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103
2021-03-26T20:35:12.079Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0}
2021-03-26T20:35:12.079Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-03-26T20:35:12.085Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,4,49]
2021-03-26T20:35:12.085Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,4,49]
2021-03-26T20:35:12.086Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,4] - 49
2021-03-26T20:35:12.086Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":4}
2021-03-26T20:35:12.086Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-03-26T20:38:32.022Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}
2021-03-26T20:38:32.024Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
2021-03-26T20:38:32.040Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82]
2021-03-26T20:38:32.040Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82]
2021-03-26T20:38:32.041Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82
2021-03-26T20:38:32.041Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0}
2021-03-26T20:38:32.042Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-03-26T20:38:32.045Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":5,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,6,2,11,254,0]}}
2021-03-26T20:38:32.047Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,5,0,30,6,0,25,6,2,11,254,0,232]
2021-03-26T20:38:32.050Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240]
2021-03-26T20:38:32.051Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
2021-03-26T20:38:32.051Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240
2021-03-26T20:38:32.051Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
2021-03-26T20:38:32.051Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-03-26T20:38:32.059Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
2021-03-26T20:38:32.060Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
2021-03-26T20:38:32.060Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103
2021-03-26T20:38:32.060Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0}
2021-03-26T20:38:32.060Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-03-26T20:38:32.065Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,5,48]
2021-03-26T20:38:32.066Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,5,48]
2021-03-26T20:38:32.066Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,5] - 48
2021-03-26T20:38:32.066Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":5}
2021-03-26T20:38:32.066Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:debug 2021-03-26 20:40:09: Saving state to file /config/zigbee2mqtt/state.json
2021-03-26T20:41:52.032Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}
2021-03-26T20:41:52.034Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
2021-03-26T20:41:52.050Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82]
2021-03-26T20:41:52.051Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82]
2021-03-26T20:41:52.051Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82
2021-03-26T20:41:52.051Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0}
2021-03-26T20:41:52.052Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-03-26T20:41:52.056Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":6,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,7,2,11,254,0]}}
2021-03-26T20:41:52.057Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,6,0,30,6,0,25,7,2,11,254,0,234]
2021-03-26T20:41:52.061Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240]
2021-03-26T20:41:52.062Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
2021-03-26T20:41:52.062Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240
2021-03-26T20:41:52.062Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
2021-03-26T20:41:52.062Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-03-26T20:41:52.070Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
2021-03-26T20:41:52.070Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
[... cut ...]
2021-03-26T20:41:52.076Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,6,51]
2021-03-26T20:41:52.076Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,6,51]
2021-03-26T20:41:52.076Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,6] - 51
2021-03-26T20:41:52.077Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":6}
2021-03-26T20:41:52.077Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

gnisch, 2021-04-02:
I had the same problem with a similar device it seems that changing the channel helped in my case, maybe yours will be the same.

trashographer, 2021-04-04:
> I had the same problem with a similar device it seems that changing the channel helped in my case, maybe yours will be the same.

Help plz, how to change the channel? I got the same problem with Smart Power Strip

gnisch, 2021-04-04:
I have it in hassio addon settings section for zigbee2mqtt under the channel. In standalone zigbee2mqtt it is in configuration.yaml I think.

closed 2021-05-12 by github-actions[bot]