Koenkk/zigbee2mqtt #6856: Problem in Edge adding devices Sonoff Zigbee Bridge

gsshelton, 2021-03-27:
## What happened
Devices fail to pair. Similar issue to [https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/67](url) but I'm not getting the same error. The same 6 devices join immediately but fail to complete interview. Won't allow update or removal; "Device cannot be configured". Forced removed them and tried adding again. 

## What did you expect to happen
Devices to pair.

## How to reproduce it (minimal and precise)
As soon as permit to join is enabled, devices try to join but fail interview process and can only be force removed. Herdsmen logs attached.

## Debug info
Zigbee2MQTT version
    1.18.1-dev commit: f8f79d3

Coordinator type
    EZSP v7

Coordinator revision
    6.5.5.0 build 432

Frontend version
    0.3.113
[zigbee2mqttlog.txt](https://github.com/Koenkk/zigbee2mqtt/files/6216446/zigbee2mqttlog.txt)

kirovilya, 2021-03-28:
EZSP v7 not supported, v8 only - https://github.com/Koenkk/zigbee-herdsman/issues/319

gsshelton, 2021-03-28:
> EZSP v7 not supported, v8 only - [Koenkk/zigbee-herdsman#319](https://github.com/Koenkk/zigbee-herdsman/issues/319)

Thanks for the heads up. I've used the .bin files from here (http://ota.tasmota.com/tasmota/release/tasmota-zbbridge.bin.gz) to upgrade the firmware, but it still says it's version V7. Is there somewhere else I need to download a .bin for V8?

kirovilya, 2021-03-28:
@gsshelton read here https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html
> Use the latest ncp-uart-sw-x.x.x_115200.ota file (as of 2020/01/25 ncp-uart-sw-6.7.8_115200.ota is recommended)
https://github.com/arendst/Tasmota/tree/development/tools/fw_SonoffZigbeeBridge_ezsp

MattWestb, 2021-03-28:
The Sonoff devs have released one updated version that is recommend but tasmota have not putting in the git. Its one EZSP 6.7.8.0 with some setting is corrected but is named EZSP 6.7.9.0.
https://github.com/arendst/Tasmota/issues/10413#issuecomment-790274262

gsshelton, 2021-03-29:
I was able to successfully re-flash the bridge and updated it to version 8. @kirovilya Thank you for pointing me in the right direction!

closed 2021-03-29 by gsshelton

Hedda, 2021-03-29:
FYI; updated the PR for zigbee2mqtt.io website to mention that need firmware with "EZSP v8" support https://github.com/Koenkk/zigbee2mqtt.io/pull/655

PS: "EmberZNet NCP 6.7.8 firmware or later" was already mentioned so not sure if and how that can part could be emphasized(?).

MattWestb, 2021-03-29:
Best writing min EZSP version (6.7.0.0) and recommended one (6.7.8.0 but for Sonoff 6.7.9.0) and also writing that 6.8.X and higher is not recommended.

Hedda, 2021-03-29:
> Thanks for the heads up. I've used the .bin files from here (http://ota.tasmota.com/tasmota/release/tasmota-zbbridge.bin.gz) to upgrade the firmware, but it still says it's version V7. Is there [...]

@gsshelton If know it was not user error flashing EFR32MG21 then suggest report to https://github.com/arendst/Tasmota/issues/

That is, if all of Tasmota's instructions for using the current .bin files from http://ota.tasmota.com/tasmota/release/tasmota-zbbridge.bin.gz to upgrade the EmberZNet NCP firmware on the Silicon Labs EFR32MG21 does not point to ncp-uart-sw_6.7.8_115200.ota which has the new recommended stable version for EZSP v8 compatible hosts then please report that as issue to upstream Tasmota so that their documentation gets updated.

gsshelton, 2021-03-29:
@Hedda In Digiblur’s guide, the step using the ncp-uart-sw-6.7.8_115200.ota file, updates the bridge’s  chipset to V8.

Hedda, 2021-03-29:
> @Hedda In Digiblur’s guide, the step using the ncp-uart-sw-6.7.8_115200.ota file, updates the bridge’s chipset to V8.

So you followed some other guide which recommended an older version?

gsshelton, 2021-03-29:
No, not "some other guide", I followed the guide from the Ilya Kirov's comment above and used the file in the guide Ilya Kirov linked to.

Hedda, 2021-03-29:
Ilya's comment and digiblur blog refer to EZSP v8 == 6.7.8 so the question was how you managed to get a firmware with EZSP v7? 

I assumed that you had followed some other guide/documentation which had not been updated and suggested to use an older firmware and only meant that if so perhaps that other guide/documentation could be updated now so that someone else does not follow it and makes the same mistake. Again I was just hoping to help get the documentation updated with the latest current info.

Hedda, 2021-05-07:
By the way, also note that WiFi based Zigbee bridges are now not recommended with the EZSP protocol. See the warnings about it:

https://www.zigbee2mqtt.io/information/supported_adapters.html

WiFi-based Serial-to-IP bridges are not recommended for Silicon Labs EZSP adapters as the serial protocol does not have enough fault-tolerance to handle packet loss or latency delays that can normally occur over WiFi connections. If cannot use a locally connected USB or UART/GPIO adapter then the recommendation is to use remote adapter that connected via Ethernet (wired) to avoid issues with EZSP caused by WiFi connections.

https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

WiFi-based Serial-to-IP bridges are not recommended for Silicon Labs EZSP adapters as the serial protocol does not have enough fault-tolerance to handle packet loss or latency delays that can normally occur over WiFi connections.

More warning about Zigbee to WiFi bridges for EZSP is not recommended for production as it does not do well with packed loss:

https://github.com/zigpy/bellows#warning-about-zigbee-to-wifi-bridges

https://www.home-assistant.io/integrations/zha/#warning-about-wi-fi-based-zigbee-to-serial-bridgesgateways

FYI, I also understand that the EFR32 board antenna design of the "SM-011 V1.0" Zigbee module used inside ITead's Sonoff ZBBridge have very high receive sensitivity to electromagnetic interference (EMI) / radio-frequency interference. This means that it will be extra important to keep it away from any other electronic devices or appliances that may generate an electromagnetic field (EM Field) or send our interfering radio signals. That includes the computer that you run your home automation application software on as well as any devices such as for example external harddrives or storage devices, power-supplies, etc., and of course any Wi-Fi access points or routers nearby.

Note! The new ITead Zigbee 3.0 USB Dongle Model: 9888010100045 (Hardware Revision Version 1.3) circuit board share the same type of antenna design so its radio is as well extremely sensitive to signal interference on its receiving end. The current recommendation for using it is to connect it via a long USB extension cable and try to move it as far away from all electronics and other possible sources of signal interference.

PS: Another general tip that always applies when building a Zigbee network is to begin by adding a few good mains-powered Zigbee router devices so that they will make use a stable baseline. Many start by buying three "IKEA TRADFRI Signal Repeater" (E1746).

https://zigbee.blakadder.com/Ikea_E1746.html

https://www.ikea.com/us/en/p/tradfri-signal-repeater-30400407/

https://www.ikea.com/gb/en/p/tradfri-signal-repeater-80424255/

https://www.ikea.com/se/sv/p/tradfri-signalfoerstaerkare-10400408/