Koenkk/zigbee2mqtt #6861: Aqara Power Plug (SP-EUC01) not updating; some attributes missing

internetfreak, 2021-03-28:
I have an issue like similar to this one https://github.com/Koenkk/zigbee2mqtt/issues/4804
Since this issue is closed, I decided to open a new one to track it.

## What happened
Got myself an aqara EU power plug (SP-EUC01), paired it with my CC2562RB (the one from slaesh). Power measurement only happens very occasionally to none at all (device is seen reporting every few minutes at most if it wants to). Already tried to reconfigure and re-pair the device and updated the firmware of my stick but nothing changes.
Also, none of the mentioned properties other than stat, power and linkquality are reported (voltage, current etc are missing in the output)

## What did you expect to happen
Power and other atributes should be reported regularly, e.g. every few seconds (my Bitron AV2010/25 for example do report every few seconds)

## How to reproduce it (minimal and precise)
1. Pair SP-EUC01
2. Add some consumer to the plug (e.g. PC/Server)
3. Watch the zigbee2mqtt log for changes under the devices address/friendly name and see whether updates come in regularly.

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC2562RB
Adapter firmware version: coordinator_20210120 (latest available)

Simsalaba, 2021-04-10:
Same problem here. Worked for a couple of days and then suddenly nothing.

rsypen, 2021-04-12:
I have same issue here.
CURRENT, voltage, temperature and energy are not being updated/reported.
Power, lqi and switch works.

## Debug info
Zigbee2MQTT version: 1.18.2
Adapter hardware: CC2531
Adapter firmware version: CC2531_SOURCE_ROUTING_20201128

Milan6407, 2021-04-15:
I have same issue here.
CURRENT, voltage, temperature and energy are not being updated/reported (N/A).
Power, lqi and switch works.

### Debug info

Zigbee2MQTT version: 1.18.2-1
Adapter hardware: CC2531
Adapter firmware version: CC2531_SOURCE_ROUTING_20201128

Koenkk, 2021-04-17:
According to https://github.com/Koenkk/zigbee-herdsman-converters/issues/1050#issuecomment-673111969 not all revisions of this device support these features.

Simsalaba, 2021-04-20:
Mine worked perfectly for a couple of days until I hit the update button in home assistant.

internetfreak, 2021-04-28:
Reconfiguring via frontend does not help as well. From all potential values only power would be really interesting for me (the others could be nice to have but I wanted to measure power consumption of selected devices). Compared to for example the Bitron devices that plug is way smaller (although still big enough to block ports on a power strip), so it seems more attractive

pyjacques, 2021-05-31:
I have the same issue.

How can i get my current firmware version like you did @Milan6407 ?
> Adapter firmware version: CC2531_SOURCE_ROUTING_20201128

Frontend offers me to update my plugs ? Is-it a good ideal ? Maybe fix this issue ? Any tips before making a mistake ? ^^
[image]

yabbah, 2021-06-06:
I have the same issue, bought this one to measure current power-usage and historical (kwh). But I can only see the current watt using. Anyone figure out a way to solve this?

internetfreak, 2021-06-06:
Updates firmware of my plug but still having the issue that the plug seems to only update once every few minutes. For example, my Bitron sockets update themselves every few seconds. Is there something to configure or am I really out of luck and it is some kind of lottery when purchasing that socket if it works correctly?

andir, 2021-06-20:
I am also not seeing the power consumption of my plug. zigbee2mqtt is logging these lines from time to time:

```
Jun 20 09:12:25 crappy zigbee2mqtt[1148515]: Zigbee2MQTT:debug 2021-06-20 09:12:25: Received Zigbee message from '0x04cf8cdf3c8240ce', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,1,3,40,27,152,57,41,92,47,63,149,57,211,196,79,64,150,57,0,64,13,69,151,57,120,251,65,64,5,33,1,0,154,32,0,8,33,22,1,7,39,0,0,0,0,0,0,0,0,9,33,1,2,11,32,0,155,16,1],"type":"Buffer"}}' from endpoint 1 with groupID null
Jun 20 09:12:25 crappy zigbee2mqtt[1148515]: Zigbee2MQTT:debug 2021-06-20 09:12:25: No converter available for 'SP-EUC01' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"247":{"data":[100,16,1,3,40,27,152,57,41,92,47,63,149,57,211,196,79,64,150,57,0,64,13,69,151,57,120,251,65,64,5,33,1,0,154,32,0,8,33,22,1,7,39,0,0,0,0,0,0,0,0,9,33,1,2,11,32,0,155,16,1],"type":"Buffer"}}'
```
This is on the latest release (9bd46932c365e886ed0b1f9b8740d8cce632230c) as well as on dev (76a0cd507fee00b9cc21d2bc85d1da405265febb).

SMAW, 2021-08-16:
Same Issue here,

Debug info:
1.21.0 commit: 70891ee 
Coordinator type: zStack3x0 
Coordinator revision: 20200805
Frontend version: 0.5.18
Adapter Hardware : Electrolama zig-a-zig-ah! (zzh!)

islepnev, 2021-08-16:
> According to [Koenkk/zigbee-herdsman-converters#1050 (comment)](https://github.com/Koenkk/zigbee-herdsman-converters/issues/1050#issuecomment-673111969) not all revisions of this device support these features. You can also try reconfiguring the device via the zigbee2mqtt frontend by clicking on it and pressing the yellow refresh icon.

Two of four plugs (firmware 09-10-2019 and 09-22-2020) were missing all attributes but on/off state, power stuck at 0.
Likely the same issue.
Fixed by reconfiguring the device (yellow refresh button).

SMAW, 2021-08-16:
> Two of four plugs (firmware 09-10-2019 and 09-22-2020) were missing all attributes but on/off state, power stuck at 0.
> Likely the same issue.
> Fixed by reconfiguring the device (yellow refresh button).

Checked, mine is also from 09-22-2020, attributes are null.. Power is sometimes working, also after an reconfigure, still same issue... Reconfigure also takes about 1 second

YoannNicod, 2021-08-18:
If I can add my experience, my 3 Aqara plugs send correctly :
- state
- power
- lqi

but these are not recieved:
- energy
- temperature
- voltage
- current

I am using slaesh stick and zigbee2mqtt of course.

One month ago, I had the opportunity to test my plugs with a **combee II and DeCOnz.**
I don't remember for the others, but **at least Energy was working fine.**

I deduce the problem we are facing probably doesn't come from the plug itself.

Koenkk, 2021-08-18:
Can someone provide the debug logging starting from the point where you attach a known load to the plug (e.g. a 100w bulb).

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

jnxxx, 2021-08-18:
I am new to all of this, but let me give it a try.

First, I am seeing the same as others reported. I get power reported (and linkquality). Attributes energy, temperature, voltage and current show null.

This is the debug log from a point I powered it on with an 8W bulb:

```
debug 2021-08-19 00:37:02: Received Zigbee message from '0x04cf8cdf3c89f4cb', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2021-08-19 00:37:02: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c89f4cb', payload '{"current":null,"energy":null,"linkquality":92,"power":0,"power_phase_c":null,"state":"ON","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null}'
debug 2021-08-19 00:37:08: Received Zigbee message from '0x04cf8cdf3c89f4cb', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":82}' from endpoint 1 with groupID 0
info  2021-08-19 00:37:08: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c89f4cb', payload '{"current":null,"energy":null,"linkquality":81,"power":8.2,"power_phase_c":null,"state":"ON","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null}'
debug 2021-08-19 00:39:26: Received Zigbee message from '0x04cf8cdf3c89f4cb', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":24}' from endpoint 1 with groupID 0
info  2021-08-19 00:39:26: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c89f4cb', payload '{"current":null,"energy":null,"linkquality":73,"power":2.4,"power_phase_c":null,"state":"ON","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null}'
debug 2021-08-19 00:39:38: Received Zigbee message from '0x04cf8cdf3c89f4cb', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":82}' from endpoint 1 with groupID 0
info  2021-08-19 00:39:38: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c89f4cb', payload '{"current":null,"energy":null,"linkquality":78,"power":8.2,"power_phase_c":null,"state":"ON","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null}'

```

To cause a change I dimmed it, and back to full.

If I use the Dev console I can get it to show me energy and temperature:
```
debug 2021-08-19 00:43:38: Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c89f4cb/set' with data '{"read":{"attributes":["currentSummDelivered","multiplier","divisor"],"cluster":"seMetering","options":{}}}'
debug 2021-08-19 00:43:38: Publishing 'set' 'read' to '0x04cf8cdf3c89f4cb'
debug 2021-08-19 00:43:38: Received Zigbee message from '0x04cf8cdf3c89f4cb', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,47],"divisor":1000,"multiplier":1}' from endpoint 1 with groupID 0
debug 2021-08-19 00:44:48: Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c89f4cb/set' with data '{"read":{"attributes":["currentTemperature"],"cluster":"genDeviceTempCfg","options":{}}}'
debug 2021-08-19 00:44:48: Publishing 'set' 'read' to '0x04cf8cdf3c89f4cb'
debug 2021-08-19 00:44:48: Received Zigbee message from '0x04cf8cdf3c89f4cb', type 'readResponse', cluster 'genDeviceTempCfg', data '{"currentTemperature":22}' from endpoint 1 with groupID 0
debug 2021-08-19 00:56:38: Received Zigbee message from '0x04cf8cdf3c89f4cb', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":31}' from endpoint 1 with groupID 0
info  2021-08-19 00:56:38: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c89f4cb', payload '{"current":null,"energy":null,"linkquality":78,"power":3.1,"power_phase_c":null,"state":"ON","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null}'
```

Temperature looks ok.
Energy I cannot validate, but could very well be ok: 0,047 kWh, I assume (it is new)
And it does seem to increase slowly.

I could not find voltage and current. To me accumulated energy is the important one.

I am not sure if it matters, but for this device I see only one entry in the "Reporting" tab, onOff. Is this what the device is asked to send automatically? For other devices it seems so.
I tried to add seMetering/currentSummDelivered, but did not seem to change anything.
If I do the same for temperature, it says it is not reportable:
`Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x04cf8cdf3c89f4cb/1 genDeviceTempCfg([{"attribute":"currentTemperature","minimumReportInterval":60,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE')'
`

Firmware: 09-22-2020

Zigbee2MQTT version: 1.21.0 commit: unknown
Coordinator type: zStack12
Coordinator revision: 20180507
Frontend version: 0.5.18

djeclemen, 2021-08-19:
Same her, nothing reported

`state
On/off state of the switch
power
Instantaneous measured power
2.8W
energy
Sum of consumed energy
nullkWh
temperature
Measured temperature value
null°C
voltage
Measured electrical potential value
nullV
current
Instantaneous measured electrical current
nullA
linkquality
Link quality (signal strength)
167lqi`

but dev console
Read result of 'genDeviceTempCfg': {"currentTemperature":30}
Read result of 'seMetering': {"currentSummDelivered":[0,457]}

Koenkk, 2021-08-19:
I've did an attempt to fix it.
- Please update to the latest dev branch (wait 2 hours from now).  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)
- Go to the frontend and press the yellow reconfigure button, provide the debug log after you click it.
- Check if energy measurements come in

islepnev, 2021-08-19:
> I've did an attempt to fix it.

Thank you! I've tried with my sockets, but no luck actually. It could be that there are multiple hardware revisions out there, here is my log just in case it could be useful.

```
1.21.0-48-geebc92f9
...
Zigbee2MQTT:debug 2021-08-19 21:38:29: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Socket","transaction":"x32lb-1"}'
Zigbee2MQTT:info  2021-08-19 21:38:29: Configuring 'Socket'
Zigbee2MQTT:debug 2021-08-19 21:38:30: Received Zigbee message from 'Socket', type 'readResponse', cluster 'haElectricalMeasurement', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2021-08-19 21:38:30: SP-EUC01 failed to setup electricity measurements (Read 0x04cf8cdf3c765eb0/1 haElectricalMeasurement(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))
Zigbee2MQTT:debug 2021-08-19 21:38:30: Error: Read 0x04cf8cdf3c765eb0/1 haElectricalMeasurement(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:293:23)
    at Endpoint.read (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:413:22)
    at Object.readEletricalMeasurementMultiplierDivisors (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:23:5)
    at Object.configure (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/xiaomi.js:876:17)
    at Configure.configure (/home/z2m/zigbee2mqtt/lib/extension/configure.js:137:13)
    at Configure.onMQTTMessage (/home/z2m/zigbee2mqtt/lib/extension/configure.js:88:21)
    at Controller.callExtensionMethod (/home/z2m/zigbee2mqtt/lib/controller.js:413:58)
Zigbee2MQTT:info  2021-08-19 21:38:30: Successfully configured 'Socket'
Zigbee2MQTT:info  2021-08-19 21:38:30: MQTT publish: topic 'zigbee2mqtt/Socket', payload '{"current":null,"device":{"applicationVersion":32,"dateCode":"09-22-2020","friendlyName":"Socket","hardwareVersion":1,"ieeeAddr":"0x04cf8cdf3c765eb0","manufacturerID":4447,"manufacturerName":"LUMI","model":"SP-EUC01","networkAddress":39376,"powerSource":"Mains (single phase)","stackVersion":2,"type":"Router","zclVersion":3},"energy":null,"last_seen":1629398310005,"linkquality":69,"power":0,"state":"OFF","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null}'
Zigbee2MQTT:info  2021-08-19 21:38:30: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Socket"},"status":"ok","transaction":"x32lb-1"}'

```

Koenkk, 2021-08-19:
@islepnev how are you running z2m? (docker, ha addon, bare metal?)

islepnev, 2021-08-19:
baremetal, git switch/pull, npm ci; npm start

SMAW, 2021-08-19:
> 
> 
> @islepnev how are you running z2m? (docker, ha addon, bare metal?)

Just updated my docker container to latest-dev, but also after removing and adding device it doens't seem to work on my plug.

islepnev, 2021-08-19:
Playing with z2m Dev console, got the following readings:

```
debug 2021-08-19 22:45:35: Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c765eb0/set' with data '{"read":{"attributes":["divisor","multiplier","currentSummDelivered","unitOfMeasure","status","meteringDeviceType","demandFormatting","summaFormatting"],"cluster":"seMetering","options":{}}}'
debug 2021-08-19 22:45:35: Received Zigbee message from 'Aquarium Light 1', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,178],"demandFormatting":0,"divisor":1000,"meteringDeviceType":0,"multiplier":1,"status":0,"summaFormatting":35,"unitOfMeasure":0}' from endpoint 1 with groupID 0
debug 2021-08-19 22:45:35: No converter available for 'SP-EUC01' with cluster 'seMetering' and type 'readResponse' and data '{"currentSummDelivered":[0,178],"demandFormatting":0,"divisor":1000,"meteringDeviceType":0,"multiplier":1,"status":0,"summaFormatting":35,"unitOfMeasure":0}'

```

178 looks like energy measurement (Wh) and it is increasing.

Other clusters:

```
debug 2021-08-19 22:31:35: Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c765eb0/set' with data '{"read":{"attributes":["currentTemperature","devTempAlarmMask","highTempThres","highTempDwellTripPoint"],"cluster":"genDeviceTempCfg","options":{}}}'
debug 2021-08-19 22:31:36: Received Zigbee message from 'Aquarium Light 1', type 'readResponse', cluster 'genDeviceTempCfg', data '{"currentTemperature":27,"devTempAlarmMask":2,"highTempDwellTripPoint":3,"highTempThres":65}' from endpoint 1 with groupID 0
debug 2021-08-19 22:31:36: No converter available for 'SP-EUC01' with cluster 'genDeviceTempCfg' and type 'readResponse' and data '{"currentTemperature":27,"devTempAlarmMask":2,"highTempDwellTripPoint":3,"highTempThres":65}'
...
debug 2021-08-19 22:50:38: Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c765eb0/set' with data '{"read":{"attributes":["currentTemperature"],"cluster":"genDeviceTempCfg","options":{}}}'
debug 2021-08-19 22:50:38: Received Zigbee message from 'Aquarium Light 1', type 'readResponse', cluster 'genDeviceTempCfg', data '{"currentTemperature":28}' from endpoint 1 with groupID 0
debug 2021-08-19 22:50:38: No converter available for 'SP-EUC01' with cluster 'genDeviceTempCfg' and type 'readResponse' and data '{"currentTemperature":28}'
```
genDeviceTempCfg / currentTemperature is increasing as socket warms up.

islepnev, 2021-08-19:
No multiplier config for voltage/current in haElectricalMeasurement, only power:
```
debug 2021-08-19 23:00:32: Received MQTT message on 'zigbee2mqtt/0x04cf8cdf3c765eb0/set' with data '{"read":{"attributes":["acPowerDivisor","acPowerMultiplier","acActivePowerOverload"],"cluster":"haElectricalMeasurement","options":{}}}'
debug 2021-08-19 23:00:32: Publishing 'set' 'read' to 'Aquarium Light 1'
debug 2021-08-19 23:00:32: Received Zigbee message from 'Aquarium Light 1', type 'readResponse', cluster 'haElectricalMeasurement', data '{"acActivePowerOverload":2300,"acPowerDivisor":10,"acPowerMultiplier":1}' from endpoint 1 with groupID 0
info  2021-08-19 23:00:32: Read result of 'haElectricalMeasurement': {"acPowerDivisor":10,"acPowerMultiplier":1,"acActivePowerOverload":2300}
```

jnxxx, 2021-08-19:
Unfortunately still the same.

The lines you added seem to add some things ro reporting. Should this be visible in the reporting tab for that device?
I still see only one line there, onOff.

Mine does not seem to respond to current or power, so it may not get to currentSummDelivered. Could be this should be right under activePower.
However, I do not see the warn message either...

If I add currentSummDelivered myself to reporting, and set max rep interval lower I do see it coming in regularly. But it says there is no converter.

```
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"0x04cf8cdf3c89f4cb","transaction":"3pq86-1"}'
Info Configuring '0x04cf8cdf3c89f4cb'
Debug Received Zigbee message from '0x04cf8cdf3c89f4cb', type 'readResponse', cluster 'haElectricalMeasurement', data '{"acPowerDivisor":10,"acPowerMultiplier":1}' from endpoint 1 with groupID 0
Info Successfully configured '0x04cf8cdf3c89f4cb'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x04cf8cdf3c89f4cb"},"status":"ok","transaction":"3pq86-1"}'
Debug Received Zigbee message from '0x04cf8cdf3c89f4cb', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,71]}' from endpoint 1 with groupID 0
Debug No converter available for 'SP-EUC01' with cluster 'seMetering' and type 'attributeReport' and data '{"currentSummDelivered":[0,71]}'
Debug Received Zigbee message from '0x04cf8cdf3c89f4cb', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,71]}' from endpoint 1 with groupID 0
Debug No converter available for 'SP-EUC01' with cluster 'seMetering' and type 'attributeReport' and data '{"currentSummDelivered":[0,71]}'
Debug Received Zigbee message from '0x04cf8cdf3c89f4cb', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":81}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c89f4cb', payload '{"current":null,"energy":null,"linkquality":78,"power":8.1,"power_phase_c":null,"state":"ON","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null}'
Debug Received Zigbee message from '0x04cf8cdf3c89f4cb', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,71]}' from endpoint 1 with groupID 0
Debug No converter available for 'SP-EUC01' with cluster 'seMetering' and type 'attributeReport' and data '{"currentSummDelivered":[0,71]}'
```

Koenkk, 2021-08-20:
@islepnev can you:
- stop z2m
- In the z2m directory execute: `nano node_modules/zigbee-herdsman-converters/devices/xiaomi.js` and change this part: https://github.com/Koenkk/zigbee-herdsman-converters/blob/0a256b5754470fa4a244cfc09594cb5605fe8d1f/devices/xiaomi.js#L864

```js
    {
        zigbeeModel: ['lumi.plug.maeu01'],
        model: 'SP-EUC01',
        description: 'Aqara EU smart plug',
        vendor: 'Xiaomi',
        fromZigbee: [fz.on_off, fz.xiaomi_switch_basic, fz.electrical_measurement, fz.metering],
        toZigbee: [tz.on_off],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint);
            try {
                await reporting.bind(endpoint, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
                await reporting.activePower(endpoint);
                await reporting.rmsCurrent(endpoint);
                await reporting.rmsVoltage(endpoint);
                await reporting.currentSummDelivered(endpoint);
                await endpoint.read('seMetering', ['multiplier', 'divisor']);
                await endpoint.read('haElectricalMeasurement', ['acVoltageMultiplier', 'acVoltageDivisor']);
                 await endpoint.read('haElectricalMeasurement', [ 'acPowerMultiplier', 'acPowerDivisor']);
                await endpoint.read('haElectricalMeasurement', [ 'acCurrentDivisor', 'acCurrentMultiplier']);
            } catch (e) {
                logger.warn(`SP-EUC01 failed to setup electricity measurements (${e.message})`);
                logger.debug(e.stack);
                // Not all plugs support this.
                // https://github.com/Koenkk/zigbee-herdsman-converters/issues/1050#issuecomment-673111969
            }

            // Voltage/current doesn't seem to be supported, maybe in futurue revisions of the device (?).
            // https://github.com/Koenkk/zigbee-herdsman-converters/issues/1050
        },
        exposes: [e.switch(), e.power(), e.energy(), e.temperature().withAccess(ea.STATE), e.voltage().withAccess(ea.STATE), e.current()],
        ota: ota.zigbeeOTA,
    },
````

Again provide the debug logging after reconfiguring via the frontend + the data/database.db entry

islepnev, 2021-08-20:
@Koenkk,

reconfiguration failed:
```
Zigbee2MQTT:debug 2021-08-20 21:34:13: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Aquarium Light 1","transaction":"ze2g6-1"}'
Zigbee2MQTT:info  2021-08-20 21:34:13: Configuring 'Aquarium Light 1'
Zigbee2MQTT:debug 2021-08-20 21:34:13: Received Zigbee message from 'Aquarium Light 1', type 'readResponse', cluster 'haElectricalMeasurement', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2021-08-20 21:34:13: SP-EUC01 failed to setup electricity measurements (Read 0x04cf8cdf3c765eb0/1 haElectricalMeasurement(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))
Zigbee2MQTT:debug 2021-08-20 21:34:14: Error: Read 0x04cf8cdf3c765eb0/1 haElectricalMeasurement(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:293:23)
    at Endpoint.read (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:413:22)
    at Object.readEletricalMeasurementMultiplierDivisors (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:23:5)
    at Object.configure (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/xiaomi.js:876:17)
    at Configure.configure (/home/z2m/zigbee2mqtt/lib/extension/configure.js:137:13)
    at Configure.onMQTTMessage (/home/z2m/zigbee2mqtt/lib/extension/configure.js:88:21)
    at Controller.callExtensionMethod (/home/z2m/zigbee2mqtt/lib/controller.js:413:58)
Zigbee2MQTT:info  2021-08-20 21:34:14: Successfully configured 'Aquarium Light 1'
Zigbee2MQTT:info  2021-08-20 21:34:14: MQTT publish: topic 'zigbee2mqtt/Aquarium Light 1', payload '{"current":null,"device":{"applicationVersion":32,"dateCode":"09-22-2020","friendlyName":"Aquarium Light 1","hardwareVersion":1,"ieeeAddr":"0x04cf8cdf3c765eb0","manufacturerID":4447,"manufacturerName":"LUMI","model":"SP-EUC01","networkAddress":39376,"powerSource":"Mains (single phase)","stackVersion":2,"type":"Router","zclVersion":3},"energy":null,"last_seen":1629484453758,"linkquality":63,"power":0,"power_phase_c":null,"state":"OFF","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null,"voltage_phase_c":null}'
Zigbee2MQTT:info  2021-08-20 21:34:14: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Aquarium Light 1"},"status":"ok","transaction":"ze2g6-1"}'
```
```
{
  "id": 3,
  "type": "Router",
  "ieeeAddr": "0x04cf8cdf3c765eb0",
  "nwkAddr": 39376,
  "manufId": 4447,
  "manufName": "LUMI",
  "powerSource": "Mains (single phase)",
  "modelId": "lumi.plug.maeu01",
  "epList": [
    1,
    242
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 81,
      "inClusterList": [
        0,
        2,
        3,
        4,
        5,
        6,
        9,
        1794,
        2820
      ],
      "outClusterList": [
        10,
        25
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "lumi.plug.maeu01",
            "manufacturerName": "LUMI",
            "powerSource": 1,
            "zclVersion": 3,
            "appVersion": 32,
            "stackVersion": 2,
            "hwVersion": 1
          }
        },
        "haElectricalMeasurement": {
          "attributes": {
            "acPowerMultiplier": 1,
            "acPowerDivisor": 10,
[... 87 lines cut ...]
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [
        33
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 32,
  "stackVersion": 2,
  "hwVersion": 1,
  "dateCode": "09-22-2020",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {
    "configured": 1065536675
  },
  "lastSeen": 1629484453758
}
```

islepnev, 2021-08-20:
@Koenkk,

```
Zigbee2MQTT:info  2021-08-20 21:48:25: Configuring 'Aquarium Light 1'
Zigbee2MQTT:warn  2021-08-20 21:48:26: SP-EUC01 failed to setup electricity measurements (ConfigureReporting 0x04cf8cdf3c765eb0/1 haElectricalMeasurement([{"attribute":"activePower","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE'))
Zigbee2MQTT:debug 2021-08-20 21:48:26: Error: ConfigureReporting 0x04cf8cdf3c765eb0/1 haElectricalMeasurement([{"attribute":"activePower","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
    at Endpoint.checkStatus (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:293:23)
    at Endpoint.configureReporting (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:627:22)
    at Object.activePower (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:172:9)
    at Object.configure (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/xiaomi.js:876:17)
    at Configure.configure (/home/z2m/zigbee2mqtt/lib/extension/configure.js:137:13)
    at Configure.onMQTTMessage (/home/z2m/zigbee2mqtt/lib/extension/configure.js:88:21)
    at Controller.callExtensionMethod (/home/z2m/zigbee2mqtt/lib/controller.js:413:58)
Zigbee2MQTT:info  2021-08-20 21:48:26: Successfully configured 'Aquarium Light 1'
Zigbee2MQTT:info  2021-08-20 21:48:26: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Aquarium Light 1"},"status":"ok","transaction":"ze2g6-2"}'
```

```
{
  "id": 3,
  "type": "Router",
  "ieeeAddr": "0x04cf8cdf3c765eb0",
  "nwkAddr": 39376,
  "manufId": 4447,
  "manufName": "LUMI",
  "powerSource": "Mains (single phase)",
  "modelId": "lumi.plug.maeu01",
  "epList": [
    1,
    242
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 81,
      "inClusterList": [
        0,
        2,
        3,
        4,
        5,
        6,
        9,
        1794,
        2820
      ],
      "outClusterList": [
        10,
        25
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "lumi.plug.maeu01",
            "manufacturerName": "LUMI",
            "powerSource": 1,
            "zclVersion": 3,
            "appVersion": 32,
            "stackVersion": 2,
            "hwVersion": 1
          }
        },
        "haElectricalMeasurement": {
          "attributes": {
            "acPowerMultiplier": 1,
            "acPowerDivisor": 10,
            "activePower": 0,
            "measurementType": 8,
            "acActivePowerOverload": 2300
          }
        },
        "genOnOff": {
          "attributes": {
            "onOff": 0
          }
        },
        "seMetering": {
          "attributes": {
            "currentSummDelivered": [
              0,
              656
            ],
            "divisor": 1000,
            "multiplier": 1,
            "unitOfMeasure": 0,
            "status": 0,
            "meteringDeviceType": 0,
            "demandFormatting": 0,
            "summaFormatting": 35
          }
        },
        "genDeviceTempCfg": {
          "attributes": {
            "currentTemperature": 28,
            "highTempThres": 65,
            "devTempAlarmMask": 2,
            "highTempDwellTripPoint": 3
          }
        },
        "genIdentify": {
          "attributes": {
            "identifyTime": 0
          }
        },
        "genGroups": {
          "attributes": {
            "nameSupport": 128
          }
        },
        "aqaraOpple": {
          "attributes": {
            "mode": 0
          }
        }
      },
      "binds": [
        {
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0020cd1fd4",
          "endpointID": 1
        },
        {
          "cluster": 2820,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0020cd1fd4",
          "endpointID": 1
        },
        {
          "cluster": 1794,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0020cd1fd4",
          "endpointID": 1
        },
        {
          "cluster": 2,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b0020cd1fd4",
          "endpointID": 1
        }
      ],
      "configuredReportings": [
        {
          "cluster": 6,
          "attrId": 0,
          "minRepIntval": 0,
          "maxRepIntval": 3600,
          "repChange": 0
        }
      ],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [
        33
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 32,
  "stackVersion": 2,
  "hwVersion": 1,
  "dateCode": "09-22-2020",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {
    "configured": 889203848
  },
  "lastSeen": 1629485306236
}
```

islepnev, 2021-08-20:
commented out problem lines, reconfigure passed and energy showed up!
no temperature however.
```
Zigbee2MQTT:debug 2021-08-20 22:00:04: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Aquarium Light 1","transaction":"ze2g6-3"}'
Zigbee2MQTT:info  2021-08-20 22:00:04: Configuring 'Aquarium Light 1'
Zigbee2MQTT:debug 2021-08-20 22:00:04: Received Zigbee message from 'Aquarium Light 1', type 'readResponse', cluster 'seMetering', data '{"divisor":1000,"multiplier":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-08-20 22:00:04: MQTT publish: topic 'zigbee2mqtt/Aquarium Light 1', payload '{"current":null,"device":{"applicationVersion":32,"dateCode":"09-22-2020","friendlyName":"Aquarium Light 1","hardwareVersion":1,"ieeeAddr":"0x04cf8cdf3c765eb0","manufacturerID":4447,"manufacturerName":"LUMI","model":"SP-EUC01","networkAddress":39376,"powerSource":"Mains (single phase)","stackVersion":2,"type":"Router","zclVersion":3},"energy":0.66,"last_seen":1629486004688,"linkquality":63,"power":0,"power_phase_c":null,"state":"OFF","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null,"voltage_phase_c":null}'
Zigbee2MQTT:debug 2021-08-20 22:00:04: Received Zigbee message from 'Aquarium Light 1', type 'readResponse', cluster 'haElectricalMeasurement', data '{"acPowerDivisor":10,"acPowerMultiplier":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-08-20 22:00:04: Successfully configured 'Aquarium Light 1'
Zigbee2MQTT:info  2021-08-20 22:00:05: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Aquarium Light 1"},"status":"ok","transaction":"ze2g6-3"}'
Zigbee2MQTT:info  2021-08-20 22:00:05: MQTT publish: topic 'zigbee2mqtt/Aquarium Light 1', payload '{"current":null,"device":{"applicationVersion":32,"dateCode":"09-22-2020","friendlyName":"Aquarium Light 1","hardwareVersion":1,"ieeeAddr":"0x04cf8cdf3c765eb0","manufacturerID":4447,"manufacturerName":"LUMI","model":"SP-EUC01","networkAddress":39376,"powerSource":"Mains (single phase)","stackVersion":2,"type":"Router","zclVersion":3},"energy":0.66,"last_seen":1629486004777,"linkquality":66,"power":0,"power_phase_c":null,"state":"OFF","temperature":null,"update":{"state":"idle"},"update_available":false,"voltage":null,"voltage_phase_c":null}'
```
```
{
  "id": 3,
  "type": "Router",
  "ieeeAddr": "0x04cf8cdf3c765eb0",
  "nwkAddr": 39376,
  "manufId": 4447,
  "manufName": "LUMI",
  "powerSource": "Mains (single phase)",
  "modelId": "lumi.plug.maeu01",
  "epList": [
    1,
    242
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 81,
      "inClusterList": [
        0,
        2,
        3,
        4,
        5,
        6,
        9,
        1794,
        2820
      ],
      "outClusterList": [
        10,
        25
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "lumi.plug.maeu01",
            "manufacturerName": "LUMI",
            "powerSource": 1,
            "zclVersion": 3,
            "appVersion": 32,
            "stackVersion": 2,
            "hwVersion": 1
          }
        },
        "haElectricalMeasurement": {
          "attributes": {
            "acPowerMultiplier": 1,
            "acPowerDivisor": 10,
            "activePower": 0,
            "measurementType": 8,
            "acActivePowerOverload": 2300
          }
        },
        "genOnOff": {
          "attributes": {
            "onOff": 0
[... 127 lines cut ...]
            await reporting.onOff(endpoint);
            try {
                await reporting.bind(endpoint, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering']);
                // await reporting.activePower(endpoint);
                // await reporting.rmsCurrent(endpoint);
                // await reporting.rmsVoltage(endpoint);
                await reporting.currentSummDelivered(endpoint);
                await endpoint.read('seMetering', ['multiplier', 'divisor']);
                // await endpoint.read('haElectricalMeasurement', ['acVoltageMultiplier', 'acVoltageDivisor']);
                 await endpoint.read('haElectricalMeasurement', [ 'acPowerMultiplier', 'acPowerDivisor']);
                // await endpoint.read('haElectricalMeasurement', [ 'acCurrentDivisor', 'acCurrentMultiplier']);
            } catch (e) {
                logger.warn(`SP-EUC01 failed to setup electricity measurements (${e.message})`);
                logger.debug(e.stack);
                // Not all plugs support this.
                // https://github.com/Koenkk/zigbee-herdsman-converters/issues/1050#issuecomment-673111969
            }

            // Voltage/current doesn't seem to be supported, maybe in futurue revisions of the device (?).
            // https://github.com/Koenkk/zigbee-herdsman-converters/issues/1050
        },
        exposes: [e.switch(), e.power(), e.energy(), e.temperature().withAccess(ea.STATE), e.voltage().withAccess(ea.STATE), e.current()],
        ota: ota.zigbeeOTA,
    },
```

Koenkk, 2021-08-21:
@islepnev thanks!

- updated the configuration, so now `energy` and `power` should work
- removed `temperature`, `voltage` and `current` since device does not seem to support this.

Can anybody else confirm if it works correctly now? Then this issue can be closed.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

islepnev, 2021-08-21:
It looks like activePower reporting is configured but the error is signaled.
I tried to configure haElectricalMeasurement/activePower reporting by GUI, got error, but reporting works.

```
Zigbee2MQTT:debug 2021-08-21 20:24:50: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Aquarium Light 1","transaction":"ledge-6"}'
Zigbee2MQTT:info  2021-08-21 20:24:50: Configuring 'Aquarium Light 1'
Zigbee2MQTT:debug 2021-08-21 20:24:50: Received Zigbee message from 'Aquarium Light 1', type 'readResponse', cluster 'haElectricalMeasurement', data '{"acPowerDivisor":10,"acPowerMultiplier":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-08-21 20:24:50: MQTT publish: topic 'zigbee2mqtt/Aquarium Light 1', payload '{"device":{"applicationVersion":32,"dateCode":"09-22-2020","friendlyName":"Aquarium Light 1","hardwareVersion":1,"ieeeAddr":"0x04cf8cdf3c765eb0","manufacturerID":4447,"manufacturerName":"LUMI","model":"SP-EUC01","networkAddress":39376,"powerSource":"Mains (single phase)","stackVersion":2,"type":"Router","zclVersion":3},"energy":1.28,"last_seen":1629566690670,"linkquality":93,"power":0,"power_phase_c":null,"state":"OFF","update":{"state":"idle"},"update_available":false,"voltage":null,"voltage_phase_c":null}'
Zigbee2MQTT:warn  2021-08-21 20:24:50: SP-EUC01 failed to setup electricity measurements (ConfigureReporting 0x04cf8cdf3c765eb0/1 haElectricalMeasurement([{"attribute":"activePower","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE'))
Zigbee2MQTT:debug 2021-08-21 20:24:50: Error: ConfigureReporting 0x04cf8cdf3c765eb0/1 haElectricalMeasurement([{"attribute":"activePower","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
    at Endpoint.checkStatus (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:293:23)
    at Endpoint.configureReporting (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:627:22)
    at Object.activePower (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:172:9)
    at Object.configure (/home/z2m/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/xiaomi.js:881:17)
    at Configure.configure (/home/z2m/zigbee2mqtt/lib/extension/configure.js:137:13)
    at Configure.onMQTTMessage (/home/z2m/zigbee2mqtt/lib/extension/configure.js:88:21)
    at Controller.callExtensionMethod (/home/z2m/zigbee2mqtt/lib/controller.js:413:58)
Zigbee2MQTT:debug 2021-08-21 20:24:50: Received Zigbee message from 'Aquarium Light 1', type 'readResponse', cluster 'seMetering', data '{"divisor":1000,"multiplier":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-08-21 20:24:51: MQTT publish: topic 'zigbee2mqtt/Aquarium Light 1', payload '{"device":{"applicationVersion":32,"dateCode":"09-22-2020","friendlyName":"Aquarium Light 1","hardwareVersion":1,"ieeeAddr":"0x04cf8cdf3c765eb0","manufacturerID":4447,"manufacturerName":"LUMI","model":"SP-EUC01","networkAddress":39376,"powerSource":"Mains (single phase)","stackVersion":2,"type":"Router","zclVersion":3},"energy":1.28,"last_seen":1629566690958,"linkquality":90,"power":0,"power_phase_c":null,"state":"OFF","update":{"state":"idle"},"update_available":false,"voltage":null,"voltage_phase_c":null}'
Zigbee2MQTT:info  2021-08-21 20:24:51: Successfully configured 'Aquarium Light 1'
Zigbee2MQTT:info  2021-08-21 20:24:51: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Aquarium Light 1"},"status":"ok","transaction":"ledge-6"}'

```

currentSummDelivered is NOT being reported. Silence from the device. Notice red square on the picture below.
[image]

Fixed this by entering '0' as Min rep change and Apply.

Here is so long awaited result:
```
debug 2021-08-21 21:04:37: Received Zigbee message from 'Aquarium Light 1', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1343]}' from endpoint 1 with groupID 0
debug 2021-08-21 21:04:42: Received Zigbee message from 'Aquarium Light 1', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1343]}' from endpoint 1 with groupID 0
...
debug 2021-08-21 21:05:15: Received Zigbee message from 'Aquarium Light 1', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1343]}' from endpoint 1 with groupID 0
debug 2021-08-21 21:05:20: Received Zigbee message from 'Aquarium Light 1', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1345]}' from endpoint 1 with groupID 0
debug 2021-08-21 21:05:26: Received Zigbee message from 'Aquarium Light 1', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1345]}' from endpoint 1 with groupID 0
[... cut ...]
debug 2021-08-21 21:08:35: Received Zigbee message from 'Aquarium Light 1', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1348]}' from endpoint 1 with groupID 0
debug 2021-08-21 21:08:40: Received Zigbee message from 'Aquarium Light 1', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1350]}' from endpoint 1 with groupID 0

```

I've set Min rep interval to 1, reporting stopped completely. Values 0.1, 0.01, 0.001 are equivalent to 0 or being ignored.

djeclemen, 2021-08-22:
> correctly now? Then this issue can be closed.

Updated to dev, 

Temp and voltage are now gone, but energy are still NULL, will keep the switch with load to see if it changes. If I do a reconfigure it gives the following warning:

SP-EUC01 failed to setup electricity measurements (ConfigureReporting 0x04cf8cdf3c89eba8/1 haElectricalMeasurement([{"attribute":"activePower","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE'))

Koenkk, 2021-08-22:
added the fixes, can you try to repair and see if it still works?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

coom, 2021-08-23:
Hi,

I saved my Zigbee2MQTT HA addon config, then removed it, and installed Zigbee2MQTT Edge, but still the same issue here.

[image]

Here is what I can observ in the log :

```
Debug No converter available for 'SP-EUC01' with cluster 'genAnalogInput' and type 'attributeReport' and data '{"presentValue":0}'
Debug No converter available for 'SP-EUC01' with cluster 'genAnalogInput' and type 'attributeReport' and data '{"presentValue":40.30699920654297}'
Debug No converter available for 'SP-EUC01' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"247":{"data":[100,16,1,3,40,20,152,57,70,182,21,66,149,57,172,142,172,58,150,57,0,224,28,69,151,57,149,29,21,67,5,33,1,0,154,32,0,8,33,32,1,9,33,0,21,11,32,0,155,16,1,15,35,0,0,68,180],"type":"Buffer"}}'
```

Is there something I miss ?

islepnev, 2021-08-23:
> added the fixes, can you try to repair and see if it still works?

@Koenkk it works, thank you!

[image]

One small adjustment, I increased report interval 5 to 60.
```
debug 2021-08-23 14:01:25: Received Zigbee message from 'Aquarium Light 1', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,2413]}' from endpoint 1 with groupID 0
debug 2021-08-23 14:02:30: Received Zigbee message from 'Aquarium Light 1', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,2415]}' from endpoint 1 with groupID 0
debug 2021-08-23 14:03:34: Received Zigbee message from 'Aquarium Light 1', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,2417]}' from endpoint 1 with groupID 0
debug 2021-08-23 14:03:45: Received Zigbee message from 'Aquarium Light 1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
debug 2021-08-23 14:03:45: Received Zigbee message from 'Aquarium Light 1', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
```

coom, 2021-08-23:
@islepnev can you confirm you're current firmware version of the plug ?

Mine is 09-22-2020, even after I tried to reset/pair it again : still no energy/power.

Using Zigbee2MQTT as a supervised addon into HA, 1.21.0-dev commit: d7ae598

It seems the value is correctly sent by the plug as I can see some lines like that in my logfiles :
```
 No converter available for 'SP-EUC01' with cluster 'genAnalogInput' and type 'attributeReport' and data '{"presentValue":37.428001403808594}'
 ```

islepnev, 2021-08-23:
@coom, I use exactly the same zigbee2mqtt: 1.21.0-58-gd7ae5988

check the contents of node_modules/zigbee-herdsman-converters/devices/xiaomi.js, it should match https://github.com/Koenkk/zigbee-herdsman-converters/blob/f19fe49024768e0b721f999490d644ef388dbd6f/devices/xiaomi.js according to https://github.com/Koenkk/zigbee-herdsman-converters/commit/f19fe49024768e0b721f999490d644ef388dbd6f

Same 09-22-2020 here.

>  No converter available for 'SP-EUC01' with cluster 'genAnalogInput' and type 'attributeReport' and data '{"presentValue":37.428001403808594}'

My device does not have "genAnalogInput" cluster at all:
[image]

coom, 2021-08-23:
@islepnev thank you for your feedback, but since I'm using Zigbee2MQTT as an Home Assistant addon, i'm not able to modify the JS file, I can't even find it : 

[image]

I think I'll have to wait for the next release ;)

coom, 2021-08-23:
Mmhhh, digged into the issue as I'm able to get into console.

Find the file xiaomi.js, and everything is ok : 

[image]

Tried to reboot the entire host, delete the plug from zigbee2mqtt, pair it again, wait for the interview to be complete and then click "reconfigure" but still nothing :

It seems I forget something.

Got this in the log when I pair my plug :

```
Debug No converter available for 'SP-EUC01' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"247":{"data":[100,16,0,3,40,19,152,57,0,0,0,0,149,57,0,0,0,0,150,57,0,96,31,69,151,57,0,0,0,0,5,33,1,0,154,32,16,8,33,32,1,9,33,0,21,11,32,0,155,16,1,15,35,0,0,98,190],"type":"Buffer"}}'
Info MQTT publish: topic 'homeassistant/switch/0x54ef44100013a0a6/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x54ef44100013a0a6/set","device":{"identifiers":["zigbee2mqtt_0x54ef44100013a0a6"],"manufacturer":"Xiaomi","model":"Aqara EU smart plug (SP-EUC01)","name":"0x54ef44100013a0a6","sw_version":"Zigbee2MQTT 1.21.0-dev"},"json_attributes_topic":"zigbee2mqtt/0x54ef44100013a0a6","name":"0x54ef44100013a0a6","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x54ef44100013a0a6","unique_id":"0x54ef44100013a0a6_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x54ef44100013a0a6/energy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x54ef44100013a0a6"],"manufacturer":"Xiaomi","model":"Aqara EU smart plug (SP-EUC01)","name":"0x54ef44100013a0a6","sw_version":"Zigbee2MQTT 1.21.0-dev"},"device_class":"energy","json_attributes_topic":"zigbee2mqtt/0x54ef44100013a0a6","last_reset_topic":"zigbee2mqtt/0x54ef44100013a0a6","last_reset_value_template":"1970-01-01T00:00:00+00:00","name":"0x54ef44100013a0a6 energy","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef44100013a0a6","unique_id":"0x54ef44100013a0a6_energy_zigbee2mqtt","unit_of_measurement":"kWh","value_template":"{{ value_json.energy }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x54ef44100013a0a6/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x54ef44100013a0a6"],"manufacturer":"Xiaomi","model":"Aqara EU smart plug (SP-EUC01)","name":"0x54ef44100013a0a6","sw_version":"Zigbee2MQTT 1.21.0-dev"},"enabled_by_default":false,"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x54ef44100013a0a6","name":"0x54ef44100013a0a6 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef44100013a0a6","unique_id":"0x54ef44100013a0a6_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info MQTT publish: topic 'homeassistant/binary_sensor/0x54ef44100013a0a6/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x54ef44100013a0a6"],"manufacturer":"Xiaomi","model":"Aqara EU smart plug (SP-EUC01)","name":"0x54ef44100013a0a6","sw_version":"Zigbee2MQTT 1.21.0-dev"},"enabled_by_default":false,"json_attributes_topic":"zigbee2mqtt/0x54ef44100013a0a6","name":"0x54ef44100013a0a6 update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x54ef44100013a0a6","unique_id":"0x54ef44100013a0a6_update_available_zigbee2mqtt","value_template":"{{ value_json.update_available}}"}'
Debug Received MQTT message on 'homeassistant/switch/0x54ef44100013a0a6/switch/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x54ef44100013a0a6/set","device":{"identifiers":["zigbee2mqtt_0x54ef44100013a0a6"],"manufacturer":"Xiaomi","model":"Aqara EU smart plug (SP-EUC01)","name":"0x54ef44100013a0a6","sw_version":"Zigbee2MQTT 1.21.0-dev"},"json_attributes_topic":"zigbee2mqtt/0x54ef44100013a0a6","name":"0x54ef44100013a0a6","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x54ef44100013a0a6","unique_id":"0x54ef44100013a0a6_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Debug Received MQTT message on 'homeassistant/sensor/0x54ef44100013a0a6/energy/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x54ef44100013a0a6"],"manufacturer":"Xiaomi","model":"Aqara EU smart plug (SP-EUC01)","name":"0x54ef44100013a0a6","sw_version":"Zigbee2MQTT 1.21.0-dev"},"device_class":"energy","json_attributes_topic":"zigbee2mqtt/0x54ef44100013a0a6","last_reset_topic":"zigbee2mqtt/0x54ef44100013a0a6","last_reset_value_template":"1970-01-01T00:00:00+00:00","name":"0x54ef44100013a0a6 energy","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef44100013a0a6","unique_id":"0x54ef44100013a0a6_energy_zigbee2mqtt","unit_of_measurement":"kWh","value_template":"{{ value_json.energy }}"}'
Debug Received MQTT message on 'homeassistant/sensor/0x54ef44100013a0a6/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x54ef44100013a0a6"],"manufacturer":"Xiaomi","model":"Aqara EU smart plug (SP-EUC01)","name":"0x54ef44100013a0a6","sw_version":"Zigbee2MQTT 1.21.0-dev"},"enabled_by_default":false,"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x54ef44100013a0a6","name":"0x54ef44100013a0a6 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef44100013a0a6","unique_id":"0x54ef44100013a0a6_linkqua [... cut ...] e2mqtt","value_template":"{{ value_json.update_available}}"}'
Debug No converter available for 'SP-EUC01' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"247":{"data":[100,16,0,3,40,19,152,57,0,0,0,0,149,57,0,0,0,0,150,57,0,224,28,69,151,57,0,0,0,0,5,33,1,0,154,32,0,8,33,32,1,9,33,0,21,11,32,0,155,16,1,15,35,0,0,98,190],"type":"Buffer"}}'
Debug No converter available for 'SP-EUC01' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"519":1}'
```

@Koenkk any ideas ?

Koenkk, 2021-08-23:
@coom 

- What is logged when you press the configure button in the frontend?
- Can you try adding to `fromZigbee` `fz.xiaomi_switch_opple_basic`, save the file and restart the container with `docker restart CONTAINER_ID` (**not** via HA since this will reset the `xiaomi.js` changes)

coom, 2021-08-23:
@Koenkk here is the log when I click reconfigure :

```
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"0x54ef44100013a0a6","transaction":"bgx6q-1"}'
Debug Received Zigbee message from '0x54ef44100013a0a6', type 'readResponse', cluster 'haElectricalMeasurement', data '{"acPowerDivisor":10,"acPowerMultiplier":1}' from endpoint 1 with groupID null
Info Successfully configured '0x54ef44100013a0a6'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x54ef44100013a0a6"},"status":"ok","transaction":"bgx6q-1"}'
```

And then I can see the 2 green popup saying that everything is well configured but still no luck.

I've modified the file according to your request :

[image]

And then restarted via docker and it has been showing something like that :

[image]

I think everything was working but **value are not updating**, even if I turn off the plug. Tried to delete it and pair it again but still no luck.

However, the value is coming to Zigbee2MQTT as this line represent the actual power (checked with a multimeter and a shelly EM) : 
```
Debug Received Zigbee message from '0x54ef44100013a0a6', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":37.428001403808594}' from endpoint 21 with groupID null
Debug Received Zigbee message from '0x54ef44100013a0a6', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":43.66600036621094}' from endpoint 21 with groupID null
Debug Received Zigbee message from '0x54ef44100013a0a6', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":40.30699920654297}' from endpoint 21 with groupID null
Debug Received Zigbee message from '0x54ef44100013a0a6', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":43.66600036621094}' from endpoint 21 with groupID null
Debug Received Zigbee message from '0x54ef44100013a0a6', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":37.428001403808594}' from endpoint 21 with groupID null
```

Tried to delete the device from Zigbee2MQTT and then pair it again and now everything is stuck at 0W / 0kWh :

[image]

Thank for your time and your patience.

Koenkk, 2021-08-23:
@coom can you also add `fz.xiaomi_power` ?

coom, 2021-08-23:
@Koenkk you're awesome Dude ! Added fz.xiaomi_power and now everything works like a charm. Value is updated right away and constantly, power and energy works.

[image]

Just for a reminder is here the final line working for me (sorry for the copy/paste it's a vmware virtual console, I can't SSH into HAOS) : 

[image]

@Koenkk Is there a way like a link so I can offer you a beer or a coffee ?

Koenkk, 2021-08-23:
@coom thanks for checking, added this now. Assuming this can be closed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

@coom click the sponsor button here https://github.com/Koenkk/zigbee2mqtt (thanks already 😄 )

closed 2021-08-23 by Koenkk

ratsoid, 2021-09-01:
Works indeed  (just updated). One small thing regarding the wording in the changelog on: https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.21.1:

`#6861 Fix energy reporting for SP-EUC01 and remove temperature, voltage and current because device doesn't support it.`

I am getting the temperature in my logs (props to @Vodochnik btw)