Koenkk/zigbee2mqtt #6862: Would like to ask you to add the tunya gas detector _TZE200_ggev5fsl

nlfva1, 2021-03-28:
Hi Koen, 

I tried to add a tunya gas detector to the devices.js as shown below, but get the error message: ReferenceError: bind is not defined.

    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_ggev5fsl'}],
        model: 'TS0601_gas_sensor',
        vendor: 'TuYa',
        description: 'gas sensor',
        fromZigbee: [fz.ias_gas_alarm_1,
                fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
                fz.tuya_data_point_dump], // This is a debug converter, it will be described in the next part
        toZigbee: [tz.tuya_data_point_test], // Another debug converter
        meta: {
                configureKey: 1,
        },
        configure: async (device, coordinatorEndpoint, logger) => {
                const endpoint = device.getEndpoint(1);
                await bind(endpoint, coordinatorEndpoint, ['genBasic']);
        },
        exposes: [e.gas(), e.battery_low(), e.tamper()],
    },

and was hoping that this would be enough, but unfortunately I get the following error message:

error 2021-03-28 18:04:11: Failed to configure '0x5c0272fffef16c06', attempt 2 (ReferenceError: bind is not defined
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js:1094:10)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.js:131:45)
    at Configure.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/configure.js:113:18)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:381:44))

## Information about the device + link
https://nl.aliexpress.com/item/1005002264019210.html?spm=a2g0s.12269583.0.0.32af4a01JzoRGQ

## data/database.db entry of the device
{"id":8,"type":"Router","ieeeAddr":"0x5c0272fffef16c06","nwkAddr":60891,"manufId":4098,"manufName":"_TZE200_ggev5fsl","powerSource":"Mains (single phase)","m
odelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic
":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ggev5fsl","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"date
Code":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted"
:true,"meta":{},"lastSeen":1616949215437}

Koenkk, 2021-03-29:
Try changing `await bind(endpoint, coordinatorEndpoint, ['genBasic']);` to `await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);` (and make sure you are on zigbee2mqtt 1.18.1)

Koenkk, 2021-03-31:
> The only problem is that the alarm switch device is not created in domoticz. 

I don't know if this device support turning on the alarm remotely, to check you should check it's behaviour with the original gateway

>  When I press the test alarm button there is also no message in the zigbee2mqtt log.txt, so I'm a bit stuck.

- Does `gas` turn `true` when triggering it with gas? 
- Can you provide the debug log when pressing this button

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

nlfva1, 2021-04-02:
I don't know if gas turn true when I trigger the button. Don't know where I can see this as there is no device in Domoticz that shows the alarm.

Here is the debug log:

Turning on:
<
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,127,223,0,34] +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,127,223,0,34] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [127,223,0] - 34 +7ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":57215,"relaycount":0,"relaylist":[]} +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +11ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,30,68,129,0,0,0,239,127,223,1,1,0,123,0,152,67,19,0,0,10,9,7,1,0,4,1,4,0,1,0,127,223,29,159] +30ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,30,68,129,0,0,0,239,127,223,1,1,0,123,0,152,67,19,0,0,10,9,7,1,0,4,1,4,0,1,0,127,223,29,159] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 30 - 2 - 4 - 129 - [0,0,0,239,127,223,1,1,0,123,0,152,67,19,0,0,10,9,7,1,0,4,1,4,0,1,0,127,223,29] - 159 +8ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":57215,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":123,"securityuse":0,"timestamp":1262488,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[9,7,1,0,4,1,4,0,1,0]}} +51ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":7,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"status":0,"transid":4,"dp":1,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[0]}},"Command":{"ID":1,"parameters":[{"name":"status","type":32},{"name":"transid","type":32},{"name":"dp","type":32},{"name":"datatype","type":32},{"name":"fn","type":32},{"name":"data","type":65}],"name":"getData"}},"address":57215,"endpoint":1,"linkquality":123,"groupID":0}' +3s
Zigbee2MQTT:debug 2021-04-02 16:22:47: Received Zigbee message from '0x5c0272fffef16c06', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0
  zigbee-herdsman:controller:endpoint DefaultResponse 0x5c0272fffef16c06/1 61184(1, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +3s
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x5c0272fffef16c06:57215/1 (0,0,1) +3s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":57215,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":11,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,7,11,1,0]}} +3s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,127,223,1,1,0,239,11,0,30,5,24,7,11,1,0,96] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +91ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100,254,3,68,128,0,1,11,205] +38ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100,254,3,68,128,0,1,11,205] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +9ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,11,205] +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,11] - 205 +9ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":11} +159ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +8ms
>

Turning off:
<
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,127,223,0,34] +4s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,127,223,0,34] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [127,223,0] - 34 +5ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":57215,"relaycount":0,"relaylist":[]} +4s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,30,68,129,0,0,0,239,127,223,1,1,0,128,0,93,115,19,0,0,10,9,8,1,0,4,1,4,0,1,1,127,223,29,159] +34ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,30,68,129,0,0,0,239,127,223,1,1,0,128,0,93,115,19,0,0,10,9,8,1,0,4,1,4,0,1,1,127,223,29,159] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 30 - 2 - 4 - 129 - [0,0,0,239,127,223,1,1,0,128,0,93,115,19,0,0,10,9,8,1,0,4,1,4,0,1,1,127,223,29] - 159 +7ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":57215,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":128,"securityuse":0,"timestamp":1274717,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[9,8,1,0,4,1,4,0,1,1]}} +57ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":8,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"status":0,"transid":4,"dp":1,"datatype":4,"fn":0, [... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82 +4ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +262ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +8ms
  zigbee-herdsman:adapter:zStack:znp:log Port closed +3m
Zigbee2MQTT:info  2021-04-02 16:22:56: zigbee-herdsman stopped
Zigbee2MQTT:info  2021-04-02 16:22:56: Stopped Zigbee2MQTT

>

Hope this helps.

Koenkk, 2021-04-04:
Try with the changes from https://github.com/Koenkk/zigbee-herdsman-converters/pull/2447/files , then gas true/false should be published.

nlfva1, 2021-04-04:
Hi Koen,

Thanks for your help so far. I tried the new info and added the code, but unfortunately not working.

In the log I get the following messages.

info  2021-04-04 16:14:23: Zigbee: allowing new devices to join.
info  2021-04-04 16:14:24: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"b6085fce","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.18.1"}'
info  2021-04-04 16:14:26: Device '0x5c0272fffef16c06' joined
info  2021-04-04 16:14:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffef16c06","ieee_address":"0x5c0272fffef16c06"},"type":"device_joined"}'
info  2021-04-04 16:14:26: Starting interview of '0x5c0272fffef16c06'
info  2021-04-04 16:14:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffef16c06","ieee_address":"0x5c0272fffef16c06","status":"started"},"type":"device_interview"}'
info  2021-04-04 16:14:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x5c0272fffef16c06"},"type":"device_connected"}'
info  2021-04-04 16:14:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x5c0272fffef16c06"},"type":"pairing"}'

info  2021-04-04 16:14:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffef16c06","ieee_address":"0x5c0272fffef16c06"},"type":"device_announce"}'
info  2021-04-04 16:14:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x5c0272fffef16c06"},"type":"device_announced"}'
info  2021-04-04 16:14:41: Successfully interviewed '0x5c0272fffef16c06', device has successfully been paired
info  2021-04-04 16:14:41: Device '0x5c0272fffef16c06' is supported, identified as: TuYa gas sensor (TS0601_gas_sensor)
info  2021-04-04 16:14:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"gas sensor","exposes":[{"access":1,"description":"Indicates whether the device detected gas","name":"gas","property":"gas","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0601_gas_sensor","supports_ota":false,"vendor":"TuYa"},"friendly_name":"0x5c0272fffef16c06","ieee_address":"0x5c0272fffef16c06","status":"successful","supported":true},"type":"device_interview"}'
info  2021-04-04 16:14:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"gas sensor","friendly_name":"0x5c0272fffef16c06","model":"TS0601_gas_sensor","supported":true,"vendor":"TuYa"},"type":"pairing"}'
info  2021-04-04 16:14:41: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"dateCode":"20190608","friendly_name":"Coordinator","ieeeAddr":"0x00124b001ccdfbf7","lastSeen":1617545681867,"networkAddress":0,"softwareBuildID":"zStack12","type":"Coordinator"},{"dateCode":"20190308","description":"TRADFRI motion sensor","friendly_name":"Bewegingsensor - Ikea","hardwareVersion":1,"ieeeAddr":"0xccccccfffea6f001","lastSeen":1617545666204,"manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1525/E1745","modelID":"TRADFRI motion sensor","networkAddress":29111,"powerSource":"Battery","softwareBuildID":"2.0.022","type":"EndDevice","vendor":"IKEA"},{"dateCode":"20191205","description":"Aqara temperature, humidity and pressure sensor","friendly_name":"Aquara 1 - binnen","hardwareVersion":30,"ieeeAddr":"0x00158d00053f5f2c","lastSeen":1617544057555,"manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","modelID":"lumi.weather","networkAddress":9027,"powerSource":"Battery","softwareBuildID":"3000-0001","type":"EndDevice","vendor":"Xiaomi"},{"dateCode":"20191205","description":"Aqara temperature, humidity and pressure sensor","friendly_name":"Aquara 2 - zijdeur","hardwareVersion":30,"ieeeAddr":"0x00158d00045acf33","lastSeen":1617544428168,"manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","modelID":"lumi.weather","networkAddress":36480,"powerSource":"Battery","softwareBuildID":"3000-0001","type":"EndDevice","vendor":"Xiaomi"},{"dateCode":"20191205","description":"Aqara temperature, humidity and pressure sensor","friendly_name":"Achterdeur - Binnen","hardwareVersion":30,"ieeeAddr":"0x00158d000659e3eb","lastSeen":1617544489137,"manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","modelID":"lumi.weather","networkAddress":38568,"powerSource":"Battery","softwareBuildID":"3000-0001","type":"EndDevice","vendor":"Xiaomi"},{"dateCode":"20191205","description":"Aqara temperature, humidity and pressure sensor","friendly_name":"Achterdeur - Buiten","hardwareVersion":30,"ieeeAddr":"0x00158d000484fa23","lastSeen":1617544523382,"manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","modelID":"lumi.weather","networkAddress":29072,"powerSource":"Battery","softwareBuildID":"3000-0001","type":"EndDevice","vendor":"Xiaomi"},{"dateCode":"20131206","description":"Zigbee LED Controller W","friendly_name":"0x00124b0021f1abc0","hardwareVersion":1,"ieeeAddr":"0x00124b0021f1abc0","lastSeen":16160574 [... cut ...]
warn  2021-04-04 16:15:02: zigbee-herdsman-converters:tuya_gas: Unrecognized DP #undefined with data {"type":"Buffer","data":[9,43,17,0,33,64]}
warn  2021-04-04 16:15:02: zigbee-herdsman-converters:tuya_gas: Unrecognized DP #undefined with data {"type":"Buffer","data":[9,43,17,0,33,64]}

In Domoticz it created only a device with link_quality, but not one for gas.

Zigbee2MQTT | 0x5c0272fffef16c06_signal | 43 | Zigbee2MQTT - 0x5c0272fffef16c06 (Link Quality) | General | Custom Sensor|0

Hope you have enough info to find the issue.

Koenkk, 2021-04-06:
Can you provide the debug log when triggering the sensor?
See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

nlfva1, 2021-04-07:
Here is the debug log when triggering the sensor:

**Turning on:**
<
zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,127,223,0,34] +2s
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,127,223,0,34] +3ms
zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [127,223,0] - 34 +7ms
zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":57215,"relaycount":0,"relaylist":[]} +2s
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +11ms
zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,30,68,129,0,0,0,239,127,223,1,1,0,123,0,152,67,19,0,0,10,9,7,1,0,4,1,4,0,1,0,127,223,29,159] +30ms
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,30,68,129,0,0,0,239,127,223,1,1,0,123,0,152,67,19,0,0,10,9,7,1,0,4,1,4,0,1,0,127,223,29,159] +3ms
zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 30 - 2 - 4 - 129 - [0,0,0,239,127,223,1,1,0,123,0,152,67,19,0,0,10,9,7,1,0,4,1,4,0,1,0,127,223,29] - 159 +8ms
zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":57215,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":123,"securityuse":0,"timestamp":1262488,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[9,7,1,0,4,1,4,0,1,0]}} +51ms
zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":7,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"status":0,"transid":4,"dp":1,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[0]}},"Command":{"ID":1,"parameters":[{"name":"status","type":32},{"name":"transid","type":32},{"name":"dp","type":32},{"name":"datatype","type":32},{"name":"fn","type":32},{"name":"data","type":65}],"name":"getData"}},"address":57215,"endpoint":1,"linkquality":123,"groupID":0}' +3s
Zigbee2MQTT:debug 2021-04-02 16:22:47: Received Zigbee message from '0x5c0272fffef16c06', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0
zigbee-herdsman:controller:endpoint DefaultResponse 0x5c0272fffef16c06/1 61184(1, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +3s
zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x5c0272fffef16c06:57215/1 (0,0,1) +3s
zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":57215,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":11,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,7,11,1,0]}} +3s
zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,127,223,1,1,0,239,11,0,30,5,24,7,11,1,0,96] +3s
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +91ms
zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100,254,3,68,128,0,1,11,205] +38ms
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100,254,3,68,128,0,1,11,205] +3ms
zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +9ms
zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +3s
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,11,205] +8ms
zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,11] - 205 +9ms
zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":11} +159ms
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +8ms

**Turning off:**
<
zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,127,223,0,34] +4s
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,127,223,0,34] +4ms
zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [127,223,0] - 34 +5ms
zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":57215,"relaycount":0,"relaylist":[]} +4s
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +9ms
zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,30,68,129,0,0,0,239,127,223,1,1,0,128,0,93,115,19,0,0,10,9,8,1,0,4,1,4,0,1,1,127,223,29,159] +34ms
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,30,68,129,0,0,0,239,127,223,1,1,0,128,0,93,115,19,0,0,10,9,8,1,0,4,1,4,0,1,1,127,223,29,159] +5ms
zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 30 - 2 - 4 - 129 - [0,0,0,239,127,223,1,1,0,128,0,93,115,19,0,0,10,9,8,1,0,4,1,4,0,1,1,127,223,29] - 159 +7ms
zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":57215,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":128,"securityuse":0,"timestamp":1274717,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[9,8,1,0,4,1,4,0,1,1]}} +57ms
zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":8,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"status":0,"transid":4,"dp":1,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[1]}},"Command":{"ID":1,"parameters":[{"name":"status","type":32},{"name":"transid","type":32},{"name":"dp","type":32},{"name":"datatype","type":32} [... cut ...]
zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +4s
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +7ms
zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,12,202] +6ms
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,12,202] +2ms
zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,12] - 202 +6ms
zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":12} +109ms
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms

nlfva1, 2021-04-10:
Hi Koen, Here is the full debug log as you requested.

Op do 8 apr. 2021 om 16:46 schreef Frank van der Vaart <frank.vandervaart@gmail.com>:
DEBUG=zigbee-herdsman* npm start
> zigbee2mqtt@1.18.1 start /opt/zigbee2mqtt> node index.js
Zigbee2MQTT:debug 2021-04-08 16:40:57: Loaded state from file /opt/zigbee2mqtt/data/state.jsonZigbee2MQTT:info  2021-04-08 16:40:57: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-04-08.16-40-45' filename: log.txtZigbee2MQTT:info  2021-04-08 16:40:59: Starting Zigbee2MQTT version 1.18.1 (commit #b6085fce)Zigbee2MQTT:info  2021-04-08 16:40:59: Starting zigbee-herdsman (0.13.71)Zigbee2MQTT:debug 2021-04-08 16:40:59: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB-Zigbee2MQTT"}}'  zigbee-herdsman:adapter Path '/dev/ttyUSB-Zigbee2MQTT' is valid for 'ZStackAdapter' +0ms  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[209,146,24,201,133,219,92,13,251,245,167,43,20,103,86,237],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB-Zigbee2MQTT"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}' +0ms  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB-Zigbee2MQTT and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +78ms  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +514ms  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,121,1,26] +0ms  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,121,1,26] +10ms  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [121,1] - 26 +11ms  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":377} +0ms  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +35ms  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +128ms  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +96ms  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,0,2,6,3,144,21,52,1,2,0,0,0,0,218] +27ms  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,0,2,6,3,144,21,52,1,2,0,0,0,0,218] +4ms  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,0,2,6,3,144,21,52,1,2,0,0,0,0] - 218 +6ms  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608} +64ms  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +16ms  zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2 +0ms  zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}) +3ms  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +72ms  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37] +80ms  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +70ms  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +5ms  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +4ms  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +89ms  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +8ms  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +0ms  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +75ms  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37] +69ms  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +49ms  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +3ms  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +5ms  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +65ms  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +10ms  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +67ms  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":132,"len":4,"offset":0,"value":{"type":"Buffer","data":[0,8,0,0]}} +66ms  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,132,0,0,174] +61ms  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,97,8,0,4,0,8,0,0,99] +28ms  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,6,97,8,0,4,0,8,0,0,99] +3ms  zigbee-herdsman:adapter:zStack:unpi:p [... cut ...] zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82] +3ms  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82 +4ms  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +259ms  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +9ms  zigbee-herdsman:adapter:zStack:znp:log Port closed +2mZigbee2MQTT:info  2021-04-08 16:43:29: zigbee-herdsman stoppedZigbee2MQTT:info  2021-04-08 16:43:29: Stopped Zigbee2MQTT

Koenkk, 2021-04-10:
Updated https://github.com/Koenkk/zigbee-herdsman-converters/pull/2447/files, can you check again?

Next time please post your logs on pastebin.com

nlfva1, 2021-04-11:
Hi Koen,

Good news. The response in the zigbee2mqtt log looks much better now:
info  2021-04-11 12:07:21: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffef16c06', payload '{"gas":true,"linkquality":113}'info  2021-04-11 12:07:25: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffef16c06', payload '{"gas":false,"linkquality":134}'

The only issue at the moment is that on successful pairing the "gas" device is not created. Only the linkquality has been created and also updated with the values from the device. Can you assist with the final step to get the proper device created?

Koenkk, 2021-04-11:
From z2m perspective it looks good now, the discovery is done by https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues so if it doesn't work after the update to the latest dev branch please create an issue there.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-04-11 by Koenkk

nlfva1, 2021-04-12:
Working like a charm. Many thanks for your help and support.