Koenkk/zigbee2mqtt #6867: Tuya compatible Skydance 5-in-1 LED Controller WZ5 (TS0601)

blazewicz, 2021-03-29:
## Information about the device + link

*ZigBee & RF 5 in1 LED Controller WZ5* manufactured by *SKYDANCE*

[image]

Product page: http://www.iskydance.com/index.php?c=product_show&a=index&id=1584

This device can be reconfigured to one of 5 modes, each is discovered as different device and have different capabilities.

| # | mode    | modelId | manufacturerName | supported dataPoints
|---|---------|---------|------------------|---------------------
| 1 | DIM     | TS0601  |                  | 
| 2 | CCT     | TS0601  | _TZE200_gz3n0tzf | 1, 3, 4
| 3 | RGB     | TS0601  |                  | 
| 4 | RGBW    | TS0601  |                  | 
| 5 | RGB+CCT | TS0601  |                  | 

## data/database.db entry of the device

Entry for "CCT" Mode

```json
{
    "id": 24,
    "type": "Router",
    "ieeeAddr": "0x60a423fffe043994",
    "nwkAddr": 25068,
    "manufId": 4098,
    "manufName": "_TZE200_gz3n0tzf",
    "powerSource": "Mains (single phase)",
    "modelId": "TS0601",
    "epList": [
        1
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 81,
            "inClusterList": [
                0,
                4,
                5,
                61184
            ],
            "outClusterList": [
                25,
                10
            ],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "TS0601",
                        "manufacturerName": "_TZE200_gz3n0tzf",
                        "powerSource": 1,
                        "zclVersion": 3,
                        "appVersion": 66,
                        "stackVersion": 0,
                        "hwVersion": 1,
                        "dateCode": ""
                    }
                }
            },
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "appVersion": 66,
    "stackVersion": 0,
    "hwVersion": 1,
    "dateCode": "",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta": {
        "configured": 2
    },
    "lastSeen": 1617004825542
}
```

## Implementation attempt

Currently I've managed to find and test following capabilities:

| dp/mode | DIM | CCT                    | RGB | RGBW | RGB+CCT
|---------|-----|------------------------|-----|------|--------
| 1       |     | `sate` (0,1)           |     |      |
| 3       |     | `brightness` (0..1000) |     |      |
| 4       |     | `color_temp` (0..1000) |     |      |

Code added:

```javascript
// devices.js
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_gz3n0tzf'}],
        model: 'TS0601_led_cct',
        vendor: 'TuYa',
        description: 'Zigbee smart LED driver (CCT)',
        extend: preset.light_onoff_brightness_colortemp(),
        fromZigbee: [
            fz.tuya_dimmer,
        ],
        toZigbee: [
            tz.tuya_dimmer_state,
            tz.tuya_dimmer_level,
            tz.tuya_dimmer_color_temp,
        ],
        exposes: [
            e.light_brightness_colortemp(),
        ]
    }
```

```javascript
// converters/toZigbee.js
    tuya_dimmer_color_temp: {
        key: ['color_temp'],
        convertSet: async (entity, key, value, meta) => {
            const dp = 4;  // TODO add new enum

            if (typeof value === 'string' && isNaN(value)) {
                // TODO: add presets 'coldest', 'cold', 'neutral', 'warm', 'warmest'
                throw new Error(`Unknown preset '${value}'`);
            }

            value = Number(value);
            if (value < 250 || value > 454) {
                throw new Error('Color temperature is out of range 250..454');
            }
            // convert range (454..250) -> (0..1000)
            value = Math.round((454 - value)/204 * 1000);

            // Always use same transid as tuya_dimmer_state (https://github.com/Koenkk/zigbee2mqtt/issues/6366)
            await tuya.sendDataPointValue(entity, dp, value, 'setData', 1);
        }
    },
```

Its my first time adding new device so any help or advice is much appreciated.

blazewicz, 2021-03-29:
I've tried contacting the manufacturer via [Contact Us](http://www.iskydance.com/index.php?c=content&a=index&id=69) page, lets see what we can get from them.

blazewicz, 2021-03-31:
By experimenting I found that it uses same color encoding and `dataPoint` as [Lidl HG06467](https://www.zigbee2mqtt.io/devices/HG06467.html) (impl: [toZigbee](https://github.com/Koenkk/zigbee-herdsman-converters/blob/04b6bbced058003eb7e2cf3b7a67f06ded7f3d3d/converters/toZigbee.js#L4349), [fromZigbee](https://github.com/Koenkk/zigbee-herdsman-converters/blob/04b6bbced058003eb7e2cf3b7a67f06ded7f3d3d/converters/fromZigbee.js#L2304)).

blazewicz, 2021-04-07:
This device appears to be straight implementation of Tuya generic RGBCW light source, all functions are directly from the creator that can be found [here](https://iot.tuya.com) after signing up and starting new project (see screenshot below).

Currently I'm working on generic implementation of Tuya specific protocol for light sources.

[image]

closed 2021-05-15 by github-actions[bot]

tbornon, 2021-10-01:
Hi,
I'm really interested by your work. Have you been able to successfully make it work with Zigbee2MQTT ?
Regards