Koenkk/zigbee2mqtt #6870: Problem with securing network config

stoinov, 2021-03-29:
## What happened
I've moved my network settings to secrets file and I got errors when starting:
```
- advanced.pan_id: should be number or 'GENERATE' (is '!secret pan_id')
- advanced.ext_pan_id should be array
- advanced.channel should be number
```
## What did you expect to happen
To start the service normally as [the help shows](https://www.zigbee2mqtt.io/information/configuration.html#specifying-mqtt-userpassword-and-network_key-in-a-different-file) the use of secrets for the `network_key` which is array.

## How to reproduce it (minimal and precise)
Move your described network settings to a secrets file like this:
```
  pan_id: '!secret pan_id'
  ext_pan_id: '!secret ext_pan_id'
  channel: '!secret channel'
```

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: zzh
Adapter firmware version: 20210120

Koenkk, 2021-03-29:
`!secret` only works for MQTT username password and the network key, the rest is not secret since it can be easily obtained by sniffing a network.

stoinov, 2021-03-30:
I was trying to use is as a config file with specific settings for different setups, but it only seems to work for entries requiring string as the data type as this works for `client_id` too.
I guess there is an exception for `network_key`.

closed 2021-03-30 by stoinov