Koenkk/zigbee2mqtt #6871: independent brightness controll for each of the RGBW channels on gledopto led controller

kovaga, 2021-03-29:
## Is your feature request related to a problem? Please describe
Inability to independently control brightness of each of the RGBW channels on gledopto led controllers.

## Describe the solution you'd like
At the moment it is possible to have two independently controlled brightness channels, one for W and one for RGB on GL-C-007-2ID

I am thinking of connecting 4 white only LEDs and controlling the brightness independently, similar to what shelly LED RGBW2 have available in their "4X White LED strip scheme" scenario.

Do you think it is possible to implement such scenarion in z2m?

Koenkk, 2021-03-29:
AFAIK these devices don't support that.

Koenkk, 2021-03-30:
It is something that has to be supported by the firmware indeed (wouldn't call it an issue).

closed 2021-03-30 by Koenkk

kovaga, 2021-03-31:
It would be simply nice, being able to control the brightness of each of the channel individually in 1 controller, rather then buying 4 :)

I have experimented a bit, and one of the possibilities is setting the brightness_rgb to maximum, then adjusting each of the RGB from 0 to 254 would adjust the voltage on each of the channel making the light brighter or dimmer.
Unfortunately it is not that simple and straightforward, as in certain combinations of RGB it also changes the voltage on the adjacent channels.

So i guess with this device  one can accurately control the brightness of at least two different LED strips, one by using the W channel and a separate brightness control, and then using one of the other colors, i.e. setting R to 254 and then adjusting the brightness using the brightness_rgb.