Koenkk/zigbee2mqtt #6872: How to get device VOCKQJK11LM binding to Z2M

zrainx, 2021-03-29:
Bug Report
What happened
I have bought a new device VOCKQJK11LM from www.aqara.com. I tried to support this device in my Z2M. It seemed the device has successfully been paired but I can't receive any Zigbee message of "attributeReport" type from it.

The package shows it's a Zigbee 3.0 device.

In an other zigbee adapter I received following log:
{"code":1002,"id": "010154ef44100001abe5", "ep": 1,"serial": 12004,"control":{"devbindreq":"0001mtemp","bindDstEndpoin":1,"bindDstAddr":"680ae2fffe0694bc"}}
{"code":1002,"id": "010154ef44100001abe5", "ep": 1,"serial": 12004,"control":{"devbindreq":"0001mhumi","bindDstEndpoin":1,"bindDstAddr":"680ae2fffe0694bc"}}
After the zigbee adapter bound the VOC device, I can receive payload message from the device.
I am not sure whether this device bind request is skipped in Z2M. And in zigbee2mqtt.io I can only find binding information between devices.

Or there was something I didn't aware to do to support the VOC device.

What did you expect to happen
I can receive some Zigbee message of "attributeReport" type from the new VOCKQJK11LM device, so I can support this new device in my Z2M

How to reproduce it (minimal and precise)
Just in What happened

Debug Info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC2538
Adapter firmware version: ’20200211‘

Koenkk, 2021-03-29:
Can you share the data/database.db entry of this device?

zrainx, 2021-03-30:
the db entry of this device：

{"id":6,"type":"EndDevice","ieeeAddr":"0x54ef44100001abe5","nwkAddr":65335,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.airmonitor.acn01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,$

Koenkk, 2021-03-30:
I see it supports the temperature (1026) and humidity (1029) cluster, but not any cluster supporting VOC measurements. Probably they use something manufacturer specific. To support this I would need to have a sniff when pairing it with the Xiaomi gateway: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

zrainx, 2021-04-01:
[pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/6244526/pcapng.zip)

Hi Koenkk ,
This is the log sniffed when it pairing with gateway. 
I'm not very sure the log is caught correctly or not.
Thanks a lot!

Koenkk, 2021-04-01:
Thanks, can you now make the sensor report some values? (and sniff that)

zrainx, 2021-04-02:
Hi Koenkk ,
Unfortunatly, The sensor VOCKQJK11LM doesn't report any value to z2m when I update your new commit. 
But I have found something interesting. I have a shunzhou gateway（not xiaomi gateway）,  and I find that, the sensor device will report  value after it receives the bind request from shunzhou gateway.
[image]

The content of the bind request is as below:

{"code":1002,"id": "010154ef44100001abe5", "ep": 1,"serial": 12004,"control":{"devbindreq":"0001mtemp","bindDstEndpoin":1,"bindDstAddr":"680ae2fffe0694bc"}}
{"code":1002,"id": "010154ef44100001abe5", "ep": 1,"serial": 12004,"control":{"devbindreq":"0001mhumi","bindDstEndpoin":1,"bindDstAddr":"680ae2fffe0694bc"}}

Can z2m submit bind request like that?

Thanks a lot.

Koenkk, 2021-04-04:
Yes, can you try with https://github.com/Koenkk/zigbee-herdsman-converters/pull/2448/files

zrainx, 2021-04-06:
Hi Koenkk ,
     I‘ve tried with commit 2448，I’m glad to find it reports  temperature and humidity to the z2m gateway, But it doesn't report VOC value.  
[image]

Yesterday, I bought a new xiaomi gateway and try again. I found the sensor VOCKQJK11LM do report VOC to the xiaomi gateway.  So I re-catch a sniffer log when it was binding to the new xiaomi gateway.

The sniffer log is as below:
[0406-xiaomi.zip](https://github.com/Koenkk/zigbee2mqtt/files/6262880/0406-xiaomi.zip)

Thank you.

Koenkk, 2021-04-06:
Can you share the network key of your Xiaomi gateway?

dasfuu, 2021-06-24:
@Koenkk 

I got it to work - kinda. I don't think i got battery/power and the configure part correct. I manually added the reporting stuff through the UI. 
And there are two batteries in there so I think the voltage for the calculation might be wrong, or it shipped with very empty batteries.

[image]
[image]

`// Import the regular converter package
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');
// Set fz (also done in the original device list)
const fz = zigbeeHerdsmanConverters.fromZigbeeConverters;
// Set e (also done in the original device list)
const e = zigbeeHerdsmanConverters.exposes.presets;

const fzn = {
    avoc: {
        cluster: 'genAnalogInput',
        type: 'attributeReport',
        convert: (model, msg, publish, options, meta) => {
            return {
                voc:  msg.data.presentValue
            };
        },
    }
};

const definition = {
    zigbeeModel: ['lumi.airmonitor.acn01'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'AAQS-S01', // Vendor model number, look on the device for a model number
    vendor: 'Xiaomi', // Vendor of the device (only used for documentation and startup logging)
    description: 'TVOC Air Quality Monitor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [
      fzn.avoc,
      fz.xiaomi_battery,
      fz.battery, 
      fz.temperature, 
      fz.humidity
    ], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    //meta: {battery: {voltageToPercentage: '3V_2100'}},
    meta: {configureKey: 2, battery: {voltageToPercentage: '3V_2100'}},    
    exposes: [e.battery(), e.temperature(), e.humidity(), e.voc()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            const binds = ['msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg'];
            await reporting.bind(endpoint, coordinatorEndpoint, binds);
            await reporting.batteryVoltage(endpoint);
            await reporting.batteryPercentageRemaining(endpoint);
            device.powerSource = 'Battery';
            device.save();
        },
};

module.exports = definition;`

Koenkk, 2021-07-01:
@dasfuu looks good, added this config now. can this be closed?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

dasfuu, 2021-07-01:
@Koenkk there is a problem with this. The configure part didn't work for me.

Koenkk, 2021-07-01:
@dasfuu does it also work without? If you put the same values in the `configure()` method it should work (the same code gets called)

dasfuu, 2021-07-01:
I got the following in my log:

```
Jul 01 19:12:42 zigbee npm[5986]: Zigbee2MQTT:error 2021-07-01 19:12:42: Failed to configure 'Aqara_Air_Quality_1', attempt 1 (ReferenceError: reporting is not defined
Jul 01 19:12:42 zigbee npm[5986]:     at Object.configure (evalmachine.<anonymous>:39:13)
Jul 01 19:12:42 zigbee npm[5986]:     at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.js:131:45)
Jul 01 19:12:42 zigbee npm[5986]:     at Configure.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/configure.js:113:18)
Jul 01 19:12:42 zigbee npm[5986]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:385:44)
Jul 01 19:12:42 zigbee npm[5986]:     at processTicksAndRejections (internal/process/task_queues.js:97:5))

```

I cannot see anything in the reporting and bind tabs. And i don't get any updates either but it updated the values on the display. When i add the entries in the ui i get updates VERY OFTEN.

See:
[image]

dasfuu, 2021-07-01:
@Koenkk Sorry for double postig. I repaired it and waited some time lone. But still nothin. As soon as i add the reporting settings it starts sending data. (Unlike the old aqara stuff)

Koenkk, 2021-07-02:
@dasfuu can you add some `console.log('TEST 123')` to the `configure()` to see where it fails?

dasfuu, 2021-07-02:
@Koenkk 

`await reporting.bind(endpoint, coordinatorEndpoint, binds);`

Seems to be the cause. I get log before but not after that line.

dasfuu, 2021-07-02:
I fixed the faulty import. now i get

```
Jul 02 19:02:17 zigbee npm[13781]: Zigbee2MQTT:error 2021-07-02 19:02:17: Failed to configure '0x54ef4410001059a1', attempt 1 (Error: Bind 0x54ef4410001059a1/1 msTemperatureMeasurement from '0x00124b0021cbec97/1' failed (AREQ - ZDO - bindRsp after 10000ms)
Jul 02 19:02:17 zigbee npm[13781]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
Jul 02 19:02:17 zigbee npm[13781]:     at listOnTimeout (internal/timers.js:549:17)
Jul 02 19:02:17 zigbee npm[13781]:     at processTimers (internal/timers.js:492:7))
Jul 02 19:02:17 zigbee npm[13781]: Zigbee2MQTT:info  2021-07-02 19:02:17: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x54ef4410001059a1"},"error":"Failed to configure (Bind 0x54ef4410001059a1/1 msTemperatureMeasurement from '0x00124b0021cbec97/1' failed (AREQ - ZDO - bindRsp after 10000ms))","status":"error","transaction":"9s9ji-2"}'
```

Still fails at

` await reporting.bind(endpoint, coordinatorEndpoint, binds);`

Koenkk, 2021-07-03:
`const reporting = zigbeeHerdsmanConverters.reporting;` is wrong, should be: `const reporting = require('zigbee-herdsman-converters/lib/reporting');
`

dasfuu, 2021-07-03:
@Koenkk I fixed tgat and then i got the error in my last comment.

Koenkk, 2021-07-03:
@dasfuu 

- Either the device does not support binding (check if it works without having:
```
const binds = ['msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg', 'genAnalogInput'];
await reporting.bind(endpoint, coordinatorEndpoint, binds);
```
- OR make sure the device is awake when these commands are executed (press a button on it to wake it up). You can force an execution on the configure via the frontend by clicking on the yellow refresh icon

dasfuu, 2021-07-03:
@Koenkk I will try that. But why does it work manually through the UI? Although I only create reportings there. The binding sturff shows up automatically then.

dasfuu, 2021-07-03:
@Koenkk the configure works. But the device doesn't report anything then. If I reapply the reportings in the UI it starts working.
After that there is an entry in the binding tab too:

[image]

Koenkk, 2021-07-03:
Ah, binding is indeed done automatically, can you try readding the bind and make sure to wake it up when configure is executed?

dasfuu, 2021-07-03:
@Koenkk It does not work. Same error. The device has only one button (double tap to switch display mode, hold to reset/pair)

Koenkk, 2021-07-03:
Can you provide 2 separate herdsman debug logs with the following content (in both situations click a button on the device right before):
1. When setting up configure reporting via the frontend
2. When executing the configure from the frontend

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

blakadder, 2021-07-08:
I managed to get all 3 reporting when manually configuring report in webui. For each time I had to press the button on the device once or the request would time out.

If it helps, here are Tasmota logs, it binds successfully on first pairing without needing to keep the device awake. Battery 
```
01:14:22.067 ZIG: {"ZbZNPReceived":"653600"}
01:14:22.073 ZIG: {"ZbZNPReceived":"45CB3C"}
01:14:22.078 MQT: tele/zigbee-gateway/RESULT = {"ZbState":{"Status":21,"Message":"Enable Pairing mode for 60 seconds"}}
01:14:22.084 ZIG: {"ZbZNPReceived":"45B6000000"}
01:14:26.061 ZIG: {"ZbZNPReceived":"45CAF13E1E6C10001044EF540000"}
01:14:26.067 MQT: tele/zigbee-gateway/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x54EF441000106C1E","ShortAddr":"0x3EF1","ParentNetwork":"0x0000"}}
01:14:26.630 ZIG: {"ZbZNPReceived":"45C1F13EF13E1E6C10001044EF5484"}
01:14:26.636 MQT: tele/zigbee-gateway/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x54EF441000106C1E","ShortAddr":"0x3EF1","PowerSource":true,"ReceiveWhenIdle":false,"Security":false}}
01:14:26.643 ZIG: ZbZNPSent 2505F13EF13E
01:14:26.653 ZIG: {"ZbZNPReceived":"650500"}
01:14:26.753 WIF: Checking connection...
01:14:27.026 ZIG: {"ZbZNPReceived":"4585F13E00F13E0101"}
01:14:27.032 MQT: tele/zigbee-gateway/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01"]}}
01:14:27.038 ZigbeeZCLSend device: 0x3EF1, group: 0x0000, endpoint:1, cluster:0x0000, cmd:0x00, send:"05000400"
01:14:27.042 ZIG: ZbZNPSent 240202F13E00000000000001000001000001301E070000010005000400
01:14:27.053 ZIG: {"ZbZNPReceived":"640200"}
01:14:27.509 ZIG: {"ZbZNPReceived":"4480F00101"}
01:14:27.514 MQT: tele/zigbee-gateway/RESULT = {"ZbConfirm":{"Endpoint":1,"Status":240,"StatusMessage":""}}
01:14:28.109 ZIG: Zigbee Devices Data saved in Flash (361 bytes)
01:14:28.163 ZIG: Zigbee Devices Data store in Flash (0x402FF800 - 361 bytes)
01:14:28.570 ZIG: ZbZNPSent 2504F13EF13E01
01:14:28.580 ZIG: {"ZbZNPReceived":"650400"}
01:14:29.032 ZIG: {"ZbZNPReceived":"4584F13E00F13E160104010203010600000300020405040C000100011900"}
01:14:29.039 MQT: tele/zigbee-gateway/RESULT = {"ZbState":{"Status":33,"Device":"0x3EF1","Endpoint":"0x01","ProfileId":"0x0104","DeviceId":"0x0302","DeviceVersion":1,"InClusters":["0x0000","0x0003","0x0402","0x0405","0x000C","0x0001"],"OutClusters":["0x0019"]}}
01:14:31.085 ZIG: auto-bind `ZbBind {"Device":"0x3EF1","Endpoint":1,"Cluster":"0x0001"}`
01:14:31.087 ZIG: ZbZNPSent 2521F13E1E6C10001044EF540101000338FB381D004B120001
01:14:31.099 ZIG: {"ZbZNPReceived":"652100"}
01:14:31.457 WIF: Sending Gratuitous ARP
01:14:33.065 ZIG: auto-bind `ZbBind {"Device":"0x3EF1","Endpoint":1,"Cluster":"0x0402"}`
01:14:33.068 ZIG: ZbZNPSent 2521F13E1E6C10001044EF540102040338FB381D004B120001
01:14:33.078 ZIG: {"ZbZNPReceived":"652100"}
01:14:33.574 ZIG: {"ZbZNPReceived":"45A1F13E00"}
01:14:33.580 MQT: tele/zigbee-gateway/RESULT = {"ZbBind":{"Device":"0x3EF1","Status":0,"StatusMessage":"SUCCESS"}}
01:14:33.791 ZIG: {"ZbZNPReceived":"45A1F13E00"}
01:14:33.801 MQT: tele/zigbee-gateway/RESULT = {"ZbBind":{"Device":"0x3EF1","Status":0,"StatusMessage":"SUCCESS"}}
01:14:35.072 ZIG: auto-bind `ZbBind {"Device":"0x3EF1","Endpoint":1,"Cluster":"0x0405"}`
01:14:35.075 ZIG: ZbZNPSent 2521F13E1E6C10001044EF540105040338FB381D004B120001
01:14:35.088 ZIG: {"ZbZNPReceived":"652100"}
01:14:35.185 ZIG: {"ZbZNPReceived":"448100000100F13E0101009C001AC20300000718000A2000201EF13E1D"}
01:14:35.189 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0001","srcaddr":"0x3EF1","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":156,"securityuse":0,"seqnumber":0,"fc":"0x18","frametype":0,"direction":1,"disableresp":1,"manuf":"0x0000","transact":0,"cmdid":"0x0A","payload":"2000201E"}}
01:14:35.194 ZIG: ZbZCLRawReceived: {"0x3EF1":{"0001/0020":30,"Endpoint":1,"LinkQuality":156}}
01:14:35.200 ZIG: {"ZbZNPReceived":"448100000204F13E0101009C0051C20300000818010A000029910CF13E1D"}
01:14:35.204 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0402","srcaddr":"0x3EF1","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":156,"securityuse":0,"seqnumber":0,"fc":"0x18","frametype":0,"direction":1,"disableresp":1,"manuf":"0x0000","transact":1,"cmdid":"0x0A","payload":"000029910C"}}
01:14:35.209 ZIG: ZbZCLRawReceived: {"0x3EF1":{"0402/0000":3217,"Endpoint":1,"LinkQuality":156}}
01:14:35.217 ZIG: {"ZbZNPReceived":"45A1F13E00"}
01:14:35.222 MQT: tele/zigbee-gateway/RESULT = {"ZbBind":{"Device":"0x3EF1","Status":0,"StatusMessage":"SUCCESS"}}
01:14:35.572 MQT: tele/zigbee-gateway/3EF1/SENSOR = {"ZbReceived":{"0x3EF1":{"Device":"0x3EF1","BatteryVoltage":3,"BatteryPercentage":100,"Temperature":32.17,"Endpoint":1,"LinkQuality":156}}}
01:14:36.417 ZIG: {"ZbZNPReceived":"448100000504F13E0101009C0030D10300000818030A000021DD15F13E1D"}
01:14:36.421 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0405","srcaddr":"0x3EF1","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":156,"securityuse":0,"seqnumber":0,"fc":"0x18","frametype":0,"direction":1,"disableresp":1,"manuf":"0x0000","transact":3,"cmdid":"0x0A","payload":"000021DD15"}}
01:14:36.426 ZIG: ZbZCLRawReceived: {"0x3EF1":{"0405/0000":5597,"Endpoint":1,"LinkQuality":156}}
01:14:36.822 MQT: tele/zigbee-gateway/3EF1/SENSOR = {"ZbReceived":{"0x3EF1":{"Device":"0x3EF1","Humidity":55.97,"Endpoint":1,"LinkQuality":156}}}
[... cut ...]
01:14:41.503 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0405","srcaddr":"0x3EF1","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":166,"securityuse":0,"seqnumber":0,"fc":"0x18","frametype":0,"direction":1,"disableresp":1,"manuf":"0x0000","transact":4,"cmdid":"0x07","payload":"00"}}
01:14:41.507 ZIG: ZbZCLRawReceived: {"0x3EF1":{"ConfigResponse":{},"Endpoint":1,"LinkQuality":166}}
01:14:41.513 MQT: tele/zigbee-gateway/3EF1/SENSOR = {"ZbReceived":{"0x3EF1":{"Device":"0x3EF1","ConfigResponse":{},"Endpoint":1,"LinkQuality":166}}}
```

dasfuu, 2021-07-08:
I can test some more on the weekend.

danieledwardgeorgehitchcock, 2021-07-19:
Hi All,

I have managed to get the device to work with the following external converter (aqara_tvoc.js):

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const utils = require('zigbee-herdsman-converters/lib/utils')
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    xiaomi_tvoc: {
        cluster: 'genAnalogInput',
        type: 'attributeReport',
        convert: (model, msg, publish, options, meta) => {
            return {
                voc:  msg.data.presentValue
            };
        },
    }
};

const device = {
        zigbeeModel: ['lumi.airmonitor.acn01'],
        model: 'AAQS-S01',
        vendor: 'Xiaomi',
        description: 'Aqara TVOC Air Quality Monitor',
        fromZigbee: [fzLocal.xiaomi_tvoc, fz.battery, fz.temperature, fz.humidity],
        toZigbee: [],
        meta: {battery: {voltageToPercentage: '3V_2100'}},
        exposes: [e.battery(), e.temperature(), e.humidity(), e.voc()],
        configure: async (device, coordinatorEndpoint, logger) => {
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg', 'genAnalogInput']);
            await reporting.batteryVoltage(device.getEndpoint(1));
            await reporting.humidity(device.getEndpoint(1));
            await reporting.temperature(device.getEndpoint(1));
            await reporting.tvoc(device.getEndpoint(1));
        },
    };

module.exports = device;
```

And an addition to the end of reporting.js:

```javascript
    tvoc: async (endpoint, overrides) => {                                                      
        const p = payload('presentValue', 10, repInterval.HOUR, 100, overrides);                
        await endpoint.configureReporting('genAnalogInput', p);                                 
    },
```

This for me, allows the device to join, bind and report all values without error. Please test for yourselves and I will get this merged into the main repo

dasfuu, 2021-07-20:
@danieledwardgeorgehitchcock @Koenkk  
looking at the model number. there seem to be different version of this new device.
I will try the converter from @danieledwardgeorgehitchcock and report if it works with my version too. So we can add the model as well.

danieledwardgeorgehitchcock, 2021-07-21:
When I looked into it, I found that what you had used was a designation number not the model number. If you look closely at the back of the box just below top left, you should see the value that I have used (I may be wrong though)?

The value that I have used also aligns more the numbering convention that Xiaomi Aqara have used with their other sensors - see the Zigbee2MQTT wiki to confirm.

my device handler should work with your device anyway as Zigbee2MQTT only uses that field for documentation purposes.

Please let me know if this isn’t the case

blakadder, 2021-07-21:
This model scheme is used for EU markets

dasfuu, 2021-07-21:
@blakadder seems to be correct

[image]

blakadder, 2021-07-21:
It is not. While Aqara used to label their products under Mi/Xiaomi brand the situation is different now. Xiaomi has their Works with Mijia labe, any manufacturer following Miot spec will have their device added in Mi Home but they don't have to do it. 
Aqara is targeting the western markets with Aqara only Homekit enabled devices that work only with latest Aqara gateways and with their own app

danieledwardgeorgehitchcock, 2021-07-21:
Interesting!

Nonetheless, the device handler that has been merged, should work with the AAQS-S01 branded device as well as the VOCKQJK11LM one.

As mentioned above, the model is used for the documentation and image references more than anything.

Please try the device that has been pushed to the Dev branch, and let me know if you have any issues.

closed 2021-08-28 by github-actions[bot]