Koenkk/zigbee2mqtt #6873: Support for Moes Tuya Thermostat TS0601 _TZE200_cwnjrr72

SoulEater45, 2021-03-29:
Today I received a new Tuya Thermostat and tried to connect it my Homeassistant instance.
Somehow it didn't got picked up and after looking into the config, it said it wasn't supported.

I am not well versed in debugging, but I looked into the zigbee2mqtt module and found out that I could add the manufacturer name in the '/app/node_modules/zigbee-herdsman-converters/devices.js' file.
Editing the line 1647 to:
```
....
fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_cwnjrr72'}, {modelID: 'TS0601', manufacturerName: '_TZE200_ckud7u2l'}, {modelID: 'TS0601', manufacturerName: '_TZE200_ywdxldoj'}],
...
```
did enable me to control the device in the web UI of Zigbee2mqtt.
The model number on the package of this device is `HY368` and differs from `HY389RT` by one and doesn't have the RT at the end. Is that suffix added usually or does it make a difference?
Are there any missing information I need to provide for the device, since I followed the steps for adding a new device and I only had to modify the lines for an existing one.

If its fine I can also make a pull request.

Koenkk, 2021-03-29:
Added the device!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-03-29 by Koenkk

daviddewaard, 2021-10-05:
I have the same model but got nog recognised by Zigbee2mqtt.   can somebody help my out??

Scalabanana, 2021-10-06:
I also have this model and it will be detected as unsupported device. Using zigbee2mqtt version 1.18.1 in homeassistant.

>Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_cwnjrr72'