Koenkk/zigbee2mqtt #6877: Quit on SRSP - ZDO - mgmtPermitJoinReq after 6000ms

m4ci3k2, 2021-03-29:
I'm running a medium size (50 devices, 3 routers) network with Slaesh's CC2652RB coordinator. Once in a few weeks I see this error
```
npm[5344]: (node:5357) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
npm[5344]:     at Timeout.waiter.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
npm[5344]:     at ontimeout (timers.js:436:11)
npm[5344]:     at tryOnTimeout (timers.js:300:5)
npm[5344]:     at listOnTimeout (timers.js:263:5)
npm[5344]:     at Timer.processTimers (timers.js:223:10)
npm[5344]: (node:5357) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error [...]
```

What is different to #5015 is that for Slaesh's stick just restarting zigbee2mqtt solves the problem (with CC2531 is was needed to actually power down the stick). It would be nice to just exit in this case, so that systemd/docker gets a chance to start it which would fix the problem.

I'm on 1.17.1 but I've looked through the commits since then and I'm quite sure that this was not changed. I'm using this network as overall stability test so I'd rather keep the current version for a few more weeks.

Koenkk, 2021-03-30:
What FW are you using? Make sure you are using the latest: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

m4ci3k2, 2021-03-31:
It's 20201026, so apparently not latest. I'll upgrade and reopen this issue if problem persists.

closed 2021-03-31 by m4ci3k2