Koenkk/zigbee2mqtt #6878: I need support for the zigbee module of the YMF40 lock

diegolz83, 2021-03-30:
I bought the Yale YMF40 lock which is shown as supported in the zigbee2mqtt project documentation. When I try to pair it with my home assistant system using the CC2531 coordinator, the addon log says that the device is not supported, something curious because that lock is in the list of supported devices.
Attached to this message is the description of the zigbee module that my device has and the
[Log.txt](https://github.com/Koenkk/zigbee2mqtt/files/6225499/Log.txt)
[image]

 addon log.

Faustino-ai, 2021-03-30:
I am not sure if what you've shown on the picture is the yale lock... But Yale YMF40 lock is supposed to look like this:
[image]
Are you sure you bought the right thing? Can you show a picture of your real product?

diegolz83, 2021-03-30:
Yes, I'm sure.

[image]

juslex, 2021-04-12:
Same problem here with the same zigbee module. In my case the lock is the ydf 40.

juslex, 2021-04-14:
> Yes, I'm sure.
> 
> [image]
> 
> In lationiamerica the lock has a red zigbee module that is sold separately and is the one shown in the first photo sent, its brand is "ASSA ABLOY iRevo" and the website says ASSA ABLOY Yale is a Zigbee member
> 
> [image]
> 
> It´s strange that zigbe2mqtt detects the device, ends the interview but then generates an unsupported device message:
> 
> Zigbee2MQTT:info 2021-03-29 18:30:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload **'{"message":"interview_successful","meta":{"friendly_name":"0x000d6f00168a65bb","supported":false}**,"type":"pairing"}'
> Zigbee2MQTT:warn 2021-03-29 18:30:06: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'ASSA ABLOY iRevo'
> 
> [image]

Hello,

I managed to integrate manually. Please send me which version of z2m you are using. Also send me a print of the z2m front end screen of the lock, I need the zigbee model number. The screen you sent is from the zigbee2mqtt assistant, I need the original z2m front end.

diegolz83, 2021-04-14:
Sure, I'll try to send this as soon as possible.

diegolz83, 2021-04-15:
Hi @juslex 

My current zigbee2mqtt version: 1.18.1-1. I attached the screens that you asked me

[image]
[image]
[image]
[image]

Thanks and I´ll be attentive

juslex, 2021-04-15:
> Hi @juslex
> 
> My current zigbee2mqtt version: 1.18.1-1. I attached the screens that you asked me
> 
> [image]
> [image]
> [image]
> [image]
> 
> Thanks and I´ll be attentive

I need the model name to try to help you with the same solution i used

diegolz83, 2021-04-15:
What only appear on zigbee module is ASSA ABLOY iRevo Brand and the serial number shown on picture.

juslex, 2021-04-15:
I need this information:

[image]

diegolz83, 2021-04-15:
Appear empty 😔

juslex, 2021-04-15:
Delete the device removing from database in frontend. Stop zigbee2mqtt try to repair the lock more closer to the coordinator.

diegolz83, 2021-04-16:
Ok, I'll do so

diegolz83, 2021-04-16:
@juslex 

[image]

juslex, 2021-04-16:
strange this zigbee model is already on devices.js it was supposed to be recognized

diegolz83, 2021-04-16:
A new error, 
[image]

My lock was already recognize but it has errors and entities don´t work

[image]

It never changes from unlocked to locked

juslex, 2021-04-16:
Try to reconfigure it.

diegolz83, 2021-04-16:
I have eliminated and integrated it 4 times but the entities don´t respond to the commands. what can i do more?

Roger4R, 2021-05-16:
Hello, 
Can you help me please?
I have the same problem. Support status = Unsupported. Zigbee model = c700000202. Yale YDF40. 
The LQI is low (just 5) because raspberry is in the other room.

diegolz83, 2021-05-16:
I have still this problem, appear unsupported device. You can solve the LQI problem installing a zigbee router as a CC2530+CC2591 device, it has a good power and I think you would have no problem if your Raspberry is in another room.

Faustino-ai, 2021-05-17:
If you are using the latest zigbee2mqtt you might want to read the how to add support for new devices again because there are some changes. You can try to add support yourself according to the guide. If it works until the point where you receive something like no converters available, you can proceed the copy the remaining parts of the code from the existing implementation of YMF40 lock.

clferraz, 2021-06-13:
I have same problem… 😞
[image]

closed 2021-07-21 by github-actions[bot]

Gera1974, 2021-08-02:
Hi sirs,

I am facing exactly the same issue. Did you manage to find any solution for this ?

Thanks,