Koenkk/zigbee2mqtt #6886: Add a description field for all devices shown in the Frontend

vertex-github, 2021-03-30:
## Is your feature request related to a problem? Please describe
Currently, zigbee devices have a deviceId and, by default, a similar friendly name.  Changing the friendly name in the frontend, impacts the device topic on MQTT.

## Describe the solution you'd like
A clear and concise description of what you want to happen.
Id like to be able to set a friendly name, but choose to retain the deviceId for the MQTT topic.  Alternatively, Id like to be able to add a 'description' field so that I can identify each device

Koenkk, 2021-03-30:
> but choose to retain the deviceId for the MQTT topic

Note that this is already possible, both ieeeAddres and friendly_name can be used to control a device.

sjorge, 2021-03-30:
> > but choose to retain the deviceId for the MQTT topic
> 
> Note that this is already possible, both ieeeAddres and friendly_name can be used to control a device.

Correct, but it doesn't publish with the ieeeAddres.

timoline, 2021-05-03:
I like to see this feature,
Also the possibility to add a description for a group