Koenkk/zigbee2mqtt #6897: HS2IRC have no response

zhouF96, 2021-03-31:
## What happened
[HS2IRC](https://www.zigbee2mqtt.io/devices/HS2IRC.html) have no response when sending the examples described in it's  [page](https://www.zigbee2mqtt.io/devices/HS2IRC.html)
## What did you expect to happen
reveive the response as described. 
## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:1.18.1
Adapter hardware:  CC1352P-2
Adapter firmware version:Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'

## log
<pre><font color="#3465A4">Zigbee2MQTT:debug</font> 2021-03-31 17:21:53: Publishing &apos;set&apos; &apos;get_list&apos; to &apos;0x000d6f001619269c&apos;
  <font color="#FFD700"><b>zigbee-herdsman:controller:endpoint </b></font>Command 0x000d6f001619269c/1 heimanSpecificInfraRedRemote.getIdAndKeyCodeList({}, {&quot;timeout&quot;:10000,&quot;disableResponse&quot;:false,&quot;disableRecovery&quot;:false,&quot;disableDefaultResponse&quot;:false,&quot;direction&quot;:0,&quot;srcEndpoint&quot;:null,&quot;reservedBits&quot;:0,&quot;manufacturerCode&quot;:4619,&quot;transactionSequenceNumber&quot;:null,&quot;writeUndiv&quot;:false}) <font color="#FFD700">+4m</font>
  <font color="#005FFF"><b>zigbee-herdsman:adapter:zStack:adapter </b></font>sendZclFrameToEndpointInternal 0x000d6f001619269c:31027/1 (0,0,1) <font color="#005FFF">+3m</font>
  <font color="#FF5FAF"><b>zigbee-herdsman:adapter:zStack:znp:SREQ </b></font>--&gt; AF - dataRequest - {&quot;dstaddr&quot;:31027,&quot;destendpoint&quot;:1,&quot;srcendpoint&quot;:1,&quot;clusterid&quot;:64642,&quot;transid&quot;:9,&quot;options&quot;:0,&quot;radius&quot;:30,&quot;len&quot;:5,&quot;data&quot;:{&quot;type&quot;:&quot;Buffer&quot;,&quot;data&quot;:[5,11,18,6,246]}} <font color="#FF5FAF">+37s</font>
  <font color="#0087FF"><b>zigbee-herdsman:adapter:zStack:unpi:writer </b></font>--&gt; frame [254,15,36,1,51,121,1,1,130,252,9,0,30,5,5,11,18,6,246,224] <font color="#0087FF">+37s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>&lt;-- [254,1,100,1,0,100] <font color="#8700FF">+37s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [254,1,100,1,0,100] <font color="#8700FF">+0ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--&gt; parsed 1 - 3 - 4 - 1 - [0] - 100 <font color="#8700FF">+1ms</font>
  <font color="#5F00D7"><b>zigbee-herdsman:adapter:zStack:znp:SRSP </b></font>&lt;-- AF - dataRequest - {&quot;status&quot;:0} <font color="#5F00D7">+37s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [] <font color="#8700FF">+0ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>&lt;-- [254,3,68,128,240,1,9,63] <font color="#8700FF">+6s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [254,3,68,128,240,1,9,63] <font color="#8700FF">+0ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--&gt; parsed 3 - 2 - 4 - 128 - [240,1,9] - 63 <font color="#8700FF">+0ms</font>
  <font color="#D75FAF"><b>zigbee-herdsman:adapter:zStack:znp:AREQ </b></font>&lt;-- AF - dataConfirm - {&quot;status&quot;:240,&quot;endpoint&quot;:1,&quot;transid&quot;:9} <font color="#D75FAF">+43s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [] <font color="#8700FF">+0ms</font>
  <font color="#005FFF"><b>zigbee-herdsman:adapter:zStack:adapter </b></font>Data confirm error (0x000d6f001619269c:31027,240,0) <font color="#005FFF">+6s</font>
  <font color="#005FFF"><b>zigbee-herdsman:adapter:zStack:adapter </b></font>Wait 2000ms <font color="#005FFF">+0ms</font>
  <font color="#005FFF"><b>zigbee-herdsman:adapter:zStack:adapter </b></font>sendZclFrameToEndpointInternal 0x000d6f001619269c:31027/1 (0,1,1) <font color="#005FFF">+2s</font>
  <font color="#FF5FAF"><b>zigbee-herdsman:adapter:zStack:znp:SREQ </b></font>--&gt; AF - dataRequest - {&quot;dstaddr&quot;:31027,&quot;destendpoint&quot;:1,&quot;srcendpoint&quot;:1,&quot;clusterid&quot;:64642,&quot;transid&quot;:10,&quot;options&quot;:0,&quot;radius&quot;:30,&quot;len&quot;:5,&quot;data&quot;:{&quot;type&quot;:&quot;Buffer&quot;,&quot;data&quot;:[5,11,18,6,246]}} <font color="#FF5FAF">+8s</font>
  <font color="#0087FF"><b>zigbee-herdsman:adapter:zStack:unpi:writer </b></font>--&gt; frame [254,15,36,1,51,121,1,1,130,252,10,0,30,5,5,11,18,6,246,227] <font color="#0087FF">+8s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>&lt;-- [254,1,100,1,0,100] <font color="#8700FF">+2s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [254,1,100,1,0,100] <font color="#8700FF">+0ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--&gt; parsed 1 - 3 - 4 - 1 - [0] - 100 <font color="#8700FF">+1ms</font>
  <font color="#5F00D7"><b>zigbee-herdsman:adapter:zStack:znp:SRSP </b></font>&lt;-- AF - dataRequest - {&quot;status&quot;:0} <font color="#5F00D7">+8s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [] <font color="#8700FF">+0ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>&lt;-- [254,3,68,128,240,1,10,60] <font color="#8700FF">+7s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [254,3,68,128,240,1,10,60] <font color="#8700FF">+0ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--&gt; parsed 3 - 2 - 4 - 128 - [240,1,10] - 60 <font color="#8700FF">+0ms</font>
  <font color="#D75FAF"><b>zigbee-herdsman:adapter:zStack:znp:AREQ </b></font>&lt;-- AF - dataConfirm - {&quot;status&quot;:240,&quot;endpoint&quot;:1,&quot;transid&quot;:10} <font color="#D75FAF">+9s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [] <font color="#8700FF">+0ms</font>
  <font color="#005FFF"><b>zigbee-herdsman:adapter:zStack:adapter </b></font>Data confirm error (0x000d6f001619269c:31027,240,1) <font color="#005FFF">+7s</font>
  <font color="#FF5FAF"><b>zigbee-herdsman:adapter:zStack:znp:SREQ </b></font>--&gt; UTIL - assocGetWithAddress - {&quot;extaddr&quot;:&quot;0x000d6f001619269c&quot;,&quot;nwkaddr&quot;:31027} <font color="#FF5FAF">+7s</font>
  <font color="#0087FF"><b>zigbee-herdsman:adapter:zStack:unpi:writer </b></font>--&gt; frame [254,10,39,74,156,38,25,22,0,111,13,0,51,121,250] <font color="#0087FF">+7s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>&lt;-- [254,36,103,74,51,121,0,0,1,4,0,0,6,1,170,0,0,0,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,235] <font color="#8700FF">+7ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [254,36,103,74,51,121,0,0,1,4,0,0,6,1,170,0,0,0,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,235] <font color="#8700FF">+0ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--&gt; parsed 36 - 3 - 7 - 74 - [51,121,0,0,1,4,0,0,6,1,170,0,0,0,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255] - 235 <font color="#8700FF">+0ms</font>
  <font color="#5F00D7"><b>zigbee-herdsman:adapter:zStack:znp:SRSP </b></font>&lt;-- UTIL - assocGetWithAddress - {&quot;nwkaddr&quot;:31027,&quot;addridx&quot;:0,&quot;noderelation&quot;:1} <font color="#5F00D7">+7s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [] <font color="#8700FF">+0ms</font>
  <font color="#005FFF"><b>zigbee-herdsman:adapter:zStack:adapter </b></font>assocRemove(0x000d6f001619269c) <font color="#005FFF">+7ms</font>
  <font color="#FF5FAF"><b>zigbee-herdsman:adapter:zStack:znp:SREQ </b></font>--&gt; UTIL - assocRemove - {&quot;ieeeadr&quot;:&quot;0x000d6f001619269c&quot;} <font color="#FF5FAF">+7ms</font>
  <font color="#0087FF"><b>zigbee-herdsman:adapter:zStack:unpi:writer </b></font>--&gt; frame [254,8,39,99,156,38,25,22,0,111,13,0,155] <font color="#0087FF">+7ms</font>
[... cut ...]
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [254,1,100,1,0,100] <font color="#8700FF">+0ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--&gt; parsed 1 - 3 - 4 - 1 - [0] - 100 <font color="#8700FF">+0ms</font>
  <font color="#5F00D7"><b>zigbee-herdsman:adapter:zStack:znp:SRSP </b></font>&lt;-- AF - dataRequest - {&quot;status&quot;:0} <font color="#5F00D7">+3s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [] <font color="#8700FF">+0ms</font>
</pre>

zhouF96, 2021-03-31:
releated commit here: https://github.com/Koenkk/zigbee-herdsman-converters/commit/66d52b8ba1fda036e527801a95009a5f84438252 
@Hacker-CB Any idea?
I've test'ed with HEIMAN's gateway and worked well. but when use z2m, it can not receive any responce while send commands.

Faustino-ai, 2021-04-01:
I just received mine last week. The "get_list" function will only work after you successfully created the modelID and learned a key.

zhouF96, 2021-04-01:
> I just received mine last week. The "get_list" function will only work after you successfully created the modelID and learned a key.

What is the procedure to create a model ID and learn a Key?

If send 
```
{
    "create": {
        "model_type": 55
    }
}
```
still no response.
 any idea?

Faustino-ai, 2021-04-01:
> > I just received mine last week. The "get_list" function will only work after you successfully created the modelID and learned a key.
> 
> What is the procedure to create a model ID and learn a Key?
> 
> If send
> 
> ```
> {
>     "create": {
>         "model_type": 55
>     }
> }
> ```
> 
> still no response.
> any idea?

This should work. Did you see anything on zigbee2mqtt log after you publish the create model_type?

zhouF96, 2021-04-02:
> This should work. Did you see anything on zigbee2mqtt log after you publish the create model_type?

<pre>  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [] <font color="#8700FF">+1ms</font>
<font color="#3465A4">Zigbee2MQTT:debug</font> 2021-04-02 11:32:27: Received MQTT message on &apos;zigbee2mqtt/0x000d6f001619269c/set&apos; with data &apos;{
  &quot;create&quot;: {
      &quot;model_type&quot;: 2
  }
}&apos;
<font color="#3465A4">Zigbee2MQTT:debug</font> 2021-04-02 11:32:27: Publishing &apos;set&apos; &apos;create&apos; to &apos;0x000d6f001619269c&apos;
  <font color="#FFD700"><b>zigbee-herdsman:controller:endpoint </b></font>Command 0x000d6f001619269c/1 heimanSpecificInfraRedRemote.createId({&quot;modelType&quot;:2}, {&quot;timeout&quot;:10000,&quot;disableResponse&quot;:false,&quot;disableRecovery&quot;:false,&quot;disableDefaultResponse&quot;:false,&quot;direction&quot;:0,&quot;srcEndpoint&quot;:null,&quot;reservedBits&quot;:0,&quot;manufacturerCode&quot;:4619,&quot;transactionSequenceNumber&quot;:null,&quot;writeUndiv&quot;:false}) <font color="#FFD700">+20s</font>
  <font color="#005FFF"><b>zigbee-herdsman:adapter:zStack:adapter </b></font>sendZclFrameToEndpointInternal 0x000d6f001619269c:36938/1 (0,0,1) <font color="#005FFF">+20s</font>
  <font color="#FF5FAF"><b>zigbee-herdsman:adapter:zStack:znp:SREQ </b></font>--&gt; AF - dataRequest - {&quot;dstaddr&quot;:36938,&quot;destendpoint&quot;:1,&quot;srcendpoint&quot;:1,&quot;clusterid&quot;:64642,&quot;transid&quot;:54,&quot;options&quot;:0,&quot;radius&quot;:30,&quot;len&quot;:6,&quot;data&quot;:{&quot;type&quot;:&quot;Buffer&quot;,&quot;data&quot;:[5,11,18,50,244,2]}} <font color="#FF5FAF">+20s</font>
  <font color="#0087FF"><b>zigbee-herdsman:adapter:zStack:unpi:writer </b></font>--&gt; frame [254,16,36,1,74,144,1,1,130,252,54,0,30,6,5,11,18,50,244,2,103] <font color="#0087FF">+20s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>&lt;-- [254,1,100,1,0,100] <font color="#8700FF">+19s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [254,1,100,1,0,100] <font color="#8700FF">+0ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--&gt; parsed 1 - 3 - 4 - 1 - [0] - 100 <font color="#8700FF">+0ms</font>
  <font color="#5F00D7"><b>zigbee-herdsman:adapter:zStack:znp:SRSP </b></font>&lt;-- AF - dataRequest - {&quot;status&quot;:0} <font color="#5F00D7">+20s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [] <font color="#8700FF">+0ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>&lt;-- [254,3,68,128,0,1,54,240] <font color="#8700FF">+106ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [254,3,68,128,0,1,54,240] <font color="#8700FF">+0ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--&gt; parsed 3 - 2 - 4 - 128 - [0,1,54] - 240 <font color="#8700FF">+1ms</font>
  <font color="#D75FAF"><b>zigbee-herdsman:adapter:zStack:znp:AREQ </b></font>&lt;-- AF - dataConfirm - {&quot;status&quot;:0,&quot;endpoint&quot;:1,&quot;transid&quot;:54} <font color="#D75FAF">+19s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [] <font color="#8700FF">+0ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>&lt;-- [254,27,68,129,0,0,130,252,74,144,1,1,0,132,0,244,225,15,0,0,7,12,11,18,50,11,244,131,74,144,29,127] <font color="#8700FF">+15ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [254,27,68,129,0,0,130,252,74,144,1,1,0,132,0,244,225,15,0,0,7,12,11,18,50,11,244,131,74,144,29,127] <font color="#8700FF">+1ms</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--&gt; parsed 27 - 2 - 4 - 129 - [0,0,130,252,74,144,1,1,0,132,0,244,225,15,0,0,7,12,11,18,50,11,244,131,74,144,29] - 127 <font color="#8700FF">+0ms</font>
  <font color="#D75FAF"><b>zigbee-herdsman:adapter:zStack:znp:AREQ </b></font>&lt;-- AF - incomingMsg - {&quot;groupid&quot;:0,&quot;clusterid&quot;:64642,&quot;srcaddr&quot;:36938,&quot;srcendpoint&quot;:1,&quot;dstendpoint&quot;:1,&quot;wasbroadcast&quot;:0,&quot;linkquality&quot;:132,&quot;securityuse&quot;:0,&quot;timestamp&quot;:1040884,&quot;transseqnumber&quot;:0,&quot;len&quot;:7,&quot;data&quot;:{&quot;type&quot;:&quot;Buffer&quot;,&quot;data&quot;:[12,11,18,50,11,244,131]}} <font color="#D75FAF">+16ms</font>
  <font color="#D7005F"><b>zigbee-herdsman:controller:log </b></font>Received &apos;zcl&apos; data &apos;{&quot;frame&quot;:{&quot;Header&quot;:{&quot;frameControl&quot;:{&quot;frameType&quot;:0,&quot;manufacturerSpecific&quot;:true,&quot;direction&quot;:1,&quot;disableDefaultResponse&quot;:false,&quot;reservedBits&quot;:0},&quot;transactionSequenceNumber&quot;:50,&quot;manufacturerCode&quot;:4619,&quot;commandIdentifier&quot;:11},&quot;Payload&quot;:{&quot;cmdId&quot;:244,&quot;statusCode&quot;:131},&quot;Command&quot;:{&quot;ID&quot;:11,&quot;name&quot;:&quot;defaultRsp&quot;,&quot;parameters&quot;:[{&quot;name&quot;:&quot;cmdId&quot;,&quot;type&quot;:32},{&quot;name&quot;:&quot;statusCode&quot;,&quot;type&quot;:32}]}},&quot;address&quot;:36938,&quot;endpoint&quot;:1,&quot;linkquality&quot;:132,&quot;groupID&quot;:0,&quot;wasBroadcast&quot;:false}&apos; <font color="#D7005F">+19s</font>
  <font color="#8700FF"><b>zigbee-herdsman:adapter:zStack:unpi:parser </b></font>--- parseNext [] <font color="#8700FF">+0ms</font>
</pre>

Then  still no expected response like 
`{
    "action": "create",
    "action_result": "success",
    "action_model_type": 55,
    "action_id": 1
}`

any idea?

Faustino-ai, 2021-04-02:
I am not so sure either. You can try to publish the "learn key" and see if the remote shows a different light color according to the instruction manual.

zhouF96, 2021-04-07:
> I am not so sure either. You can try to publish the "learn key" and see if the remote shows a different light color according to the instruction manual.

@Faustino-ai  I tried , but nothing happened. Would you please give me your log to help me debug? 

Only when I unplug and plug the HS2IRC can it send 
`{"battery":100,"devices":[{"id":1,"key_codes":[29,30],"model_type":99}],"linkquality":108}`
something like this. Very strange.

Faustino-ai, 2021-04-08:
> @Faustino-ai I tried , but nothing happened. Would you please give me your log to help me debug?
>
> Only when I unplug and plug the HS2IRC can it send [...]

I will let you know once I use that remote again. I am currently still in the middle of making my smart lock work. Won't be doing any other stuff until I figure out the smart lock. Meanwhile try to see if others can help.

coraxxx, 2021-04-19:
Hi! I have the same problem, any solution found?

hidaris, 2021-04-20:
Not yet, we are still plagued by this problem.

coraxxx, 2021-05-16:
Hi everyone! Any news about the solution?

Faustino-ai, 2021-05-17:
Hi everyone. Apparently mine doesn't work anymore after updating to the latest version. However, I came from version 1.17.1 and it still worked until my update. Now I reinstalled z2m using 1.17.1 version and it works again. Will try to figure out something. If still can't make it then probably have ask Koen.

Faustino-ai, 2021-05-17:
The interviewing process of the latest version is very problematic compared to 1.17.1. For the latest version I still can't get my HS2IRC to be successfully interviewed where as it connected to 1.17.1 like a breeze. Any idea how to solve this issue? I checked the devices.js (heiman.js in latest version), fromZigbee and toZigbee for this device and nothing seems to change except for the some lines in devices.js but I changed it back. Still didn't work though

coraxxx, 2021-05-17:
> Hi everyone. Apparently mine doesn't work anymore after updating to the latest version. However, I came from version 1.17.1 and it still worked until my update. Now I reinstalled z2m using 1.17.1 [...]

Sorry for my stupid question, but I can't figure it out myself. Could you please explain how I can roll back the z2m version? I'm using Hassos on RPI4

Faustino-ai, 2021-05-18:
> Sorry for my stupid question, but I can't figure it out myself. Could you please explain how I can roll back the z2m version? I'm using Hassos on RPI4

Unfortunately, I tried to find solution for this too, but apparently there aren't any easy ways to downgrade your z2m as Hassos addon. My suggestion would be (not sure if it will work) try to pull the older version on your own github repo and see if there's any way you can add that into HA add-on store.

Asc0tt, 2021-06-18:
I still have this problem. Is there a solution? Is there anything I can do to help you find a solution?
Within 2 weeks I will try to reinstall the older version of z2m and compare the behavior HS2IRC

roadnull, 2021-07-16:
Good afternoon, everyone. My first comment on github. I found an interesting point in the source file.
src/zcl/definition/cluster.ts
` heimanSpecificInfraRedRemote: {
        // from HS2IRC-3.0海曼智能红外转发控制器API-V01文档
        ID: 0xfc82,
        manufacturerCode: ManufacturerCode.Heiman,
        attributes: {},
        commands: {`

It seems to me that the matter is in curly brackets after the word attributes.

zhouF96, 2021-07-19:
> Good afternoon, everyone. My first comment on github. I found an interesting point in the source file.
> src/zcl/definition/cluster.ts
> ` heimanSpecificInfraRedRemote: { // from [...]

OK , Will try it.

Koenkk, 2021-07-24:
Fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-07-24 by Koenkk