Koenkk/zigbee2mqtt #6901: Request help with new device, Tuya based water valve

jimmyeao, 2021-03-31:
Tried going through the add support for a device doc, but became confused. I think this is a Tuya based device, not sure what to put for things like "supports" or "exposes"

I tried to set it up like this:
```
    //water valve
   {
    fingerprint: [ // Since a lot of Tuya devices use the same modelID, but use different data points
                   // it's usually necessary to provide a fingerprint
        {
            modelID: 'TS0601', // The model ID from: Device with modelID 'TS0601' is not supported.
                               // You may need to add \u0000 at the end of the name in some cases
            manufacturerName: '_TZE200_wt9agwf3' // The manufacturer name from: Device with modelID 'TS0601' is not supported.
        },
    ],
    model: 'TS0601',
    vendor: 'TuYA',
    description: 'Water valve',
    supports: 'thermostat, temperature',
    extend: preset.switch(),
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.setTime, // Add this if you are getting no converter for 'commandSetTimeRequest'
    meta: {
        configureKey: 1,
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
},
```

## Information about the device + link

.https://www.facebook.com/100205581602548/posts/smart-water-valve-powered-by-tuya-wifi-zigbee-easy-to-use-indoor-and-outdoor/139992237623882/.

## data/database.db entry of the device
.{"id":39,"type":"Router","ieeeAddr":"0x5c0272fffe915ecb","nwkAddr":17081,"manufId":4098,"manufName":"_TZE200_wt9agwf3","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0013\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0013\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0013\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0013","65506":21,"modelId":"TS0601","manufacturerName":"_TZE200_wt9agwf3","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1617196637660}.

ArJay60, 2021-04-03:
This is the same as #6641. Currently working on it with manufacturer. But slow progress.

ArJay60, 2021-05-04:
Valve is in release 1.18.3

closed 2021-05-04 by Koenkk