Koenkk/zigbee2mqtt #6905: Gledopto GL-C-008-2ID: incorrect attribute changed when group with this controller is toggled

andreypolyak, 2021-03-31:
## What happened

Hi,
I am using Gledopto GL-C-008-2ID controller which has two independent channels to control: RGB and CCT. It has two endpoints -- one for each channel. I control each channel on this device either individually (turn on or off appropriate light in HA) or as part of the group with other lights. Now, the problem is that when I turn any channel on or off it changes ```state_cct``` or ```state_rgb``` attribute in MQTT. But when I turn the group on or off it changes ```state``` attribute.

Here is the example. Default state (note -- ```state_cct```, ```state_rgb``` and ```state``` have ```OFF``` value):
```json
{
    "brightness": 254,
    "brightness_rgb": 253,
    "color": {
        "x": 0.612,
        "y": 0.374
    },
    "color_rgb": {
        "x": 0.589,
        "y": 0.385
    },
    "color_temp": 933,
    "color_temp_rgb": 134,
    "last_seen": 1617227125243,
    "linkquality": 72,
    "state": "OFF",
    "state_cct": "OFF",
    "state_rgb": "OFF"
}
```

Turned on RGB channel, only ```state_rgb``` changed:
```json
{
    "brightness": 254,
    "brightness_rgb": 253,
    "color": {
        "x": 0.612,
        "y": 0.374
    },
    "color_rgb": {
        "x": 0.589,
        "y": 0.385
    },
    "color_temp": 933,
    "color_temp_rgb": 134,
    "last_seen": 1617227274817,
    "linkquality": 84,
    "state": "OFF",
    "state_cct": "OFF",
    "state_rgb": "ON"
}
```

Turned it off back (all states are ```OFF```):
```json
{
    "brightness": 254,
    "brightness_rgb": 253,
    "color": {
        "x": 0.612,
        "y": 0.374
    },
    "color_rgb": {
        "x": 0.589,
        "y": 0.385
    },
    "color_temp": 933,
    "color_temp_rgb": 134,
    "last_seen": 1617227422395,
    "linkquality": 147,
    "state": "OFF",
    "state_cct": "OFF",
    "state_rgb": "OFF"
}
```

Turned on the group which only had RGB endpoint (```state``` changed but not ```state_rgb```):
```json
{
    "brightness": 254,
    "brightness_rgb": 253,
    "color": {
        "x": 0.612,
        "y": 0.374
    },
    "color_rgb": {
        "x": 0.589,
        "y": 0.385
    },
    "color_temp": 933,
    "color_temp_rgb": 134,
    "last_seen": 1617227495168,
    "linkquality": 108,
    "state": "ON",
    "state_cct": "OFF",
    "state_rgb": "OFF"
}
```

This is how the group is set up:
```yaml
'38':
  friendly_name: Group Bedroom Color
  retain: false
  optimistic: true
  devices:
    - 0x00124b0021126cf0/rgb
    - %other lights here%
```

## What did you expect to happen

In my opinion ```state_cct``` and ```state_rgb``` attributes should be changed when the group with endpoints from this controller are toggled.

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC26252
Adapter firmware version: 20210218

Koenkk, 2021-04-04:
Indeed a bug, will look into it.

Koenkk, 2021-04-04:
Should be fixed now, can you confirm?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

You can remove the `state` manually from `state.json`.

andreypolyak, 2021-04-05:
Ho Koen,
so I tried the Edge version of addon and unfortunately it didn't work reliably. Sometimes when I toggled the group it changed the state of the endpoint, but sometimes it didn't. I was not able to find any patterns when it works and when not. I'll try to prepare the logs where this will be shown later this week.

andreypolyak, 2021-04-12:
Hi @Koenkk,
sorry for the late response. So I tested everything once again and was not able to reproduce the problem that I written in my last comment. But I was able to find another one.

To demo it I created the short video:
https://user-images.githubusercontent.com/7446762/114464699-c71cb080-9bee-11eb-8617-abfd471e1150.mov

I have created group with only one endpoint (RGB channel for Gledopto) and following settings:
```yaml
'44':
  friendly_name: Test Group
  retain: false
  optimistic: true
  devices:
    - 0x00124b002113ac0e/rgb
```

When I turn on or off this group state of endpoint changes accordingly. But when I turn on or off this RGB channel it doesn't change the state of the group.

Koenkk, 2021-04-13:
Should also be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

andreypolyak, 2021-04-16:
Hi @Koenkk,
everything work as expected now. Thank you so much!

closed 2021-04-16 by andreypolyak