Koenkk/zigbee2mqtt #6906: Adding Xiaomi / Aqara OTA Updates

sorryusernameisalreadytaken, 2021-04-01:
I have a few Aqara Xiaomi devices and have also done some research as to why they are not getting OTA updates over zigbee2mqtt.
In the end I got an OTA ZIP with numerous OTA files from xiaomi from a telegram-group.

For the [LLKZMK11LM ](https://www.zigbee2mqtt.io/devices/LLKZMK11LM.html)I am pretty sure that there is an update and I think that my 4x adapter does not have it. According to this post it should be version [0.0.0_0046](https://community.hubitat.com/t/new-xiaomi-aqara-relay-2-channel/13335/169).
I think it is the OTA file:
**OTA_lumi_relay_c2acn01_V1.3.42.20191230_68389C.20200429113233.ota**

Only now I ask myself several questions:
1. Can I trust the firmware or can I verify it somehow?
2. How could I flash the OTA manually?
3. Do we want and can OTA file links at xiaomi like Philips with sniff and include the links, if so, how does it work?

[ota_files.zip](https://github.com/Koenkk/zigbee2mqtt/files/6240732/ota_files.zip)

Attached the file list:
```
+---lumi.airrtc.tcpco2ecn01
|   \---12_484d603c_885a430ec8da27e378e4f89c8dae804d
|           OTA_lumi.airrtc.tcpco2ecn01_OTA_v12.20180828161433.ota
|
+---lumi.airrtc.tcpecn01
|   \---30_0000001E_7ac0c8b0_8789dfe0a84b62d2273f5dedcc74caa3
|           OTA_withCRC_LMES_HVACController_V1.2.30_20170710.20181024102131.ota
|
+---lumi.airrtc.tcpecn02
|   \---12_e4226584_fbafd72bcc06b322f3e5f825de5246ab
|           OTA_lumi.airrtc.tcpecn02_OTA_v12.20180828161528.ota
|
+---lumi.ctrl_86plug.aq1
|   \---33_7eb6491b_f4889ea6c8ff478d32a269f7d0536f45
|           LM15_PA_86SP_AQ1_V1.0.33_20180625_v33.20180717175418.ota
|
+---lumi.ctrl_86plug.v1
|   \---11_f5dcb0dd_41b1ced5ffd3742e4a61b1325ee206ed
|           LM15_86SP_aq_V1.0.11_20170302_OTA_v11_withCRC.20170417201259.ota
|
+---lumi.ctrl_dimmer3.es1
|   \---30_99620999_570f21443673761208b50ec0400b6dc6
|           OTA_withCRC_LMES_Dimmer3Controller_V1.2.30_20170801.20170818101543.ota
|
+---lumi.ctrl_dualchn.es1
|   \---30_c0e78b9c_86ffab33d58a29a8cee396c363870c94
|           OTA_withCRC_LMES_DualController_V1.3.30_20170801.20170818100757.ota
|
+---lumi.ctrl_hvac.es1
|   +---19_92358b36_a4874175dc935ff239a82e6d7614cc18
|   |       OTA_NOCRC_LMES_HVACController_V1.2.19_20170802.20170818101012.ota
|   |
|   \---30_7ac0c8b0_8789dfe0a84b62d2273f5dedcc74caa3
|           OTA_withCRC_LMES_HVACController_V1.2.30_20170710.20170818101250.ota
|
+---lumi.ctrl_ln1.aq1
|   +---30_0000001E_ec0d052f_62f658d41a45f39f2d87e83a4b5755d6
|   |       PA_LM15_ln1_AQ1_V1.0.30_20170906_v30_withCRC.20170920094345.ota
|   |
|   \---33_d0710f7c_cd866651d3576e9c4609ff10de0599dd
|           LM15_ln1_AQ1_V1.0.33_20180625_v33.20181011142900.ota
|
+---lumi.ctrl_ln1.v1
|   +---30_46f6ea4d_100b565bfc73e7fe40732bee10e5f18f
|   |       LM15_ln1_AQ_V1.0.30_20180312_v30_withCRC.20180507182122.ota
|   |
|   \---32_4deb9897_98ca9796a724ba0f9cc5a696f590a373
|           LM15_ln1_AQ_V1.0.32_20180625_v32.20181008194104.ota
|
+---lumi.ctrl_ln2.aq1
|   +---30_0000001E_f3167a6f_b33bb003f950b470835aa7bec8ea8a92
|   |       PA_LM15_ln2_AQ1_V1.0.30_20170906_v30_withCRC.20170920094437.ota
|   |
|   \---33_fcf80a37_ce5e8fa10971f3855be390d6a62b9cba
|           PA_LM15_ln2_AQ1_V1.0.33_20180625_v33.20181011142943.ota
|
+---lumi.ctrl_ln2.v1
|   +---30_266c3ae7_62d350e348e9e89de4383e49f8d6ef4d
|   |       LM15_ln2_AQ_V1.0.30_20180312_v30_withCRC.20180507182208.ota
|   |
|   \---32_806aa849_e688f851ae1102c3ac3400c98aab7976
|           LM15_ln2_V1.0.32_20180625_v32.20181008194246.ota
|
+---lumi.ctrl_neutral1.v1
|   +---21_00000015_7eae51ea_6df4c7023ef91436d165779a33ba39aa
|   |       OTA_LM15_LNN_V2.6.21_20171125_neutral1_19ms_DIO19Led.20171218154458.ota
|   |
|   \---22_47a34a58_1fa54acb8bc4e9a960e9b9bda34eb767
|           OTA_LM15_LNN_V2.6.22_20180503_neutral1_19ms_DIO19Led.20181011142357.ota
|
+---lumi.ctrl_neutral2.v1
|   +---21_00000015_e505fc69_26fa61537edc59cd23d84180e46e0684
|   |       OTA_LM15_LNN_V2.6.21_20171125_neutral2_19ms_DIO19Led.20171220112633.ota
|   |
|   \---22_79ebb1df_38e14e3866e4ac0250cce4ba398180de
|           OTA_LM15_LNN_V2.6.22_20180503_neutral2_19ms_DIO19Led.20181011142447.ota
|
+---lumi.ctrl_rgb.es1
|   \---30_dc48ef43_0519e275ce8584937cd503745692009c
|           OTA_WithCRC_LMES_RGBController_V1.2.30_20170801.20170920100827.ota
|
+---lumi.curtain.aq2
|   +---0.0.0_0030_F2DAF8AE_72769485efa612d5e5fcf03c9cc8c672
|   |       OTA_lumi.curtain.aq2_0.0.0_0030_20200706_C8C672.20200713103412.ota
|   |
|   +---11_0000000B_9c44c600_ad6cea36f3f01e1a97b65b2def3d2922
|   |       lumi.curtain.aq2_1.1.11_20181019_v11.20181023102952.ota
|   |
|   \---13_0000000D_9c81cd41_cde5d443baf876ab272d90d9b7b5fad5
|           OTA_lumi.curtain.aq2_1.1.13_20190906_v13_FAD5.20190906162202.ota
|
+---lumi.curtain.v1
|   +---30_3dcf4dfd_bc5d722825e68d688e94adb41be024e9
|   |       lumi.curtain_1.1.30_20170823_v30_withCRC.20170823172848.ota
|   |
|   +---32_7df7c5ff_9f2dbfbaf10d1b07a3184e3e752fdad2
|   |       lumi.curtain_1.1.32_20180813_v32.20180820101424.ota
|   |
|   \---34_97A16B5A_cfa3ae2e8094d0f0b0b5aabbed1f0a7b
|           OTA_lumi.curtain_0.0.0_0034_20200717_1F0A7B.20200717154021.ota
|
+---lumi.eemeter.zbtecn01
|   \---35_744cfdb_b77447aca1bb5cef84eccf23c094c3aa
|           lumi.zzjq_1.1.35_20180824_v35.20180824161828.ota
|
+---lumi.light.aqcn02
|   +---0.0.0_0031_B2BCF3DF_03e9c17cb6a7885b3955dcbff57e7f1c
|   |       OTA_lumi.light.aqcn02_0.0.0_0031_20200701_7E7F1C.20200710171134.ota
|   |
|   +---22_f20c26e8_0ba35c6da458c0e598932188ba37058c
|   |       OTA_lumi.light.aqcn02_1.1.22_20180930_v22_058C.20181011143043.ota
|   |
|   \---23_00000017_a019c705_da60f9f1204e5c2095aaea0568a29c97
|           OTA_lumi.light.aqcn02_1.1.23_20181122_v23_9C97.20181122170517.ota
|
+---lumi.lock.acn02
|   \---9_00000009_9eedaaef_cccbc45ad3beefa3232886ff7fa9ff09
|           OTA_LMACN02_DoorLock_V4.1.09_20180317.20180411152155.ota
|
+---lumi.lock.aq1
|   \---19_6e422aa6_0d13199b176de770f894c90a0628b588
|           OTA_LMAQ_DoorLock_V2.2.19_20171108.20180129142422.ota
|
+---lumi.lock.v1
|   \---13_0000000D_9ba8906a_3f4d33c61fce832550e94cefd5385ab4
|           OTA_LM_WM_DoorLock_V2.3.13_20180409.20180412161023.ota
|
+---lumi.plug.aq1
|   \---30_ac32f660_3622c27b1744f777981cbe2c5e32a84b
|           LM15_SP_aq_V1.3.30_20180724_v30_withCRC.20180724160524.ota
|
+---lumi.plug.v1
|   +---0.0.0_0030_270a6c92_e8da7eafd9322c0af6cdbf4c93cd7cd3
|   |       LM15_SP_mi_V1.3.30_20170929_v30_withCRC.20180514181348.ota
|   |
|   +---0.0.0_0090_4442E0E_c585663f837a0948b0ef65b59ce2ab53
|   |       OTA_lumi.plug_V90_20191220_E2AB53.20200106152408.ota
|   |
|   \---30_270a6c92_e8da7eafd9322c0af6cdbf4c93cd7cd3
|           LM15_SP_mi_V1.3.30_20170929_v30_withCRC.20180514181348.ota
|
+---lumi.relay.c2acn01
|   +---0.0.0_0036_8691F18B_e92ad148ac2de53568745a44a594007f
|   |       OTA_lumi_relay_c2acn01_V1.3.36.20190308_94007F.20191009160743.ota
|   |
|   +---0.0.0_0042_6CD27E92_33ff98239d745a64cbe3e0408768389c
|   |       OTA_lumi_relay_c2acn01_V1.3.42.20191230_68389C.20200429113233.ota
|   |
|   +---30_0000001E_c0e78b9c_86ffab33d58a29a8cee396c363870c94
|   |       OTA_withCRC_LMES_DualController_V1.3.30_20170801.20181024095320.ota
|   |
|   \---35_3b451bb3_2b39856873feffe89987e517ce06c78a
|           OTA_lumi_relay_c2acn01_V1.3.35.20180920.20181008170031.ota
|
+---lumi.relay.c4acn01
|   +---12_1ce41eab_41da90b9d85c0e136dffcd236133c075
|   |       OTA_LMC4ACN01_FourController_V1.1.12_20180706.20180719103618.ota
|   |
|   \---20_CC0AB1E7_63445e24e3f5b5a3d4b7d3bb2a8ffec9
|           OTA_lumi.relay.c4acn01_V2.1.20_201900821_8FFEC9.20190906162416.ota
|
\---lumi.switch.b2nacn02
    \---0.0.0_0062_BD7A2BBE_58490ce9b329b1360d350c77bc3282b0
            OTA_lumi.switch.b2nacn02_0.0.0_0062_20200827_3282B0.20200831122819.ota

```

mrIngrid, 2021-04-01:
Nice job 🙂.

Koenkk, 2021-04-01:
@kirovilya are these the same FWs that you found?

kirovilya, 2021-04-01:
In my case, I found an unpleasant feature. OTA-files for different devices had the same parameters: imageType, manufacturerCode. As a result, during the update, the incorrect firmware was uploaded and the device turned into a brick.
for example:
`Updated existing entry ({"fileVersion":32,"fileSize":199550,"manufacturerCode":4447,"imageType":257,"sha512":"0d3bc193aca543a6e4e0f40745d35d0e6e6200ae77241366c2040052cc2f2c801b6625fb59a0a83624ae4fcec1f4e08c044653997d4da23a274589ca99fe69fa","url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/lumi.light.aqcn02_32.ota","path":"images/Xiaomi/lumi.light.aqcn02_32.ota"})
`
and
`
Updated existing entry ({"fileVersion":62,"fileSize":190302,"manufacturerCode":4447,"imageType":257,"sha512":"ae207109dfd2afe1e85079d87cb39ace45c3d00392dba09d65ce38104570d805ade1591fd149ed810f98bd9d1c6636e772ff0cbcc1ec9444f67f48c51e912424","url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/OTA_lumi.switch.b2nacn02_0.0.0_0062_20200827_3282B0.20200831122819.ota","path":"images/Xiaomi/OTA_lumi.switch.b2nacn02_0.0.0_0062_20200827_3282B0.20200831122819.ota"})
`

I have not yet found a solution (there was no time)

kirovilya, 2021-04-01:
And this is not the only case in my set of OTA-files.
Now I think there are 2 solutions:
1) Add one more parameter to the OTA-file repository descriptions - the model name. Fill it in manually when adding an ota-file. The model name can be found in the firmware file itself.
2) Make it possible to upload custom the firmware, not from a common repository.

pipiche38, 2021-04-02:
@kirovilya I had similar case with Schneider Wiser devices, where basically all firmware images are using the same ImageType

kirovilya, 2021-04-02:
@Koenkk what do you think of this situation? what is the best way to proceed?

sorryusernameisalreadytaken, 2021-04-02:
I have a little hope that you can get interesting links with a hacked xiaomi lumi gateway but maybe that's just such a stupid idea :D
Got me a DGNWG02LM, is on the way to me. As soon as it arrives, I would start right away to get me access. Still have the project here in mind, but have also found a few current instructions:
https://github.com/dgiese/dustcloud-nexmon
https://github.com/T-REX-XP/XiaomiGatewayHack

If someone can help me with questions, I would be very pleased.

My dream goal would be to be able to fake requests for foreign Aqara / Xiaomi devices to then get the appropriate OTA's or links.

Koenkk, 2021-04-04:
@kirovilya for gledopto we already specify the model since the image types are the same: https://github.com/Koenkk/zigbee-OTA/blob/41b4f685acf7bb38d5bed1f2ecdc1b4c9c693d3e/index.json#L799 . Could this also be used here?

kirovilya, 2021-04-04:
@Koenkk  Oh, I didn't notice. Yes, I'll try to do it as well. thanks

Koenkk, 2021-04-06:
Closing this, merged now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-04-06 by Koenkk

TheJulianJES, 2022-07-16:
I've got some ``lumi.plug.maeu01`` plugs with newer firmware on them. (The ``lumi.plug.mmeu01`` have potentially updated firmware too).
I did find a way to get power measurement working on all ``lumi.plug.maeu01`` plugs with the latest firmware (with ZHA) and having the updated images could be nice for that.

I don't have a Xiaomi gateway to play around with at the moment. For those who found the original images, is it possible for you to link the Telegram group or check if there are updates?

@Otnow IIRC you added also multiple Xiaomi images to the repo too, but I think for your method to work, you need to have the devices and a Xiaomi gateway, right? (Maybe I'll just pick up an "Aqara Hub E1" if that works)

Otnow, 2022-07-16:
@TheJulianJES, I managed to get up-to-date links to OTA files for the devices you requested without having them physically:

- `lumi.plug.maeu01`: https://github.com/Koenkk/zigbee-OTA/pull/140
- `lumi.plug.mmeu01`: https://github.com/Koenkk/zigbee-OTA/pull/139

And additionally:

- `lumi.plug.maus01`: https://github.com/Koenkk/zigbee-OTA/pull/141

TheJulianJES, 2022-07-17:
Thank you so much!

sorryusernameisalreadytaken, 2022-07-18:
> @TheJulianJES, I managed to get up-to-date links to OTA files for the devices you requested without having them physically:
> 
> * `lumi.plug.maeu01`: [Update OTA file for Xiaomi `SP-EUC01` zigbee-OTA#140](https://github.com/Koenkk/zigbee-OTA/pull/140)
> * `lumi.plug.mmeu01`: [Update OTA file for Xiaomi `ZNCZ04LM` zigbee-OTA#139](https://github.com/Koenkk/zigbee-OTA/pull/139)
> 
> And additionally:
> 
> * `lumi.plug.maus01`: [Add OTA file for Xiaomi `ZNCZ12LM` zigbee-OTA#141](https://github.com/Koenkk/zigbee-OTA/pull/141)

@Otnow can you help us here getting more details about your trick to get the direct dl links?
https://github.com/Koenkk/zigbee2mqtt/discussions/13193

Otnow, 2022-07-18:
@sorryusernameisalreadytaken, answered in discussion https://github.com/Koenkk/zigbee2mqtt/discussions/13193.

nunoary, 2022-09-28:
@Otnow Can you please share the firmware for lumi.plug (ZNCZ02LM), but version V90.

As I've seen around here ([this](https://github.com/Koenkk/zigbee2mqtt/discussions/2921?sort=top#discussioncomment-713155) and [this](https://github.com/Koenkk/zigbee2mqtt/issues/13856)) some users are having the same problem as me. Sockets with firmware older than 2018 cannot be updated to V92, available here. So I thought it would be possible to upgrade to V90 first and then to V92, but the V90 has been deleted and is not available.

Otnow, 2022-09-28:
> Can you please share the firmware for lumi.plug (ZNCZ02LM), but version V90.

Of course, here's the file you want, which has been removed: [OTA_lumi.plug_V90_20191220_E2AB53.20200106152408.ota](https://github.com/Koenkk/zigbee-OTA/raw/d946ddf454255d16b5d26e09bcd8233256b4ca53/images/Xiaomi/OTA_lumi.plug_V90_20191220_E2AB53.20200106152408.ota)

nunoary, 2022-09-29:
> 
> Of course, here's the file you want, which has been removed: [OTA_lumi.plug_V90_20191220_E2AB53.20200106152408.ota](https://github.com/Koenkk/zigbee-OTA/raw/d946ddf454255d16b5d26e09bcd8233256b4ca53/images/Xiaomi/OTA_lumi.plug_V90_20191220_E2AB53.20200106152408.ota)

Thanks @Otnow. 

Unfortunately I can't update to this version (V90) either. Plugs with the V34 (2018) I can update to the V92 without problems but the ones with the V22 (2017) don't update.

volcanodave, 2022-10-26:
> Thanks @Otnow.
>
>
>
> Unfortunately I can't update to this version (V90) either. Plugs with the V34 (2018) I can update to the V92 without problems but the ones with the V22 (2017) don't update.

I'm having the same issue. Is there anywhere to get the pre V90 files so that a progressive update to V92 is possible?

jkerekes99, 2023-01-09:
> Thanks @Otnow.
>
> Unfortunately I can't update to this version (V90) either. Plugs with the V34 (2018) I can update to the V92 without problems but the ones with the V22 (2017) don't update.

Hey all,
All my ZNCZ02LM plugs are on firmware 02-28-2017 and refuse to do OTA, even when trying with the V90 file.

How can I get these plugs updated?

nunoary, 2023-01-09:
I read somewhere that if the firmware is older than v34 (my outlet's had v22), then the update procedure must be done twice: first, the firmware will be updated to v30, the next step is to v92 from 12/24/2021.

So we need the v30 of firmware. I don't know if @Otnow can provide the firmware version 30 but he was kind enough to provide the v90 file, which I tested and couldn't update.

Otnow, 2023-01-09:
> So we need the v30 of firmware. I don't know if @Otnow can provide the firmware version 30 but he was kind enough to provide the v90 file, which I tested and couldn't update.

I'm not entirely sure, but in archive [ota_files.zip](https://github.com/Koenkk/zigbee2mqtt/files/6240732/ota_files.zip) (_lumi.plug.v1 directory_) from topic https://github.com/Koenkk/zigbee2mqtt/issues/6906 I found a firmware file similar to what you are looking for: [LM15_SP_mi_V1.3.30_20170929_v30_withCRC.20180514181348.ota.zip](https://github.com/Koenkk/zigbee2mqtt/files/10374315/LM15_SP_mi_V1.3.30_20170929_v30_withCRC.20180514181348.ota.zip) (_do not forget to extract the firmware file from the archive_).

nunoary, 2023-01-09:
@Otnow many, many thanks, it worked...finally.

@volcanodave @jkerekes99 First I updated to v30 with the file sent by **Otnow** then I updated to v92 and it worked

volcanodave, 2023-01-09:
Thanks you @nunoary and @Otnow! I'll give it a try later today

volcanodave, 2023-01-10:
> @Otnow many, many thanks, it worked...finally.
> 
> @volcanodave @jkerekes99 First I updated to v30 with the file sent by **Otnow** then I updated to v92 and it worked

Worked for me! Thanks so much for your help :)

ShonkyCH, 2023-01-11:
Regarding the incremental upgrade for the ZNC02LM I have the override file created and the firmware in place, but Z2M finds v92 rather than the v30 I added. I have ota.json inext to configuration.yaml

```
ota:
  ikea_ota_use_test_url: false
  zigbee_ota_override_index_location: ota.json
```

and ota.json
```
[
    {
        "url": "LM15_SP_mi_V1.3.30_20170929_v30_withCRC.20180514181348.ota"
    }
]
```

How do I stop Z2M seeing the latest version so I can do the stepped upgrade?

jwillmer, 2023-01-24:
@jkerekes99 how did you do the update? If I specify it in my own ota.json it wont be picked up since zigbee2mqtt finds a newer file online. At least this is what I think happens. It always tries to install the latest version.

ShonkyCH, 2023-01-25:
> @jkerekes99 how did you do the update? If I specify it in my own ota.json it wont be picked up since zigbee2mqtt finds a newer file online. At least this is what I think happens. It always tries to install the latest version.
Restarted z2m? Did you read my last post?

jwillmer, 2023-01-25:
@ShonkyCH  I did restart z2m. And I know that it is reading my `ota.json` since it reported formatting errors the first time I tried it. But when I issue a update request by clicking on `Check all` I can see in the logs that the device responds with: `{"installed_version":22,"latest_version":92,"state":"available"}`. 

My ota.json
```
[
    {
        "fileVersion": 30,
        "fileSize": 191714,
        "manufacturerCode": 4447,
        "imageType": 257,
        "sha512": "8fd5a04c36991ed8b7e4de3a842e15162b6c8e07e3b0dce5e74ef6295cd653b2e69d44ab5e5e0274c6baf7b9633b4639319036b57e3a383606e70bf1742295>
        "modelId": "lumi.plug.aq1",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/LM15_SP_aq_V1.3.30_20180724_v30_withCRC.20180724160524.ota",
        "path": "images/Xiaomi/LM15_SP_aq_V1.3.30_20180724_v30_withCRC.20180724160524.ota",
        "force": true
    }
]
```

ShonkyCH, 2023-01-25:
Presumably the bad JSON is just a copy paste from a terminal. The SHA512 is truncated.

You have both a URL and a path. Not sure both will work. The url you have also redirects - not sure if Z2M can handle that.

You don't need anything other than the URL as per my post. I'd say copy the file locallly (url can still be a local file) and just put it next to your ota.json file and leave out all the other stuff. For a once off upgrade I'd forgo it all.

You are also using the"aq" version where I used the "mi" version. Are you sure you have the right one? I used the mi version for a ZNCZ02LM device.

jwillmer, 2023-01-25:
Awesome. Now I see version 30 when searching for updates! And... it worked :)

ShonkyCH, 2023-01-25:
Maybe a note about what was wrong with what you had done to help others who come here in the future.

jwillmer, 2023-01-25:
As you pointed out the copy&paste I did was missing parts - dl the referenced file and your json worked in the end. Also the restart button of z2m did not really work for me. I had to restart the docker container from outside to change the settings.

hakong, 2023-02-09:
@kirovilya did you ever find a way to revert to the correct firmware for the LLKZMK11LM (lumi.relay.c2acn01)? I just turned one of mine into a lumi.ctrl_86plug.aq1.

kirovilya, 2023-02-10:
@hakong I didn’t do it myself, but you can probably solder the wires to the chip inside and flash the desired firmware via jtag. you only need to find the appropriate firmware, not OTA

Nedjumiiro, 2023-02-24:
Lamp Aqara ZNLDP12LM (lumi.light.aqcn02) automatically updated in ZHA firmware from the socket lumi.ctrl_86plug.aq. On a five-fold inclusion for pairing does not respond, does not light up. When removed from ZHA, it returns there as an outlet. Can I somehow return the firmware or a new lamp to the trash? Not even 10 minutes worked.

Nedjumiiro, 2023-02-25:
> @kirovilya did you ever find a way to revert to the correct firmware for the LLKZMK11LM (lumi.relay.c2acn01)? I just turned one of mine into a lumi.ctrl_86plug.aq1.

Did you manage to restore the relay firmware?

hakong, 2023-02-27:
> > @kirovilya did you ever find a way to revert to the correct firmware for the LLKZMK11LM (lumi.relay.c2acn01)? I just turned one of mine into a lumi.ctrl_86plug.aq1.
> 
> Did you manage to restore the relay firmware?

No, not yet. @Koenkk do you know if that's possible?

Koenkk, 2023-02-27:
If you have the firmware file, you can follow https://www.zigbee2mqtt.io/guide/configuration/ota-device-updates.html#ota-index-override-file

hakong, 2023-03-03:
I managed to send the correct firmware to the relay but it looks like it didn't write/flash it. Maybe there's downgrade protection?
It's completely dead now. Power cycled and tried to reset it. No lights, nothing.

oywino, 2024-02-09:
Doe anyone know where to find FW files for the Aqara Motion sensor?

timurlutfullin, 2024-11-30:
Тоже столкнулся с такой проблемой, под рукой не было шлюза для обновления. Обновил через zigbee2mqtt подсунув нужную прошивку.
Имеем zncz02lm прошивка 22 от 02-28-2017 штатно постоянно выдает ошибку по OTA с 22 на 92, через MiHome обновлений не видит. 
Надо обновить на промежуточную 30 прошивку. Обсуждение тут https://github.com/Koenkk/zigbee2mqtt/issues/6906
Синтаксис локального обновления тут https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files
Все делаем через FileEditor
1. В настройках /homeassistant/zigbee2mqtt/configuration.yaml добавляем строки
```
ota:
    zigbee_ota_override_index_location: ota.json
```
2. Создаем /homeassistant/zigbee2mqtt/ota.json и пишем там
```
[
    {
        "url": "LM15_SP_mi_V1.3.30_20170929_v30_withCRC.20180514181348.ota",
        "force": true
    }
]
```
3. В  /homeassistant/zigbee2mqtt заливаем прошивку распакованную прошивку  https://github.com/Koenkk/zigbee2mqtt/files/10374315/LM15_SP_mi_V1.3.30_20170929_v30_withCRC.20180514181348.ota.zip
4. Полностью перегружаем систему, а не только HA. 
5. Видим что zigbee2mqtt теперь предлагает обновиться с 22 на 30. Обновляемся.
6. Откатываем пункты 1, 2, 3 и полностью перезагружаемся.
7. Теперь уже zigbee2mqtt предлагает нам обновиться с 30 на 92, штатно обновляемся.