Koenkk/zigbee2mqtt #6911: YSRSAI YSR-MINI-01_rgbcct does not turn off

fabir-git, 2021-04-01:
## What happened
The YSRSAI YSR-MINI-01_rgbcct does not turn off the led strip when I send the off command (HomeAssistant or via zigbee2mqtt frontend).
After sending the command it is shown as off (in HomeAssistant and the Zigbee2mqtt frontend)
Every other command (change brightness / colour works without problems).

## What did you expect to happen
The LED strip turns off.

## How to reproduce it (minimal and precise)
- send the off command
- device does not turn off
- frontend lists the device as off

## Debug info
Zigbee2MQTT version: 1.18.2
Adapter hardware: CC2652RB (Slaesh)
Adapter firmware version: zStack3x0, 20210120

Log:
`Debug Received MQTT message on 'zigbee2mqtt/LED-Stripe/set' with data '{"state": "OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/LED-Stripe', payload '{"brightness":253,"color":{"x":0.459,"y":0.411},"color_mode":"xy","color_temp":2732,"linkquality":21,"state":"OFF"}'`

fabir-git, 2021-05-02:
Issue is still active

closed 2021-06-10 by github-actions[bot]

Clement87, 2021-11-09:
Hi, I have the exact same issue.

stumbik, 2021-11-28:
Hi, I had the same problem and it was caused by adding specific "transition" configuration. When I remove it and restart z2m then everything started working as before.

kornelremete, 2021-12-15:
same issue for me.

smekac95, 2022-01-04:
Hello, I have the same problem with YSRSAI YSR-MINI-01 with all variants.

vorlocsev, 2022-01-11:
same issue exists

Boolkinator, 2023-01-15:
same issue