Koenkk/zigbee2mqtt #6912: Zigbee2MQTT restart every 5 seconds since last update

freakbn777, 2021-04-01:
# Bug Report
## What happened
Since update done by 'git pull' today, Zigbee2MQTT restart every ~5 seconds
I flashed the latest firmware on my CC2531
I updated Node (14.16.0) and NPM (6.14.11)
I also reseted the key (2s button push)
And notthing in FAQ regarding usual CC2531 problems allow me to solve that issue

Debug log: 
https://pastebin.com/zZyZLeFE

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: Latest
Adapter hardware: CC2531
Adapter firmware version: Latest

freakbn777, 2021-04-01:
I used debug mode and that showed me that few modules are missing or too old, I installed and updated then, now service don't restart every second.

But since I have copy back my /data folder, I have this message and /data/log/xxxx/log.txt is created but empty.
https://pastebin.com/5eXdtnWT

Koenkk, 2021-04-01:
- First log looks like you need to execute: `cd /opt/zigbee2mqtt && npm ci`
- Second log shows a permission error (run z2m as root or fix your permission on the data directory).

freakbn777, 2021-04-01:
Thanks, When I installed it, it was done with Pi user but now it disable and I use an another account, I made chown -R regarding your reply, and also the first command. Everything is running again, devices are coming back automatically.

Many thanks for your help :)

closed 2021-04-01 by freakbn777