Koenkk/zigbee2mqtt #6914: CR11S8UZ is unsupported device?

kpeu3i, 2021-04-01:
I'm trying to connect CR11S8UZ (https://www.zigbee2mqtt.io/devices/CR11S8UZ.html) using the latest installation zigbee2mqtt (docker).

The device has successfully been paired but further messages can no be recognized: 
```
Received message from unsupported device with Zigbee model '51725b7bcba945c8a595b325127461e9' and manufacturer name '欧瑞博'
```

Full logs: https://pastebin.com/yxWP6BSL

What's wrong here?

Koenkk, 2021-04-01:
Updating to Zigbee2MQTT 1.18.2 should fix it.

kpeu3i, 2021-04-01:
Thank you for your quick response! The newest version (1.18.2, released ~11h ago :) ) fixed the issue. But I've found that both bottom buttons won't work. Is it expected behavior?

[image]

kpeu3i, 2021-04-01:
seems this here https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L2551

Koenkk, 2021-04-04:
Can you provide the debug logging when clicking that button?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

kpeu3i, 2021-04-04:
I've enabled logging as described in the link above and restarted the Docker container as well. Unfortunately, nothing happens once these bottom buttons are pressed (logs are empty). But when I press other buttons, I see a lot of new lines the logs (each keypress is separated by new line):

```
zigbee2mqtt      | 2021-04-04T13:15:12.930Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,68,129,0,0,23,0,253,147,1,10,0,33,0,151,122,61,0,0,6,25,0,8,3,0,0,253,147,29,59]
zigbee2mqtt      | 2021-04-04T13:15:12.931Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,68,129,0,0,23,0,253,147,1,10,0,33,0,151,122,61,0,0,6,25,0,8,3,0,0,253,147,29,59]
zigbee2mqtt      | 2021-04-04T13:15:12.931Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 2 - 4 - 129 - [0,0,23,0,253,147,1,10,0,33,0,151,122,61,0,0,6,25,0,8,3,0,0,253,147,29] - 59
zigbee2mqtt      | 2021-04-04T13:15:12.932Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":23,"srcaddr":37885,"srcendpoint":1,"dstendpoint":10,"wasbroadcast":0,"linkquality":33,"securityuse":0,"timestamp":4029079,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[25,0,8,3,0,0]}}
zigbee2mqtt      | 2021-04-04T13:15:12.933Z zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":23,"data":{"type":"Buffer","data":[25,0,8,3,0,0]},"address":37885,"endpoint":1,"linkquality":33,"groupID":0,"wasBroadcast":false}'
zigbee2mqtt      | Zigbee2MQTT:debug 2021-04-04 13:15:12: Received Zigbee message from '0x00124b000cc8d641', type 'raw', cluster '23', data '{"data":[25,0,8,3,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
zigbee2mqtt      | 2021-04-04T13:15:12.937Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt      | Zigbee2MQTT:info  2021-04-04 13:15:12: MQTT publish: topic 'zigbee2mqtt/0x00124b000cc8d641', payload '{"action":"button_1_click","linkquality":33}'
zigbee2mqtt      | Zigbee2MQTT:info  2021-04-04 13:15:12: MQTT publish: topic 'zigbee2mqtt/0x00124b000cc8d641', payload '{"action":"","linkquality":33}'
zigbee2mqtt      | Zigbee2MQTT:info  2021-04-04 13:15:12: MQTT publish: topic 'zigbee2mqtt/0x00124b000cc8d641/action', payload 'button_1_click'

zigbee2mqtt      | 2021-04-04T13:15:18.856Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,68,129,0,0,23,0,253,147,1,10,0,63,0,46,33,67,0,0,6,25,0,8,11,0,0,253,147,29,177]
zigbee2mqtt      | 2021-04-04T13:15:18.856Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,68,129,0,0,23,0,253,147,1,10,0,63,0,46,33,67,0,0,6,25,0,8,11,0,0,253,147,29,177]
zigbee2mqtt      | 2021-04-04T13:15:18.856Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 2 - 4 - 129 - [0,0,23,0,253,147,1,10,0,63,0,46,33,67,0,0,6,25,0,8,11,0,0,253,147,29] - 177
zigbee2mqtt      | 2021-04-04T13:15:18.857Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":23,"srcaddr":37885,"srcendpoint":1,"dstendpoint":10,"wasbroadcast":0,"linkquality":63,"securityuse":0,"timestamp":4399406,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[25,0,8,11,0,0]}}
zigbee2mqtt      | 2021-04-04T13:15:18.858Z zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":23,"data":{"type":"Buffer","data":[25,0,8,11,0,0]},"address":37885,"endpoint":1,"linkquality":63,"groupID":0,"wasBroadcast":false}'
zigbee2mqtt      | Zigbee2MQTT:debug 2021-04-04 13:15:18: Received Zigbee message from '0x00124b000cc8d641', type 'raw', cluster '23', data '{"data":[25,0,8,11,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
zigbee2mqtt      | 2021-04-04T13:15:18.862Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt      | Zigbee2MQTT:info  2021-04-04 13:15:18: MQTT publish: topic 'zigbee2mqtt/0x00124b000cc8d641', payload '{"action":"button_2_click","linkquality":63}'
zigbee2mqtt      | Zigbee2MQTT:info  2021-04-04 13:15:18: MQTT publish: topic 'zigbee2mqtt/0x00124b000cc8d641', payload '{"action":"","linkquality":63}'
zigbee2mqtt      | Zigbee2MQTT:info  2021-04-04 13:15:18: MQTT publish: topic 'zigbee2mqtt/0x00124b000cc8d641/action', payload 'button_2_click'

zigbee2mqtt      | 2021-04-04T13:15:22.876Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,68,129,0,0,23,0,253,147,1,10,0,42,0,147,246,70,0,0,6,25,0,8,7,0,0,253,147,29,199]
zigbee2mqtt      | 2021-04-04T13:15:22.876Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,68,129,0,0,23,0,253,147,1,10,0,42,0,147,246,70,0,0,6,25,0,8,7,0,0,253,147,29,199]
zigbee2mqtt      | 2021-04-04T13:15:22.876Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 2 - 4 - 129 - [0,0,23,0,253,147,1,10,0,42,0,147,246,70,0,0,6,25,0,8,7,0,0,253,147,29] - 199
zigbee2mqtt      | 2021-04-04T13:15:22.877Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":23,"srcaddr":37885,"srcendpoint":1,"dstendpoint":10,"wasbroadcast":0,"linkquality":42,"securityuse":0,"timestamp":4650643,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[25,0,8,7,0,0]}}
zigbee2mqtt      | 2021-04-04T13:15:22.878Z zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":23,"data":{"type":"Buffer","data":[25,0,8,7,0,0]},"address":37885,"endpoint":1,"linkquality":42,"groupID":0,"wasBroadcast":false}'
zigbee2mqtt      | Zigbee2MQTT:debug 2021-04-04 13:15:22: Received Zigbee message from '0x00124b000cc8d641', type 'raw', cluster '23', data '{"data":[25,0,8,7,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
[... cut ...]
zigbee2mqtt      | 2021-04-04T13:15:26.087Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt      | Zigbee2MQTT:info  2021-04-04 13:15:26: MQTT publish: topic 'zigbee2mqtt/0x00124b000cc8d641', payload '{"action":"button_4_click","linkquality":0}'
zigbee2mqtt      | Zigbee2MQTT:info  2021-04-04 13:15:26: MQTT publish: topic 'zigbee2mqtt/0x00124b000cc8d641', payload '{"action":"","linkquality":0}'
zigbee2mqtt      | Zigbee2MQTT:info  2021-04-04 13:15:26: MQTT publish: topic 'zigbee2mqtt/0x00124b000cc8d641/action', payload 'button_4_click'
```

Koenkk, 2021-04-06:
@kirovilya since you added this device in https://github.com/Koenkk/zigbee-herdsman-converters/pull/735 , do you have any idea? I guess we have to sniff traffic with original gateway: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

kirovilya, 2021-04-09:
I was still unable to get the data from the bottom buttons. the first four are working fine.

kirovilya, 2021-04-09:
I didn't sniff traffic

kpeu3i, 2021-04-17:
Is it possible to somehow identify these buttons without traffic sniffing?

kpeu3i, 2021-05-23:
any ideas?

closed 2021-07-02 by github-actions[bot]

SAOPP, 2022-09-01:
This devices is worked for me when I was at cc2531, at zigstar poe lan with z2m builds from july to today (1st Sep) device is all time is not fully interviewed... Any ideas, it's a bug or what?