Koenkk/zigbee2mqtt #6918: Devices lost after upgrade

ai-xyz, 2021-04-01:
## What happened
Hi i've upgraded from 1.18.1 to 1.18.2 and three of my bulbs disappeared. They are still listed in the configuration.yaml, but they are not visible anymore in the Frontend, HA or when i subcribe to the topic zigbee2mqtt/#
I already tried downgrading, but they are still lost. It was an AC03647 and two AC03642. But not all of my AC03647 and AC03642 disappeared, only those three. I was able to pair them again without altering the configuration.yaml

## What did you expect to happen

## How to reproduce it (minimal and precise)
Upgrade from 1.18.1 to 1.18.2

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

ai-xyz, 2021-04-01:
I already tested one of the lamps which is in a group together with another lamp that is still visible and i can change the groups brightness and it works even though one of the two bulbs is not visible.

nurikk, 2021-04-02:
Hi!
Can you attach complete state.json file from `Settings -> Tools -> Download state` ?

code-in-progress, 2021-04-02:
Same problem here. Just updated to 1.18.2. Here is my states.json file:

[z2m_state.zip](https://github.com/Koenkk/zigbee2mqtt/files/6250494/z2m_state.zip)

closed 2021-05-03 by nurikk