Koenkk/zigbee2mqtt #6919: missing device_mode for Philips Hue Wall Switch Module (929003017102)

obelix05, 2021-04-01:
## What happened

I want to add the Philips Hue Wall Switch Module ([929003017102](https://www.zigbee2mqtt.io/devices/929003017102.html)) to my zigbee2mqtt network and after that I wanted to change the `device_mode` of the device to `single_push_button` but this doesn't work. After checking the logs I found out that the device doesn't respond on the request i try to send.
Also the `device_mode` is not part of the published state from the device in the zigbee2mqtt log.

## What did you expect to happen

The `device_mode` should be shown and changed.

## How to reproduce it (minimal and precise)

1. Add the device to zigbee2mqtt
2. wait until the pairing is finished
3. open home assistant mqtt console and listen to topic `zigbee2mqtt/living_room_wall_switch/#`
4. publish topic `zigbee2mqtt/living_room_wall_switch/get` with payload `{"device_mode": ""}`
5. check the zigbee2mqtt docker log and wait for the error message: https://pastebin.com/raw/dr67qeGx
6. try to change the device_mode to `single_push_button` and publish topic `zigbee2mqtt/living_room_wall_switch/set` with payload `{"device_mode": "single_push_button"}`
7. check the zigbee2mqtt docker log and wait for the error message: https://pastebin.com/raw/WfeXMUnK

## Debug info
Zigbee2MQTT version: 1.18.2 (Docker)
Adapter hardware: Conbee II
Adapter firmware version: 0x26660700

Could someone help me please?

alexryd, 2021-04-18:
I think you need to wake up the device to get a response. Try toggling the switch right after you send a `get` or `set` request.

RubenN, 2021-04-20:
Yes, @alexryd is right: my two switches work perfectly.

obelix05, 2021-05-06:
Ok, so now I got them to change their device_mode. But I'm a bit confused. I want to use the Wall Switch as push button, so I want to get the information "Is switch pushed": "on" or "off". In my setup it doesn't matter if the device mode is "single_push"button" or "single_rocker", if I push the button the device starts with the answer: left_press, continues with left_hold and after a few minutes it says left_hold_release and so my light will turn off.
Do you have any ideas what could cause this behaviour?

obelix05, 2021-06-08:
issue is still valid and the device_mode still doesn't work for me

obelix05, 2021-07-10:
I still don’t get the device_mode working. The switch is currently configured as single_push_Button and if I press the button I get the „_pressed“ and 1 second later the „_pressed_release“ message. If it is configured as single_rocker I get multiple times the „_pressed“ message and after a few minutes the „_pressed_release“ message come up and then there is no more communication. Is this the right behaviour?

closed 2021-08-19 by github-actions[bot]