Koenkk/zigbee2mqtt #6921: Livarno Lux E27 bulb RGB UNSUPPORTED_ATTRIBUTE when try restore color on expose

shirou93, 2021-04-02:
## What happened
[image]
When try restore color got this error:

(Propably this issue have all Lidl livarno bulbs)

Zigbee2MQTT:error 2021-04-02 07:02:30: Publish 'get' 'color' to 'Światło Pokój' failed: 'Error: Read 0x842e14fffe1b95c6/1 lightingColorCtrl(["currentX","currentY","enhancedCurrentHue","currentSaturation"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

## What did you expect to happen
Restore color 
## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.18.2-1
Adapter hardware: CC2652RB
Adapter firmware version: CC2652RB_coordinator_20210120.zip

sjorge, 2021-04-02:
Can you try using the read convertor to see if it is enhancedCurrentHue that is causing the issue, the bulb might only support the non enhanced version.

(Make sure to use your friendly_name)
```
mosquitto_pub -t zigbee2mqtt/0x14b457fffe2bd760/set/read -m '{"attributes": ["enhancedCurrentHue"], "cluster": "lightingColorCtrl"}'
 ```

It only logs to debug logging, so you need to enable that too, if that fails you can try reading the non enhanced variant.

```
mosquitto_pub -t zigbee2mqtt/0x14b457fffe2bd760/set/read -m '{"attributes": ["currentHue"], "cluster": "lightingColorCtrl"}'
```

shirou93, 2021-04-02:
bash: mosquitto_pub: command not found - container in HomeAssistant

nurikk, 2021-04-03:
> Can you try using the read convertor to see if it is enhancedCurrentHue that is causing the issue, the bulb might only support the non enhanced version.
>
> (Make sure to use your friendly_name)
>
> ``` [...]

hey guys, @sjorge @shirou93 I've implemented UI for this api, try to update on latest dev branch

@sjorge I would like to hear your feedback for future improvements
 
[image]

sjorge, 2021-04-03:
@nurikk seems to work well enough, I did notice a minor issue with the results not being cleared when doing a new read.
e.g. read genBasic.zlcVersion, then read OnOff.onOff both results are displayed below. Aside from that, it's very usable.

I'm not sure we should show this by default though, it was only ever intended to help us (us being developers) to get a way to have a user test something in a quick and easy way. cc: @Koenkk any opinion on this? Should this be hidden after a configuration property? Or just have it show up for everyone?

shirou93, 2021-04-03:
Hey guys, we getting off the point ... How do I use this tool to help fix the issue?

sjorge, 2021-04-03:
> Hey guys, we getting off the point ... How do I use this tool to help fix the issue?

If you can get on the latest dev release, you should now be able to use the frontend and try to reed this the attribute mentioned earlier.

[image]

Is what it looks like when it's not supported, which I suspect is the case for the Livarno.

If it is support it will look like this (other hue value most likely though)

[image]

shirou93, 2021-04-03:
> If you can get on the latest dev release, you should now be able to use the frontend and try to reed this the attribute mentioned earlier.
>
> [image]
>
> Is what it looks like when it's not supported, [...]

what exactly type on textbox attribute?

sjorge, 2021-04-03:
> what exactly type on textbox attribute?

It's actually a drop down, you select 'enhancedCurrentHue'

shirou93, 2021-04-03:
> > what exactly type on textbox attribute?
> 
> It's actually a drop down, you select 'enhancedCurrentHue'

[image]

sjorge, 2021-04-03:
Cluster => LColorCtrl
Attribute => enhancedCurrentHue

shirou93, 2021-04-03:
> Cluster => LColorCtrl
> Attribute => enhancedCurrentHue

Thx. 
[image]

sjorge, 2021-04-03:
I opened a PR that marks enhancedHue as unsupported for this device, it should fall back to reading the non enhanced variant. https://github.com/Koenkk/zigbee-herdsman-converters/pull/2445

shirou93, 2021-04-03:
All Lidl bulbs? Propably all this bulbs have same firmware

sjorge, 2021-04-03:
Does the E14 get the same result when reading the attribute manually? If so I’ll update the PR.

I could indeed be all Lidl bulbs, but just making that assumption is risky.

shirou93, 2021-04-03:
> Does the E14 get the same result when reading the attribute manually? If so I’ll update the PR.
> 
> I could indeed be all Lidl bulbs, but just making that assumption is risky.

On Lidl  online store this bulbs is one product with another thread. I'm pretty sure this is same  supported atributes.

shirou93, 2021-04-03:
[image]

sjorge, 2021-04-03:
@Koenkk are you OK with applying this to all RGB bulbs from Lidl?

Koenkk, 2021-04-04:
@sjorge yes, they probably share the same fw.

shirou93, 2021-04-06:
@sjorge  thanks for fix. I have one more question. This issue causes change color bulp in home assistant ui on default choised from zigbee2mqtt? When i change different color on HomeAssistant UI after a while in UI HomeAssistant color change to last choised in zigbee2mqtt ui (bulb no change color phisicly). It's wery strange
[image]

sjorge, 2021-04-06:
I’m not using homeassisant so I’m not sure what the problem could be.

shirou93, 2021-04-06:
@sjorge I tryed only on z2m ui and looks like when change RGB color and after change color temp, color rgb last state rgb return to ui without change state phisicly in bulp. 
[image]

shirou93, 2021-04-09:
@sjorge SMART+ CLASSIC MULTICOLOUR 60 10W E27 (AC25697) have same issue 
"when change RGB color xy and after change color temp, color rgb xy last state rgb return to ui without change state phisicly in bulb."

sjorge, 2021-04-09:
Can you read the same attribute like for the lidl one? If it shows the same log entry it’s an easy fix.

shirou93, 2021-04-09:
I thought it was related.  Ledvance  not have issue UNSUPPORTED_ATTRIBUTE.

sjorge, 2021-04-09:
There are currently a few know issues with color_temp/xy/hs not being in sync when changing one.
There are 2 PR open to address this, one is making some good progress the other is on hold waiting for the other one.

It's a not so trivial issue is a mix of read/report/set and each play a part, and it's not one size fits all unfortunately.

shirou93, 2021-04-09:
@sjorge  thanks for your help.

closed 2021-04-09 by shirou93