Koenkk/zigbee2mqtt #6943: Logic issue - 2 x Hue new remotes bound to a single group

877dev, 2021-04-03:
Hi,

I have two Hue lamps (9290022166 ) in a group called "Stairs_lamps", and have bound 2 x Hue v2 dimmers (929002398602) to that group (one for upstairs, one for downstairs).

The Hue dimmer v2 (929002398602) has four buttons:

- Button 1 - power on/power off
- Button 2 - brightness up
- Button 3 - brightness down
- Button 4 - actions/scenes

This causes the following to occur:

When I press dimmer#1 (button1), both lamps in the group turn ON.
When I press dimmer#2 (button1), both lamps stay ON.

Obviously caused as both dimmers are sending the ON command first.

Can I change this so that the behavior toggles instead?

i.e..
Press dimmer#1 (button1), both lamps in the group turn ON.
Press dimmer#2 (button1), both lamps in the group turn OFF.

877dev, 2021-04-04:
Tried to add a converter for the Hue Dimmer v2, changing the `on` to `toggle` in fromZigbee.js:

```
    hue_dimmer_switch_v2: {
        cluster: 'manuSpecificPhilips',
        type: 'commandHueNotification',
        convert: (model, msg, publish, options, meta) => {
            const buttonLookup = {1: 'toggle', 2: 'up', 3: 'down', 4: 'off'};
            const button = buttonLookup[msg.data['button']];
            const typeLookup = {0: 'press', 1: 'hold', 2: 'press_release', 3: 'hold_release'};
            const type = typeLookup[msg.data['type']];
            const payload = {action: `${button}_${type}`};

            // duration
            if (type === 'press') globalStore.putValue(msg.endpoint, 'press_start', Date.now());
            else if (type === 'hold' || type === 'release') {
                payload.action_duration = (Date.now() - globalStore.getValue(msg.endpoint, 'press_start')) / 1000;
            }

            // simulated brightness
            if (options.simulated_brightness && (button === 'down' || button === 'up') && type !== 'release') {
                const opts = options.simulated_brightness;
                const deltaOpts = typeof opts === 'object' && opts.hasOwnProperty('delta') ? opts.delta : 35;
                const delta = button === 'up' ? deltaOpts : deltaOpts * -1;
                const brightness = globalStore.getValue(msg.endpoint, 'brightness', 255) + delta;
                payload.brightness = numberWithinRange(brightness, 0, 255);
                globalStore.putValue(msg.endpoint, 'brightness', payload.brightness);
            }

            return payload;
        },
```

Here is the devices.js section:

```
 {
        zigbeeModel: ['RWL022'],
        model: '929002398602',
        vendor: 'Philips',
        description: 'Hue dimmer switch',
        fromZigbee: [fz.ignore_command_on, fz.ignore_command_off, fz.ignore_command_step, fz.ignore_command_stop,
            fz.hue_dimmer_switch_v2, fz.battery],
        exposes: [e.battery(), e.action(['on_press', 'on_hold', 'on_press_release', 'on_hold_release',
            'off_press', 'off_hold', 'off_press_release', 'off_hold_release', 'up_press', 'up_hold', 'up_press_release', 'up_hold$
            'down_press', 'down_hold', 'down_press_release', 'down_hold_release'])],
        toZigbee: [],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'manuSpecificPhilips', 'genPowerCfg'$
            const options = {manufacturerCode: 0x100B, disableDefaultResponse: true};
            await endpoint.write('genBasic', {0x0031: {value: 0x000B, type: 0x19}}, options);
            await reporting.batteryPercentageRemaining(endpoint);
        },
    },

```

However it does not work, although zigbee logs show the toggle_press & toggle_release, and I am getting UnhandledPromiseRejectionWarning' warnings. I guess there is a little more to it than that....

Koenkk, 2021-04-06:
Changing something in Z2M (e.g. `fromZigbee.js`) does not change the behaviour when the device is being bound to a group. This is determined by the firmware of the hue remote.

877dev, 2021-04-07:
Ah I see, so there is no way to fix this for now. Maybe a request to Philips might be the only option?

Koenkk, 2021-04-07:
Only a FW update can change the behaviour of the buttons indeed

closed 2021-05-15 by github-actions[bot]