Koenkk/zigbee2mqtt #6944: WooX R7060 not supported

qbahc, 2021-04-03:
Hi everyone,

Have some problem with WOOX R7060. It's already added to Zigbee2MQTTnbut I see it as an unsupported device.
So I do not see it in HA as an entity.

I already checked Zigbee2MQTT website and WOOX r7060 should be supported.

What I do wrong?

CC2531 and the newest sort of HA and Zigbee2MQTT

Thx in advance

Have a nice day!

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version:

qbahc, 2021-04-03:
Sorry, checked again. Have Zigbee2MTQQ 1.18.1-1. 
But can't do update via HA.

How can I do it?

Thx.

qbahc, 2021-04-06:
Done by Zigbee2MQTT Edge.

closed 2021-04-06 by qbahc

driek12, 2022-05-19:
I have the same issue with this device. Im using a sonof zifgbee 3 dongle. How did you fix it?