Koenkk/zigbee2mqtt #6945: Hey Good afternoon! everything stopped here again at home, all devices no longer respond, and this error occurs when triggering

vpereiradn, 2021-04-03:
Hey Good afternoon!
everything stopped here again at home, all devices no longer respond, and this error occurs when triggering

Zigbee2MQTT:error 2021-04-03 12:50:14: Publish 'set' 'state' to 'Lampada area de Lazer' failed: 'Error: Command 0x804b50fffeb34c4d/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))

kashesandr, 2021-04-05:
Experiencing the same

vpereiradn, 2021-04-05:
nao entendi!!

kashesandr, 2021-04-06:
I had to reconnect all the devices after the zigbee2mqtt upgrade...

vpereiradn, 2021-04-06:
I ended up doing it too, I had to pair all devices here at home, now everything is working perfectly

kevincaradant, 2021-04-12:
Hi

Is it normal @Koenkk ? I'm currently hesitate to upgrade on 1.18.2 in this case. I have a lot of devices and some are difficulties to reach physically :)

thank you

kevincaradant, 2021-04-12:
Alright thank you for your answer :). I was scared to miss some breaking changes on this version.

closed 2021-05-20 by github-actions[bot]