Koenkk/zigbee2mqtt #6949: Impossilbe to restore: database.db overwritten to null at container start

masantiago, 2021-04-04:
## What happened
Database.db is always reset to empty at Docker migration and it is imposible to restore previous database. However, it was persisting correctly while working with my previous Docker.
## What did you expect to happen
Recover my original database
## How to reproduce it (minimal and precise)
Same issue as https://github.com/Koenkk/zigbee2mqtt/issues/4687 and still happens. I don't know how to restore my database.
## Debug info
Zigbee2MQTT version: 1.17 and 1.18
Adapter hardware: CC2531

Koenkk, 2021-04-04:
Make sure to also restore your configuration.yaml and coordinator_backup.json file. Also don't change your network key, channel or pan ID since this will reinit z2m and clear the database.db

masantiago, 2021-04-04:
Thank you very much @Koenkk . I have just succeeded it and I suppose what's going on. Let me explain if it helps anybody.

My Z2M container comes from a [Kubernetes chart](https://github.com/k8s-at-home/charts/tree/master/charts/stable/zigbee2mqtt). It is embedded inside a POD of a K3S deployment. The folder /data is mounted in a PVC. So far, so good. I have noticed that everytime the POD goes down (because I restart it), Z2M container updates the database.db with the latest data of the service that probably is located in memory. 

For the database recovery, I was updating the database.db **before** restarting the POD, then Z2M always was overwritting with the data in memory. Therefore, I have done it in a different manner. Instead of restarting a POD by killing it, I deleted the Deployment and then the POD is removed. Later, I replaced the /data folder with my backup. And, finally, I redeployed the Chart (helm upgrade) that is creating a new deployment but using the updated PVC.

Now, the UI is successfully filled out with my backup 😄

closed 2021-04-04 by masantiago