Koenkk/zigbee2mqtt #6959: Samsung Motion Sensor IM6001-MTP01 not reporting temperature

ssmall, 2021-04-04:
## What happened

I had to re-pair all of my Zigbee sensors due to a data loss during the upgrade to version 1.18.2 (root cause of the data loss was user error, not related to this bug report). Previously, this sensor was reporting all values including temperature without issue. However, after re-pairing the sensor it no longer reports temperature reading. I have at least 5 other identical motion sensors that all got re-paired the same way and none of them are having this issue.

I have noticed that sometimes the temperature appears in the state of the device in the Zigbee2MQTT frontend, but never gets published to the MQTT topic and usually disappears when I refresh the page.

## What did you expect to happen

After re-pairing the device, the temperature should be reported as normal.

## How to reproduce it (minimal and precise)

1. remove device from Zigbee2MQTT
2. re-add device

## Debug info
Zigbee2MQTT version: 1.18.2
Adapter hardware: CC2652RB stick (zStack3x0)
Adapter firmware version: 20210120

Debug logs (look for device ID `0x286d970001094ca4`): [zigbee2mqtt.log](https://github.com/Koenkk/zigbee2mqtt/files/6255049/zigbee2mqtt.log)

ssmall, 2021-04-04:
This is what the state of the device looks like on the frontend:

[image]

Notice that `temperature` is missing. Compared to the state of a different sensor of the exact same model:

[image]

ssmall, 2021-04-04:
I just tried removing and re-adding another device of the same model and I'm seeing the same behavior, so I'm concerned this is an issue with the new version of Zigbee2MQTT and not a problem with the devices.

ssmall, 2021-04-04:
Seeing the same issue when I downgrade to the previous version I was running, 1.17.1.

ssmall, 2021-04-23:
@Koenkk can you please advice here? This issue is preventing me from updating to the latest version and could represent a regression for Samsung motion sensors.

Koenkk, 2021-04-23:
I can't find any recent changes to this. Can you press the yellow reconfigure button (you can find it by clicking on the device in the frontend)? Make sure to wakeup the device before doing that.

ssmall, 2021-05-03:
This is no longer an issue with the latest version, 1.18.3. Closing the issue.

closed 2021-05-03 by ssmall