Koenkk/zigbee2mqtt #6966: CC2652RB: "Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)"

bertinets, 2021-04-05:
I'm opening an issue here, not sure if it's related to this one:
https://github.com/Koenkk/zigbee2mqtt/issues/6517

The adapter I have and the error I get are the same as in that issue. However, as it seems that that issue was solved from 1.18.0 and I am running 1.18.2, which is supposed to be working, the issue may be related to my environment.

## What happened
I have just bought a CC2652RB from Slaesh which I connected to a RPi (running Ubuntu 18.04.5 LTS). 
The serial port of the device is exposed to the network using ser2net.
I run a docker image of Zigbee2MQTT (v. 1.18.2) with socat (https://hub.docker.com/r/forepe/zigbee2mqtt-socat) configured with minimal settings.

typical log here: https://pastebin.com/vTNSd24F

Running 'python3 ./test.py' on the RPi returns:

Opening port /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_4F_3F_0E-if00-port0, baud 115200

--> Bootloader is working!
--> ZigBee coordinator is working!
`---> transportrev 2, product 1, majorrel 2, minorrel 7, maintrel 1, revision: 20200715

Check done!

with the debug FW from Slaesh's website OR

Opening port /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_4F_3F_0E-if00-port0, baud 115200

--> Bootloader is working!
--> ZigBee coordinator is working!
`---> transportrev 2, product 1, majorrel 2, minorrel 7, maintrel 1, revision: 20210120

with te newest FW

Running 'test -w /dev/ttyUSB0 && echo success || echo failure' returns
success

## What did you expect to happen
I have another docker image with embedded socat that connects to a CC2531 connected to the same RPi. With that adapter everything works absolutely fine, so I would expect Zigbee2MQTT to work as well with Slaesh's CC2652RB.

## How to reproduce it (minimal and precise)
Zigbee2MQTT docker image: https://hub.docker.com/r/forepe/zigbee2mqtt-socat

My configuration.yaml:

homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://10.42.197.8'
serial:
  port: /dev/zigbee
advanced:
  pan_id: 57563

ser2net version 2.10.1

## Debug info
Zigbee2MQTT version: 1.18.2
Adapter hardware: CC2652RB
Adapter firmware version: both the 20210120 or the debug version of Slaesh

skyw33, 2021-04-05:
Not sure if this is related, but after upgrading to 1.18.2 I can only access my device if I set it up as /dev/ttyACM0.  If I use /dev/ttyUSB-zigbee which is a softlink to /dev/ttyACM0, then it can't find it.  I have udev rules setup because I also have a zwave stick and my system randomly assigns the ACM0 or ACM1 addresses.  Here are my rules:

SUBSYSTEM=="tty", ATTRS{idVendor}=="0451", ATTRS{idProduct}=="16a8", ATTRS{seria
l}=="__0X00124B00193B3ADF", SYMLINK+="ttyUSB-zigbee"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0658", ATTRS{idProduct}=="0200", ATTRS{seria
l}=="01FFFFFF-FFFF-FFFF-FFFF-100915210555", SYMLINK+="ttyUSB-zwave"

I'm using docker on a synology NAS.  Firmware version is zStack12 on a generic Zigbee CC2531 USB Dongle.

bertinets, 2021-04-05:
Thanks for the hint, but I guess this is probably not related: maybe it's because of my configuration (i.e. ser2net - socat), but with Z2M 1.18.2 I can access my CC2531 USB dongle with both 
/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00014D90A2-if00
and 
/dev/ttyACM1

But I guess this is related to the fact that, within docker, the serial port used in both cases on my server is the same, i.e. /dev/zigbee. As I understand it, this the virtual port created by socat to listen to the physical port on the RPi shared by ser2net over the network. 

for my cc2652rb neither 
/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_4F_3F_0E-if00-port0
nor 
/dev/ttyUSB0
works.

thebencourt, 2021-04-08:
I'm getting the same with a CC2531 after the latest update.
I've tried running via docker and bare metal (Debian 10) on a raspberry pi 4. If I restart it will occasionally work but I'm then getting timeouts and other issues.

```
Zigbee2MQTT:error 2021-04-08 16:38:14: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
```

bertinets, 2021-04-10:
I realised that I don't need a Z2M container with embedded socat because Z2M is able to connect to a remote adapter itself, so I'm closing this issue. I'll open a new one, though, because with the original Z2M docker container (koenkk/zigbee2mqtt) I have the very same problem.

closed 2021-04-10 by bertinets