Koenkk/zigbee2mqtt #6967: MFKZQ01LM Aqara cube disappears from UI dashboard

rom169, 2021-04-05:
## What happened
Cube disappears from UI dashboard when receiving "heartbeat" without movement. 
This is caused by missing action attribute in payload:
`info  2021-03-29 22:44:37: MQTT publish: topic 'zig2mqtt/LivingCubeXiaomi', payload '{"angle":34,"battery":100,"last_seen":1617050677361,"linkquality":57,"side":2,"voltage":3005}'`

Payload with action attribute makes it appear again:
`info  2021-03-29 23:23:48: MQTT publish: topic 'zig2mqtt/LivingCubeXiaomi', payload '{"action":"wakeup","angle":34,"battery":100,"last_seen":1617053028576,"linkquality":57,"side":2,"voltage":3005}'`

## What did you expect to happen
Cube stays in the dasboard, e.g. by always sending action

## How to reproduce it (minimal and precise)
Do not move the cube and wait for "heartbeat"

## Debug info
Zigbee2MQTT version: 1.18.2
Adapter hardware: CC2652RB
Adapter firmware version: 20210120

nurikk, 2021-04-06:
Dashboard is ment to show only devices which can be controlled(lights, relays) or publish some constantly updating values like sensors. Switches should not be visible in dashboard

closed 2021-04-06 by Koenkk

rom169, 2021-04-06:
> Dashboard is ment to show only devices which can be controlled(lights, relays) or publish some constantly updating values like sensors. Switches should not be visible in dashboard 

That means battery level, network signal and cube movements are not considered constantly updating values? If so, why does the device appear in UI when payload includes action attribute?