Koenkk/zigbee2mqtt #6973: Exposes doesn't have capability for threshold and timer

ArJay60, 2021-04-06:
Im trying to add a new type device, a water valve. I have the valve working but I am not able to add capabilities except state (e.switch). Is it possible to add
1) expose for threshold. The percentage from 0 - 100 that the valve is open.
2) expose for a downtimer. Range from 0 - 86400 seconds. Valve report every 30 secs when timer is set its timer status together with other info (see below)

Below the output from Zigbee2MQTT when running a timer with a theshold. Valve is reporting every 30 seconds it's timer state

Zigbee2MQTT:info  2021-04-06 15:49:13: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":30,"state":"ON","threshold":80,"timer":30}'
Zigbee2MQTT:info  2021-04-06 15:49:13: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":30,"state":"ON","threshold":80,"timer":30}'
Zigbee2MQTT:info  2021-04-06 15:49:13: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":33,"state":"ON","threshold":80,"timer":30}'
Zigbee2MQTT:info  2021-04-06 15:49:13: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":30,"state":"ON","threshold":80,"timer":30}'
Zigbee2MQTT:info  2021-04-06 15:49:43: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":33,"state":"OFF","threshold":80,"timer":30}'
Zigbee2MQTT:info  2021-04-06 15:49:43: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":33,"state":"OFF","threshold":80,"timer":30}'
Zigbee2MQTT:info  2021-04-06 15:49:43: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":33,"state":"OFF","threshold":80,"timer":30}'
Zigbee2MQTT:info  2021-04-06 15:49:43: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":33,"state":"OFF","threshold":80,"timer":30}'
Zigbee2MQTT:info  2021-04-06 15:49:43: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":33,"state":"OFF","threshold":80,"timer":30}'
Zigbee2MQTT:info  2021-04-06 15:49:44: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":33,"state":"OFF","threshold":80,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:44: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":33,"state":"OFF","threshold":80,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:44: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":33,"state":"OFF","threshold":80,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:44: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":33,"state":"OFF","threshold":80,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:44: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":30,"state":"OFF","threshold":80,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:44: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":30,"state":"OFF","threshold":80,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:44: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":30,"state":"OFF","threshold":80,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:45: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":33,"state":"OFF","threshold":80,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:45: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":33,"state":"OFF","threshold":80,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:45: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":30,"state":"OFF","threshold":0,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:45: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":30,"state":"OFF","threshold":0,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:45: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":30,"state":"OFF","threshold":0,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:46: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":30,"state":"OFF","threshold":0,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:46: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":33,"state":"OFF","threshold":0,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:46: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":30,"state":"OFF","threshold":0,"timer":0}'
Zigbee2MQTT:info  2021-04-06 15:49:46: MQTT publish: topic 'zigbee2mqtt/Valve', payload '{"linkquality":33,"state":"OFF","threshold":0,"timer":0}'

sjorge, 2021-04-07:
You could do this for the threshold, it what is used for various  TRV's valve position

```
exposes.numeric('threshold', exposes.access.ALL).withValueMin(0).withValueMax(100).withDescription('Valve open percentage'),
```

update axpose.access.ALL to whatever it should be.

I guess you could do something similar for the timer.

ArJay60, 2021-04-08:
I have made the following additions to exposes.js

class Valve extends Base {
    constructor() {
        super();
        this.type = 'valve';
        this.features = [];
        this.features.push(new Binary('state', access.ALL, 'ON', 'OFF').withValueToggle('TOGGLE').withDescription('Open/Clase state of this valve'));
    }

    withThreshold() {
        assert(!this.endpoint, 'Cannot add feature after adding endpoint');
        this.features.push(new Numeric('threshold', access.ALL).withValueMin(0).withValueMax(100).withDescription('Threshold of this valve'));
        return this;
    }
}

in the section module.exports added a line:

valve: () => new Valve(),

in the section under presets in the same section added a line:

valve: () => new Valve().withThreshold(),

Finally added the next line in devices.js to my device:

    exposes: [e.switch(), e.valve()],

But still it doesn't work for threshold!! Can you provide some extra help???

Below you can also find my additions to and from zigbee converters (I use a local file instead of adding directly to the converters files:

const fzLocal = {
    fe_valve: {
        cluster: 'manuSpecificTuya',
        type: ['commandGetData', 'commandSetDataResponse'],
        convert: (model, msg, publish, options, meta) => {
            const value = tuya.getDataValue(msg.data.datatype, msg.data.data);
            const dp = msg.data.dp;
            switch (dp) {
            case tuya.dataPoints.state: {
                return {state: value ? 'ON': 'OFF'};
            }
            case tuya.dataPoints.fethreshold: {
                return {threshold: value};
            }
            case tuya.dataPoints.fetimer: {
                return {timer: value};
            }
            default: { // TODO: Unknown dp, assumed value type
                //const normalised = (value - 10) / (1000 - 10);
                //return {unknown: value};
                meta.logger.warn(`zigbee-herdsman-converters:FrankeverValve: NOT RECOGNIZED DP #${
                dp} with data ${JSON.stringify(msg.data)}`);
            }
        }
        },
    },
};

And 

const tzLocal = {
    frankever_threshold: {
    key: ['threshold'],
    convertSet: async (entity, key, value, meta) => {
        //input to multiple of 10 with max value of 100
        const thresh = Math.min(10*(Math.floor(value/10)), 100);
        // sendTuyaDataPoint* functions take care of converting the data to proper format
        await tuya.sendDataPointValue(entity, tuya.dataPoints.fethreshold, thresh, 'setData', 1);
    },
    },
    frankever_timer: {
    key: ['timer'],
    convertSet: async (entity, key, value, meta) => {
        //input in seconds with maximum of 86400s or 24 hours
        const timer = Math.min(value, 86400);
        // sendTuyaDataPoint* functions take care of converting the data to proper format
        await tuya.sendDataPointValue(entity, tuya.dataPoints.fetimer, timer, 'setData', 1);
    },
    },
};

sjorge, 2021-04-08:
You don't need to create the whole Valve() class as this is just one device, others might not use the same terminology.

```
exposes: [
     e.switch(),
     exposes.numeric('threshold', exposes.access.ALL).withValueMin(0).withValueMax(100).withDescription('Valve open percentage'),
],
```

Or if you keep the Valve class the expose entry should like like:

```
exposes: [ e.valve().withThreshold()],
```

But in this case I'd make the valve class more generic:

```
class Valve extends Base {
    constructor() {
        super();
        this.type = 'valve';
        this.features = [];
        this.features.push(new Binary('state', access.ALL, 'on', 'off').withValueToggle('toggle').withDescription('Open/Close state of this valve'));
    }

    withPosition(name) {
        assert(!this.endpoint, 'Cannot add feature after adding endpoint');
        this.features.push(new Numeric(name, access.ALL).withValueMin(0).withValueMax(100).withDescription('Valve postion'));
        return this;
    }
}
```

```
exposes: [e.valve().withPosition('threshold')],
```

ArJay60, 2021-04-08:
Better to not add what is not necessary. So I removed the class and with your additions its working fine for now. I have one question. I have set up the timer in the same manner as threshold (range 0 - 86400). If you want to set the seconds for intance with a slider in the Zigbee2MQTT Dashboard this is a very rude way of doing things because a slight move causes  a huge number of seconds more or less. Is there another way of doing this?  For instance with hours, minutes and seconds as separate values?

sjorge, 2021-04-08:
> If you want to set the seconds for intance with a slider in the Zigbee2MQTT Dashboard this is a very rude way of doing things because a slight move causes a huge number of seconds more or less. Is there another way of doing this? For instance with hours, minutes and seconds as separate values?
ping @nurikk

ArJay60, 2021-04-08:
Do you also have an idea how I can work on the get messages like the one below? I get them for state, threshold and timer.

No converter available for 'get' 'threshold' ()

nurikk, 2021-04-08:
> ping @nurikk

you can use larger units like minutes, 5*minutes, N* minutes
create as many exposes features as you want

sjorge, 2021-04-08:
> Do you also have an idea how I can work on the get messages like the one below? I get them for state, threshold and timer.
> 
> No converter available for 'get' 'threshold' ()

Yeah, you need to implement a convertGet, it' a bit confusing at first that it is located in toZigbee.js...
https://github.com/Koenkk/zigbee-herdsman-converters/blob/689d775e3b0b5fb48637e70bb7aeb16dfcaeb5e2/converters/toZigbee.js#L4804

That basically links a attribute and tells z2m how to read it. The fromZigbee.js side of things that  then takes the result handlest hat.

ArJay60, 2021-04-08:
I've got things working. No capabilities in device to get status. It's sending status itself during a running downtimer. I've created this device in an external converter file and added two lines to lib/tuya.js to make it working. How can I with some help convert this to be part of the supported devices??

> you can use larger units like minutes, 5_minutes, N_ minutes
> create as many exposes features as you want

Done this. Works for me now.

ArJay60, 2021-04-08:
I've got things working. No capabilities in device to get status. It's sending status itself during a running downtimer. I've created this device in an external converter file and added two lines to lib/tuya.js to make it working. How can I with some help convert this to be part of the supported devices??

ArJay60, 2021-04-09:
Created working solution.

closed 2021-04-09 by ArJay60