Koenkk/zigbee2mqtt #6978: Request help with new device, Tuya based temperature display

Peterpc58, 2021-04-06:
The device does not give a ModelID, it reports this:
Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
What do I fill in at ModelID and manufacturerName?
## Information about the device + link
Tuya ZigBee Smart Home Temperature And Humidity Sensor With LED Screen Works With Google Assistant and Tuya Zigbee Hub
https://nl.aliexpress.com/item/4000843004202.html?spm=a2g0s.9042311.0.0.27424c4d6KGtOH

## data/database.db entry of the device
How can I read this file? Tried DB browser for SQ lite, but have to fill in a Password.

kizio1000, 2021-04-18:
Same here. I have bought it but much to my surprise ( regarding facet that it look very similar to already supported devices ) it's not working with z2m.

vajpeters, 2021-05-11:
Yups, confirmed: same problem here. Old version was detected, I bought two new ones and guess what: new version. Any succes yet?

kizio1000, 2021-05-12:
After switching to edge I have support for this sensor.
[image]
[image]

vajpeters, 2021-05-12:
[image]

kizio1000, 2021-05-12:
Try pairing couple times. Yesterday I have moved from CC2531 to CC2652p and had to pair him 4 times before he was correctly interviewed.

vajpeters, 2021-05-12:
Ok, latest edge version solves the issue! Issue van be close!

closed 2021-05-12 by Koenkk