Koenkk/zigbee2mqtt #6979: SNZB-03 works with coordinator but not with router

manunited10, 2021-04-07:
Hi,
I just bought a Sonoff SNZB-03 and it connects and works with coordinator. But when I pair it to the router, it lists it as unsupported. Can anyone help me on this please. I tried several times by connecting/disconnecting removing the battery, even re-flashing the router but no success.
(I know there's a topic explaining how to introduce a new unsupported device. But I thought this couldn't be the case here as connect-to-coordinator method works)

Zigbee2MQTT version: 1.18.2
Coordinator: CC2531 with revision 20201127
Router: CC2530 with firmware 2018-09-26 (should be the latest, no idea why date is 2018)

Thank you in advance.

Koenkk, 2021-04-07:
These devices can be tricky to pair, especially when done through router. I recommend to pair it with the coordinator and after that moved it to the desired location. The SNZB-03 should automatically switch over to a router when the connection is lost with the coordinator.

manunited10, 2021-04-07:
> These devices can be tricky to pair, especially when done through router. I recommend to pair it with the coordinator and after that moved it to the desired location. The SNZB-03 should automatically switch over to a router when the connection is lost with the coordinator.

Thank you for your quick reply. I will try this solution and report the result here.

manunited10, 2021-04-19:
Pairing with coordinator and let it move to router by itself. worked. At the moment it's connected to both the coordinator and the router at the same time, and it's working. Thanks Koenkk.

closed 2021-04-19 by manunited10