Koenkk/zigbee2mqtt #6980: Adding support for RGB Genie 5 Zone Remote ZB-5001

tels7ar, 2021-04-07:
This remote appears similar to some of the other Sunricher / RGB Genie models.  For example, there is already a supported model ZGRC-KEY-013 RGB Genie remote which has 3 zones.  This remote reports as ZGRC-KEY-012.

I tried copying the config for the 013 model and just changing the identifiers from 013 to 012, i.e.:
```
    {
        zigbeeModel: ['ZGRC-KEY-012'],
        model: 'ZGRC-KEY-012',
        vendor: 'RGB Genie',
        description: '5 Zone remote and dimmer',
        fromZigbee: [fz.battery, fz.command_move, fz.legacy.ZGRC013_brightness_onoff,
            fz.legacy.ZGRC013_brightness, fz.command_stop, fz.legacy.ZGRC013_brightness_stop, fz.command_on,
            fz.legacy.ZGRC013_cmdOn, fz.command_off, fz.legacy.ZGRC013_cmdOff, fz.command_recall],
        exposes: [e.battery(), e.action(['brightness_move_up', 'brightness_move_down', 'brightness_stop', 'on', 'off', 'recall_*'])],
        toZigbee: [],
        meta: {configureKey: 1, multiEndpoint: true, battery: {dontDividePercentage: true}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint);
        },
    }
```
that results in partial success: the device is recognized, and the first 4 buttons work (master on, master off, zone 1 on, zone 1 off).  The master on and zone 1 on button both send action 'on_1', and the master off and zone 1 off button both send action 'off_1' However, pressing the rest of the buttons produces no results. Here's a sample log message:
```
Apr 07 05:01:34 odroid npm[7314]: Zigbee2MQTT:info  2021-04-07 05:01:34: MQTT publish: topic 'zigbee2mqtt/0x000d6f000fc74eb4', payload '{"action":"on_1","click":"1_on">
Apr 07 05:01:34 odroid npm[7314]: Zigbee2MQTT:info  2021-04-07 05:01:34: MQTT publish: topic 'zigbee2mqtt/0x000d6f000fc74eb4', payload '{"action":"","linkquality":0}'
Apr 07 05:01:34 odroid npm[7314]: Zigbee2MQTT:info  2021-04-07 05:01:34: MQTT publish: topic 'zigbee2mqtt/0x000d6f000fc74eb4', payload '{"click":"","linkquality":0}'
Apr 07 05:01:34 odroid npm[7314]: Zigbee2MQTT:info  2021-04-07 05:01:34: MQTT publish: topic 'zigbee2mqtt/0x000d6f000fc74eb4/action', payload 'on_1'
Apr 07 05:01:34 odroid npm[7314]: Zigbee2MQTT:info  2021-04-07 05:01:34: MQTT publish: topic 'zigbee2mqtt/0x000d6f000fc74eb4/click', payload '1_on'
```
Also, the configure fails to work:
```
error 2021-04-07 05:04:30: Failed to configure '0x000d6f000fc74eb4', attempt 3 (Error: ConfigureReporting 0x000d6f000fc74eb4/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51316 - 1 - 6 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/srv/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:67:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
```
I then tried a minimal config:
```
    {
        zigbeeModel: ['ZGRC-KEY-012'],
        model: 'ZGRC-KEY-012',
        vendor: 'RGB Genie',
        description: '5 Zone remote and dimmer',
        fromZigbee: [],
        exposes: [],
        toZigbee: [],
    },
```
which produced essentially the same results (master and zone 1 buttons act the same, no response from other buttons):
```
debug 2021-04-07 05:12:35: Received Zigbee message from '0x000d6f000fc74eb4', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2021-04-07 05:12:35: No converter available for 'ZGRC-KEY-012' with cluster 'genOnOff' and type 'commandOn' and data '{}'
debug 2021-04-07 05:12:38: Received Zigbee message from '0x000d6f000fc74eb4', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2021-04-07 05:12:38: No converter available for 'ZGRC-KEY-012' with cluster 'genOnOff' and type 'commandOff' and data '{}'
debug 2021-04-07 05:12:41: Received Zigbee message from '0x000d6f000fc74eb4', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2021-04-07 05:12:41: No converter available for 'ZGRC-KEY-012' with cluster 'genOnOff' and type 'commandOn' and data '{}'
debug 2021-04-07 05:12:43: Received Zigbee message from '0x000d6f000fc74eb4', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2021-04-07 05:12:43: No converter available for 'ZGRC-KEY-012' with cluster 'genOnOff' and type 'commandOff' and data '{}'
```
At this point I am unsure how to continue.  I've been trying to follow the troubleshooting steps in https://github.com/Koenkk/zigbee2mqtt/issues/642 for when you were assisting in getting the 013 model working, but I haven't had much luck with them.  For example, I tried https://github.com/Koenkk/zigbee2mqtt/issues/642#issuecomment-443692915 but I just get the message
```
Apr 07 04:36:49 odroid npm[6965]: Zigbee2MQTT:error 2021-04-07 04:36:49: Failed to configure '0x000d6f000fc74eb4', attempt 3 (TypeError: shepherd.find is not a function
Apr 07 04:36:49 odroid npm[6965]:     at Object.configure (/srv/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js:12769:37)
Apr 07 04:36:49 odroid npm[6965]:     at Configure.configure (/srv/zigbee2mqtt/lib/extension/configure.js:131:45)
Apr 07 04:36:49 odroid npm[6965]:     at Configure.onZigbeeEvent (/srv/zigbee2mqtt/lib/extension/configure.js:113:18)
Apr 07 04:36:49 odroid npm[6965]:     at Controller.callExtensionMethod (/srv/zigbee2mqtt/lib/controller.js:382:44))
```
Please let me know what I should try next to troubleshoot this.  I am particularly unclear on how I'm supposed to get the other buttons working.
## Information about the device + link
Product link for this device: https://rgbgenie.com/?product=rgbgenie-5-zone-remote-and-dimmer-zigbee

## data/database.db entry of the device
```
{"id":3,"type":"EndDevice","ieeeAddr":"0x000d6f000fc74eb4","nwkAddr":51316,"manufId":4644,"manufName":"Sunricher","powerSource":"Unknown","modelId":"ZGRC-KEY-012","epList":[1,2,3,4,5],"endpoints":{"1":{"profId":260,"epId":1,"devId":1,"inClusterList":[0,1,3,2821,4096],"outClusterList":[3,4,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"ZGRC-KEY-012","manufacturerName":"Sunricher","powerSource":0,"zclVersion":2,"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"","swBuildId":"2.2.3_r10"}},"genPowerCfg":{"attributes":{"batteryVoltage":30}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00193b174b","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00193b174b","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":1,"inClusterList":[0,1,3,2821,4096],"outClusterList":[3,4,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":1,"inClusterList":[0,1,3,2821,4096],"outClusterList":[3,4,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":1,"inClusterList":[0,1,3,2821,4096],"outClusterList":[3,4,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":1,"inClusterList":[0,1,3,2821,4096],"outClusterList":[3,4,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"","swBuildId":"2.2.3_r10","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1617770118729}
```

tels7ar, 2021-04-16:
@Koenkk any chance you will have some time to look this over and offer any suggestions?  I think with a bit more troubleshooting guidance I could probably get it working.

Koenkk, 2021-04-17:
If for each pressed button you can provided the logging we can fill the fromZigbee part. E.g. for the logging you provided now use:
```js
    {
        zigbeeModel: ['ZGRC-KEY-012'],
        model: 'ZGRC-KEY-012',
        vendor: 'RGB Genie',
        description: '5 Zone remote and dimmer',
        fromZigbee: [fz.command_on, fz.command_off],
        exposes: [],
        toZigbee: [],
        meta: {multiEndpoint: true}, // <-- not sure if this is needed, this will and the endpoint number to the `action`, first try without it and see if each zone generates a unique action.
    },
```

closed 2021-05-25 by github-actions[bot]

drhaber, 2021-06-14:
Recently purchased this device on [amazon](https://www.amazon.com/gp/product/B07GBK3KPS/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1).

[Documentation ](https://www.zigbee2mqtt.io/devices/ZB-5004.html)is labeled 

ZB-5001 and ZB-5004 are visually similar and the remote when pared correctly displays as RGBgenie ZB-5001