Koenkk/zigbee2mqtt #6981: z2m does not start after an update

salopette, 2021-04-07:
Hello, I just made an update, now z2m doesn't start anymore.
Where is the mistake?

TI-Board
Zigbee2MQTT version 1.18.2-dev 
error 2021-04-07 08:49:27: Failed to call 'Configure' 'onZigbeeStarted' (TypeError: Cannot read property 'ID' of undefined
    at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:272:68)
    at Configure.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/configure.js:97:48)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:382:44))
error 2021-04-07 08:49:27: Failed to call 'OnEvent' 'onZigbeeStarted' (TypeError: Cannot read property 'ID' of undefined
    at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:272:68)
    at OnEvent.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/onEvent.js:10:48)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:382:44))
error 2021-04-07 08:49:27: Failed to call 'Report' 'onZigbeeStarted' (TypeError: Cannot read property 'ID' of undefined
    at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:272:68)
    at Report.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/legacy/report.js:179:48)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:382:44))
info  2021-04-07 08:49:27: Started frontend on port 0.0.0.0:8080
error 2021-04-07 08:49:31: Not connected to MQTT server!

`Apr 07 08:48:09 raspberrypi systemd[1]: Stopped zigbee2mqtt.
Apr 07 08:48:09 raspberrypi systemd[1]: Started zigbee2mqtt.
Apr 07 08:48:10 raspberrypi npm[11818]: > zigbee2mqtt@1.18.2-dev start
Apr 07 08:48:10 raspberrypi npm[11818]: > node index.js
Apr 07 08:48:14 raspberrypi npm[11818]: Zigbee2MQTT:info  2021-04-07 08:48:14: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-04-07.08-48-12' filename: log.txt
Apr 07 08:48:14 raspberrypi npm[11818]: Zigbee2MQTT:info  2021-04-07 08:48:14: Starting Zigbee2MQTT version 1.18.2-dev (commit #ab0ae8c)
Apr 07 08:48:14 raspberrypi npm[11818]: Zigbee2MQTT:info  2021-04-07 08:48:14: Starting zigbee-herdsman (0.13.89)
Apr 07 08:48:17 raspberrypi npm[11818]: Zigbee2MQTT:info  2021-04-07 08:48:17: zigbee-herdsman started
Apr 07 08:48:17 raspberrypi npm[11818]: Zigbee2MQTT:info  2021-04-07 08:48:17: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'
Apr 07 08:48:17 raspberrypi npm[11818]: Zigbee2MQTT:info  2021-04-07 08:48:17: Set transmit power to 'null'
Apr 07 08:48:17 raspberrypi npm[11818]: Zigbee2MQTT:info  2021-04-07 08:48:17: Currently 57 devices are joined:
Apr 07 08:48:17 raspberrypi npm[11818]: Zigbee2MQTT:info  2021-04-07 08:48:17: BWM-Flur (0x0017880104b4de0f): 9290012607 - Philips Hue motion sensor (EndDevice)
Apr 07 08:48:17 raspberrypi npm[11818]: Zigbee2MQTT:info  2021-04-07 08:48:17: Wohnzimmer-Sensor (0x00158d00022cca23): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Apr 07 08:48:17 raspberrypi npm[11818]: Zigbee2MQTT:info  2021-04-07 08:48:17: BWM-WC-EG (0xec1bbdfffe302dcc): BW-IS3 - BlitzWolf Rechargeable Zigbee PIR motion sensor (EndDevice)
Apr 07 08:48:17 raspberrypi npm[11818]: Zigbee2MQTT:info  2021-04-07 08:48:17: Flur-TRV (0x5c0272fffec3c99c): SEA801-Zigbee/SEA802-Zigbee - Saswell Thermostatic radiator valve (EndDevice)
Apr 07 08:48:17 raspberrypi npm[11818]: Zigbee2MQTT:error 2021-04-07 08:48:17: Failed to call 'Configure' 'onZigbeeStarted' (TypeError: Cannot read property 'ID' of undefined
Apr 07 08:48:17 raspberrypi npm[11818]:     at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:272:68)
Apr 07 08:48:17 raspberrypi npm[11818]:     at Configure.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/configure.js:97:48)
Apr 07 08:48:17 raspberrypi npm[11818]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:382:44))
Apr 07 08:48:17 raspberrypi npm[11818]: Zigbee2MQTT:error 2021-04-07 08:48:17: Failed to call 'OnEvent' 'onZigbeeStarted' (TypeError: Cannot read property 'ID' of undefined
Apr 07 08:48:17 raspberrypi npm[11818]:     at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:272:68)
Apr 07 08:48:17 raspberrypi npm[11818]:     at OnEvent.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/onEvent.js:10:48)
Apr 07 08:48:17 raspberrypi npm[11818]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:382:44))
Apr 07 08:48:17 raspberrypi npm[11818]: Zigbee2MQTT:error 2021-04-07 08:48:17: Failed to call 'Report' 'onZigbeeStarted' (TypeError: Cannot read property 'ID' of undefined
Apr 07 08:48:17 raspberrypi npm[11818]:     at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:272:68)
Apr 07 08:48:17 raspberrypi npm[11818]:     at Report.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/legacy/report.js:179:48)
Apr 07 08:48:17 raspberrypi npm[11818]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:382:44))
Apr 07 08:48:17 raspberrypi npm[11818]: Zigbee2MQTT:info  2021-04-07 08:48:17: Started frontend on port 0.0.0.0:8080
Apr 07 08:48:17 raspberrypi npm[11818]: (node:11830) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'ID' of undefined
Apr 07 08:48:17 raspberrypi npm[11818]:     at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:272:68)
Apr 07 08:48:17 raspberrypi npm[11818]:     at Controller.start (/opt/zigbee2mqtt/lib/controller.js:122:40)
Apr 07 08:48:17 raspberrypi npm[11818]:     at async start (/opt/zigbee2mqtt/index.js:43:5)
Apr 07 08:48:17 raspberrypi npm[11818]: (node:11830) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
Apr 07 08:48:17 raspberrypi npm[11818]: (node:11830) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Apr 07 08:48:20 raspberrypi npm[11818]: /opt/zigbee2mqtt/lib/zigbee.js:272
Apr 07 08:48:20 raspberrypi npm[11818]:                 endpointName: utils.getKey(endpointNames, endpoint.ID, null, ((v) => v === 'default' ? null : v)),
Apr 07 08:48:20 raspberrypi npm[11818]:                                                                    ^
Apr 07 08:48:20 raspberrypi npm[11818]: TypeError: Cannot read property 'ID' of undefined
Apr 07 08:48:20 raspberrypi npm[11818]:     at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:272:68)
Apr 07 08:48:20 raspberrypi npm[11818]:     at Frontend.onWebSocketConnection (/opt/zigbee2mqtt/lib/extension/frontend.js:91:48)
Apr 07 08:48:20 raspberrypi npm[11818]:     at WebSocketServer.emit (events.js:314:20)
Apr 07 08:48:20 raspberrypi npm[11818]:     at /opt/zigbee2mqtt/lib/extension/frontend.js:69:30
Apr 07 08:48:20 raspberrypi npm[11818]:     at Frontend.authenticate (/opt/zigbee2mqtt/lib/extension/frontend.js:62:9)
Apr 07 08:48:20 raspberrypi npm[11818]:     at /opt/zigbee2mqtt/lib/extension/frontend.js:67:18
Apr 07 08:48:20 raspberrypi npm[11818]:     at WebSocketServer.completeUpgrade (/opt/zigbee2mqtt/node_modules/ws/lib/websocket-server.js:330:5)
Apr 07 08:48:20 raspberrypi npm[11818]:     at WebSocketServer.handleUpgrade (/opt/zigbee2mqtt/node_modules/ws/lib/websocket-server.js:246:10)
Apr 07 08:48:20 raspberrypi npm[11818]:     at Frontend.onUpgrade (/opt/zigbee2mqtt/lib/extension/frontend.js:66:18)
Apr 07 08:48:20 raspberrypi npm[11818]:     at Server.emit (events.js:314:20)
Apr 07 08:48:20 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Apr 07 08:48:20 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Apr 07 08:48:20 raspberrypi systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Apr 07 08:48:20 raspberrypi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 5.
Apr 07 08:48:20 raspberrypi systemd[1]: Stopped zigbee2mqtt.
`

salopette, 2021-04-07:
`Apr 07 08:57:41 raspberrypi systemd[1]: Started zigbee2mqtt.
Apr 07 08:57:43 raspberrypi npm[1305]: > zigbee2mqtt@1.18.2-dev start
Apr 07 08:57:43 raspberrypi npm[1305]: > node index.js
Apr 07 08:57:46 raspberrypi npm[1305]: Zigbee2MQTT:info  2021-04-07 08:57:46: Lo                                                                                                             gging to console and directory: '/opt/zigbee2mqtt/data/log/2021-04-07.08-57-45'                                                                                                              filename: log.txt
Apr 07 08:57:47 raspberrypi npm[1305]: Zigbee2MQTT:info  2021-04-07 08:57:47: St                                                                                                             arting Zigbee2MQTT version 1.18.2-dev (commit #ab0ae8c)
Apr 07 08:57:47 raspberrypi npm[1305]: Zigbee2MQTT:info  2021-04-07 08:57:47: St                                                                                                             arting zigbee-herdsman (0.13.89)
Apr 07 08:57:49 raspberrypi npm[1305]: Zigbee2MQTT:info  2021-04-07 08:57:49: zi                                                                                                             gbee-herdsman started
Apr 07 08:57:49 raspberrypi npm[1305]: Zigbee2MQTT:info  2021-04-07 08:57:49: Co                                                                                                             ordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"pr                                                                                                             oduct":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'
Apr 07 08:57:49 raspberrypi npm[1305]: Zigbee2MQTT:info  2021-04-07 08:57:49: Se                                                                                                             t transmit power to 'null'
Apr 07 08:57:49 raspberrypi npm[1305]: Zigbee2MQTT:info  2021-04-07 08:57:49: Cu                                                                                                             rrently 57 devices are joined:
Apr 07 08:57:49 raspberrypi npm[1305]: Zigbee2MQTT:info  2021-04-07 08:57:49: BW                                                                                                             M-Flur (0x0017880104b4de0f): 9290012607 - Philips Hue motion sensor (EndDevice)
Apr 07 08:57:49 raspberrypi npm[1305]: Zigbee2MQTT:info  2021-04-07 08:57:49: Wo                                                                                                             hnzimmer-Sensor (0x00158d00022cca23): WSDCGQ11LM - Xiaomi Aqara temperature, hum                                                                                                             idity and pressure sensor (EndDevice)
Apr 07 08:57:49 raspberrypi npm[1305]: Zigbee2MQTT:info  2021-04-07 08:57:49: BW                                                                                                             M-WC-EG (0xec1bbdfffe302dcc): BW-IS3 - BlitzWolf Rechargeable Zigbee PIR motion                                                                                                              sensor (EndDevice)
Apr 07 08:57:49 raspberrypi npm[1305]: Zigbee2MQTT:info  2021-04-07 08:57:49: Fl                                                                                                             ur-TRV (0x5c0272fffec3c99c): SEA801-Zigbee/SEA802-Zigbee - Saswell Thermostatic                                                                                                              radiator valve (EndDevice)
Apr 07 08:57:49 raspberrypi npm[1305]: Zigbee2MQTT:error 2021-04-07 08:57:49: Fa                                                                                                             iled to call 'Configure' 'onZigbeeStarted' (TypeError: Cannot read property 'ID'                                                                                                              of undefined
Apr 07 08:57:49 raspberrypi npm[1305]:     at Zigbee.resolveEntity (/opt/zigbee2                                                                                                             mqtt/lib/zigbee.js:272:68)
Apr 07 08:57:49 raspberrypi npm[1305]:     at Configure.onZigbeeStarted (/opt/zi                                                                                                             gbee2mqtt/lib/extension/configure.js:97:48)
Apr 07 08:57:49 raspberrypi npm[1305]:     at Controller.callExtensionMethod (/o                                                                                                             pt/zigbee2mqtt/lib/controller.js:382:44))
Apr 07 08:57:49 raspberrypi npm[1305]: Zigbee2MQTT:error 2021-04-07 08:57:49: Fa                                                                                                             iled to call 'OnEvent' 'onZigbeeStarted' (TypeError: Cannot read property 'ID' o                                                                                                             f undefined
Apr 07 08:57:49 raspberrypi npm[1305]:     at Zigbee.resolveEntity (/opt/zigbee2                                                                                                             mqtt/lib/zigbee.js:272:68)
[... cut ...]
Apr 07 08:57:51 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited                                                                                                             , code=exited, status=1/FAILURE
Apr 07 08:57:51 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result                                                                                                              'exit-code'.
Apr 07 08:57:51 raspberrypi systemd[1]: zigbee2mqtt.service: Service RestartSec=                                                                                                             100ms expired, scheduling restart.
`

salopette, 2021-04-07:
> 0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.12
3 info using node@v12.22.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.18.2-dev~prestart: zigbee2mqtt@1.18.2-dev
6 info lifecycle zigbee2mqtt@1.18.2-dev~start: zigbee2mqtt@1.18.2-dev
7 verbose lifecycle zigbee2mqtt@1.18.2-dev~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.18.2-dev~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.18.2-dev~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.18.2-dev~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.18.2-dev~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.18.2-dev~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.18.2-dev start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:314:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:314:20)
13 verbose stack     at maybeClose (internal/child_process.js:1022:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)
14 verbose pkgid zigbee2mqtt@1.18.2-dev
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 5.10.20-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v12.22.1
19 verbose npm  v6.14.12
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.18.2-dev start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.18.2-dev start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

salopette, 2021-04-07:
I have now removed everything and reinstalled z2m.
configuration and database written back, but it does not start.

`0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.12
3 info using node@v14.16.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.18.2~prestart: zigbee2mqtt@1.18.2
6 info lifecycle zigbee2mqtt@1.18.2~start: zigbee2mqtt@1.18.2
7 verbose lifecycle zigbee2mqtt@1.18.2~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.18.2~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle zigbee2mqtt@1.18.2~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.18.2~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.18.2~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.18.2~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.18.2 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:315:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:315:20)
13 verbose stack     at maybeClose (internal/child_process.js:1048:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
14 verbose pkgid zigbee2mqtt@1.18.2
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 5.10.20-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v14.16.1
19 verbose npm  v6.14.12
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.18.2 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.18.2 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
`
`pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.18.2 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2021-04-07 12:04:55: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-04-07.12-04-53' filename: log.txt
Zigbee2MQTT:info  2021-04-07 12:04:55: Starting Zigbee2MQTT version 1.18.2 (commit #abd8a09)
Zigbee2MQTT:info  2021-04-07 12:04:55: Starting zigbee-herdsman (0.13.88)
Zigbee2MQTT:error 2021-04-07 12:06:05: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-04-07 12:06:05: Failed to start zigbee
Zigbee2MQTT:error 2021-04-07 12:06:05: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-04-07 12:06:05: Exiting...
Zigbee2MQTT:error 2021-04-07 12:06:05: Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:172:23
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.2 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.18.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-04-07T10_06_05_884Z-debug.log
`

salopette, 2021-04-07:
Update to dev, help not. ever bootloop

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.12
3 info using node@v14.16.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.18.2-dev~prestart: zigbee2mqtt@1.18.2-dev
6 info lifecycle zigbee2mqtt@1.18.2-dev~start: zigbee2mqtt@1.18.2-dev
7 verbose lifecycle zigbee2mqtt@1.18.2-dev~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.18.2-dev~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.18.2-dev~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.18.2-dev~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.18.2-dev~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.18.2-dev~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.18.2-dev start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:315:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:315:20)
13 verbose stack     at maybeClose (internal/child_process.js:1048:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
14 verbose pkgid zigbee2mqtt@1.18.2-dev
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 5.10.20-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v14.16.1
19 verbose npm  v6.14.12
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.18.2-dev start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.18.2-dev start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

salopette, 2021-04-07:
`Apr 07 14:05:23 raspberrypi systemd[1]: Stopped zigbee2mqtt.
Apr 07 14:05:23 raspberrypi systemd[1]: Started zigbee2mqtt.
Apr 07 14:05:25 raspberrypi npm[2626]: > zigbee2mqtt@1.18.2-dev start /opt/zigbee2mqtt
Apr 07 14:05:25 raspberrypi npm[2626]: > node index.js
Apr 07 14:05:28 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:28: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-04-07.14-05-27' filename: log.txt
Apr 07 14:05:29 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:29: Starting Zigbee2MQTT version 1.18.2-dev (commit #ab0ae8c)
Apr 07 14:05:29 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:29: Starting zigbee-herdsman (0.13.89)
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:31: zigbee-herdsman started
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:31: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:31: Set transmit power to 'null'
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:31: Currently 9 devices are joined:
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:31: SW-Andrej-Bett (0x00158d0002d7b87a): WXKG02LM_rev2 - Xiaomi Aqara double key wireless wall switch (2018 model) (EndDevice)
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:31: BWM Hunde-Licht (0x0017880104b55f1c): 9290012607 - Philips Hue motion sensor (EndDevice)
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:31: BWM-WC-EG (0xec1bbdfffe302dcc): BW-IS3 - BlitzWolf Rechargeable Zigbee PIR motion sensor (EndDevice)
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:31: Flur-TRV (0x5c0272fffec3c99c): SEA801-Zigbee/SEA802-Zigbee - Saswell Thermostatic radiator valve (EndDevice)
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:31: BWM-Flur (0x0017880104b4de0f): 9290012607 - Philips Hue motion sensor (EndDevice)
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:31: Wohn-TRV (0x5c0272fffec216fe): SEA801-Zigbee/SEA802-Zigbee - Saswell Thermostatic radiator valve (EndDevice)
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:31: BWM Treppe EG (0x0017880108641ebc): 9290012607 - Philips Hue motion sensor (EndDevice)
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:31: WC-TRV (0x5c0272fffec3c8c5): SEA801-Zigbee/SEA802-Zigbee - Saswell Thermostatic radiator valve (EndDevice)
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:error 2021-04-07 14:05:31: Failed to call 'Configure' 'onZigbeeStarted' (TypeError: Cannot read property 'ID' of undefined
Apr 07 14:05:31 raspberrypi npm[2626]:     at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:272:68)
Apr 07 14:05:31 raspberrypi npm[2626]:     at Configure.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/configure.js:97:48)
Apr 07 14:05:31 raspberrypi npm[2626]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:382:44))
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:error 2021-04-07 14:05:31: Cannot find '0x7cb03eaa00ac9427/3' of group 'Color_rgbw'
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:error 2021-04-07 14:05:31: Cannot find '0xf0d1b8000012bc0d/1' of group 'Color_rgbw'
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:error 2021-04-07 14:05:31: Cannot find '0xf0d1b8000012aca2/1' of group 'Color_rgbw'
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:error 2021-04-07 14:05:31: Cannot find '0x7cb03eaa00a92675/3' of group 'Juli_rgbw'
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:error 2021-04-07 14:05:31: Cannot find '0x7cb03eaa00adc1ab/3' of group 'Juli_rgbw'
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:error 2021-04-07 14:05:31: Cannot find '0x7cb03eaa00a91032/3' of group 'Juli_rgbw'
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:error 2021-04-07 14:05:31: Failed to call 'OnEvent' 'onZigbeeStarted' (TypeError: Cannot read property 'ID' of undefined
Apr 07 14:05:31 raspberrypi npm[2626]:     at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:272:68)
Apr 07 14:05:31 raspberrypi npm[2626]:     at OnEvent.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/onEvent.js:10:48)
Apr 07 14:05:31 raspberrypi npm[2626]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:382:44))
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:error 2021-04-07 14:05:31: Failed to call 'Report' 'onZigbeeStarted' (TypeError: Cannot read property 'ID' of undefined
Apr 07 14:05:31 raspberrypi npm[2626]:     at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:272:68)
Apr 07 14:05:31 raspberrypi npm[2626]:     at Report.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/legacy/report.js:179:48)
Apr 07 14:05:31 raspberrypi npm[2626]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:382:44))
Apr 07 14:05:31 raspberrypi npm[2626]: Zigbee2MQTT:info  2021-04-07 14:05:31: Started frontend on port 0.0.0.0:8080
Apr 07 14:05:31 raspberrypi npm[2626]: (node:2644) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'ID' of undefined
Apr 07 14:05:31 raspberrypi npm[2626]:     at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:272:68)
[... cut ...]
Apr 07 14:05:32 raspberrypi npm[2626]: npm ERR!     /home/pi/.npm/_logs/2021-04-07T12_05_32_399Z-debug.log
Apr 07 14:05:32 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Apr 07 14:05:32 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Apr 07 14:05:32 raspberrypi systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Apr 07 14:05:32 raspberrypi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 34.
Apr 07 14:05:32 raspberrypi systemd[1]: Stopped zigbee2mqtt.
`

TheStigh, 2021-04-07:
What zigbee adapter are you using? You said "TI", there are several models to choose from...
Please **unplug and re-plug the adapter** to your PI while Z2M are running, then restart Z2M after re-plugged.

salopette, 2021-04-07:
my adapter is TI LAUNCHXL-CC26X2R1 

I've been fighting for 5 hours without success.

Have already deleted everything and reinstalled it several times.
And also the Ti board re-flashed.

I have 57 devices that are now without function :-(
What I still have left is my conf.yaml and yesterday's .db, but I can't restore anything.

salopette, 2021-04-07:
Here is another attempt to start.

> pi@raspberrypi:/opt/zigbee2mqtt $ sudo systemctl stop zigbee2mqtt
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.18.2-dev start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2021-04-07 14:56:09: Logging to console and directory: '/opt/zigbee2mqtt/da                                                                                                ta/log/2021-04-07.14-56-08' filename: log.txt
Zigbee2MQTT:info  2021-04-07 14:56:10: Starting Zigbee2MQTT version 1.18.2-dev (commit #ab0ae                                                                                                8c)
Zigbee2MQTT:info  2021-04-07 14:56:10: Starting zigbee-herdsman (0.13.89)
Zigbee2MQTT:error 2021-04-07 14:57:12: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-04-07 14:57:12: Failed to start zigbee
Zigbee2MQTT:error 2021-04-07 14:57:12: Check https://www.zigbee2mqtt.io/information/FAQ.html#                                                                                                help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-04-07 14:57:12: Exiting...
Zigbee2MQTT:error 2021-04-07 14:57:12: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.                                                                                                js:67:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.2-dev start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.18.2-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output a                                                                                                bove.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-04-07T12_57_12_672Z-debug.log
pi@raspberrypi:/opt/zigbee2mqtt $

TheStigh, 2021-04-07:
First is to get the board started. I assume you have tried to re-flash to be sure all went well and you haven't forgotten any steps?
Second, are you SURE you point to the correct device in you configuration, meaning do you have other USB devices attached?

salopette, 2021-04-07:
everything fits in the configuration,
Port and mqtt.
I have refläsht the board to be sure, but it is the same FW that was installed before.

TheStigh, 2021-04-07:
Assume you have tried everything from the [FAQ](https://www.zigbee2mqtt.io/information/FAQ.html#) ?

It is essential to not taking any shortcuts when troubleshooting.

Personally, I've had this issue several times using dev-env, but it almost always resolves by;
- Power cycle harware (for me, an Intel NUC)
- Unplug and re-plug the adapter (use both CC2531 and 1352-P2 depending)
- Delete everything in /share/zigbee2mqtt
- Change PAN_ID (change of PAN will break existing zigbee network and require re-pairing)

I see you mention 57 devices not working anymore... Are you using the Dev-version in your production environment? This is not a wise decision, dev is dev, meaning things can break. Production env should ALWAYS be with the Release version.

cysp74, 2021-04-07:
I faced with the same issue.

Something is wrong with the code after commit 6bf27b9976448d534122af80efebdfca81a7d185 and had to revert to this. Since this commit cannot add new devices and could not start z2m neither.
Tried to reflash CC2652R - no effect.
Tried to change pan_id - no effect.

Sadly my database.db wiped out - but I realized, devices couldn't join and any attempt from devices to connect caused z2m restart, so I went back to commit mentioned above and now works as before - expect I have almost empty database.db

Koenkk, 2021-04-07:
This bug was indeed introduced in the dev branch. Fixed the `TypeError: Cannot read property 'ID' of undefined
at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:272:68)` error now.

salopette, 2021-04-07:
Can you restore everything now?
if I reset the PAnID to the old one?

TheStigh, 2021-04-07:
> 
> 
> Can you restore everything now?
> if I reset the PAnID to the old one?

Do you have a backup of old files ? Then it should be doable

Koenkk, 2021-04-08:
Only if you have the backup file indeed. If you've changed your pan id and don't have the old backup file anymore you need to repair devices (since changing the pan id requires repairing all devices)

salopette, 2021-04-09:
I have now changed the pan id, 80% of my devices have gradually come back on their own. I had to re-pair the rest.

closed 2021-05-17 by github-actions[bot]